espressif/esp-idf #7584: WiFi AP, mDNS cannot be queried

stdenits, 2021-09-20:
## Environment

- Development Kit:      [ESP32-C3-DevKitM-1]
- Module or chip used:  [ESP32­-C3-­MINI-­1]
- IDF version (run ``git describe --tags`` to find it):    [ESP-IDF v4.4-dev-3042-g220590d599 2nd stage bootloader]
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):     [???]
- Operating System:     [Windows]
- (Windows only) environment type: [???].
- Using an IDE?: [VSCode]
- Power Supply:         [USB]

## Problem Description

Can ping WiFi AP with it own IP address, but can not do it via "mDNSname.local" or just "mDNSname" (netbios name?).
Also, can not connect to WS server.

#MDNS_STRICT_MODE  is set to 1

### Expected Behavior

### Actual Behavior

### Steps to reproduce

### Code to reproduce this issue
Can send full project via e-mail.

## Debug Logs

```
Query A: lampvest.local
Host was not found!
I (3685) ESP32_Server: Query PTR: _http._tcp.local
W (6785) ESP32_Server: No results found!
```
[log.zip](https://github.com/espressif/esp-idf/files/7197086/log.zip)

## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/7197041/sdkconfig.zip)

stdenits, 2021-09-21:
Tried and example for [mDNS and STA](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mdns).
I can not to ping any instances either except STA IP address.

```
I (726357) mdns-test: Query A: tinytester.local
W (728457) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!
I (729457) mdns-test: Query both A and AAA: esp32-mdns.local
I (730557) mdns-test: Query A esp32-mdns.local finished
I (730557) mdns-test: Query AAAA esp32-mdns.local finished
I (730557) mdns-test: Query A: esp32.local
W (732657) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!
I (732657) mdns-test: Query PTR: _arduino._tcp.local
W (735757) mdns-test: No results found!
I (735757) mdns-test: Query PTR: _http._tcp.local
W (738857) mdns-test: No results found!
I (738857) mdns-test: Query PTR: _printer._tcp.local
W (741957) mdns-test: No results found!
I (741957) mdns-test: Query PTR: _ipp._tcp.local
W (745057) mdns-test: No results found!
I (745057) mdns-test: Query PTR: _afpovertcp._tcp.local
W (748157) mdns-test: No results found!
I (748157) mdns-test: Query PTR: _smb._tcp.local
W (751257) mdns-test: No results found!
I (751257) mdns-test: Query PTR: _ftp._tcp.local
W (754357) mdns-test: No results found!
I (754357) mdns-test: Query PTR: _nfs._tcp.local
W (757457) mdns-test: No results found!
```

stdenits, 2021-09-23:
mDNS logs.

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d68
entry 0x403ce000
I (30) boot: ESP-IDF v4.4-dev-3042-g220590d599 2nd stage bootloader
I (30) boot: compile time 09:16:26
I (30) boot: chip revision: 3
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=18440h ( 99392) map
I (122) esp_image: segment 1: paddr=00028468 vaddr=3fc8ee00 size=02b3ch ( 11068) load
I (124) esp_image: segment 2: paddr=0002afac vaddr=40380000 size=0506ch ( 20588) load
I (132) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=9b024h (634916) map
I (235) esp_image: segment 4: paddr=000cb04c vaddr=4038506c size=09b98h ( 39832) load
I (243) esp_image: segment 5: paddr=000d4bec vaddr=50000000 size=00004h (     4) load
I (243) esp_image: segment 6: paddr=000d4bf8 vaddr=50000008 size=00074h (   116) load
I (254) boot: Loaded app from partition at offset 0x10000
I (256) boot: Disabling RNG early entropy source...
I (273) cpu_start: Pro cpu up.
I (282) cpu_start: Pro cpu start user code
I (282) cpu_start: cpu freq: 160000000
I (282) cpu_start: Application information:
I (284) cpu_start: Project name:     wifi_softAP
I (290) cpu_start: App version:      1
I (294) cpu_start: Compile time:     Sep 23 2021 10:02:27
I (300) cpu_start: ELF file SHA256:  3829486c7db1b8d0...
I (306) cpu_start: ESP-IDF:          v4.4-dev-3042-g220590d599
I (313) heap_init: Initializing. RAM available for dynamic allocation:
I (320) heap_init: At 3FC96240 len 00029DC0 (167 KiB): DRAM
I (326) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (333) heap_init: At 5000007C len 00001F84 (7 KiB): RTCRAM
I (340) spi_flash: detected chip: generic
I (344) spi_flash: flash io: dio
I (348) sleep: Configure to isolate all GPIO pins in sleep state
I (355) sleep: Enable automatic switching of GPIO sleep configuration
I (362) cpu_start: Starting scheduler.
I (366) ESP32_Server: app_main() runned on core 0

I (376) gpio: GPIO[7]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:3 
I (376) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (386) ESP32_mDNS: Initializing SPIFFS
I (416) ESP32_mDNS: Partition size: total: 956561, used: 819766
I (416) ESP32_Server: ESP_WIFI_MODE_AP / STA
I (416) ESP32_mDNS: mdns hostname set to: [lampvest.local]
I (426) pp: pp rom version: 9387209
I (426) net80211: net80211 rom version: 9387209
I (436) wifi:wifi driver task: 3fca287c, prio:23, stack:6656, core=0
I (436) system_api: Base MAC address is not set
I (446) system_api: read default base MAC address from EFUSE
I (446) wifi:wifi firmware version: d683787
I (456) wifi:wifi certification version: v7.0
I (456) wifi:config NVS flash: enabled
I (456) wifi:config nano formating: disabled
[... 186 lines cut ...]

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[17370]: 
    Q: oyugjiei.local. A IN

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[17370]:
    Q: xgbdzmy.local. A IN

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[18360]: 
    Q: ._googlecast._tcp.local. PTR IN

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[18360]:
    Q: ._googlecast._tcp.local. PTR IN

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[19320]: 
    Q: ._lxi._tcp.local. PTR IN
    Q: ._scpi-raw._tcp.local. PTR IN
    Q: ._vxi-11._tcp.local. PTR IN
    Q: ._hislip._tcp.local. PTR IN

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[19370]: 
    Q: ._googlecast._tcp.local. PTR IN

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[19380]: 
    Q: ._googlecast._tcp.local. PTR IN
```

david-cermak, 2021-09-27:
Hi @stdenits 

Does it mean that the basic mdns example from IDF doesn't work for you on esp32c3? This is our  very basic test scenario which is constantly tested. We've been reported some issues related to C3's wifi performance on earlier `v4.4-dev` which could influence multicasts and mdns, but your version seems quite recent. I've also tested Wifi AP with mdns on 220590d599e  and the hostname is correctly resolved by the standard `dns-sd` engine (and avahi) and also with `dig` one shot query.
 
Anyway, if the most basic scenario doesn't work for you indicates something wrong with
* the device itself: Very unlikely as the networking part works okay and you're also receiving mdns queries (based on the log above) -- but could you please try another board/chip?
* application code: Again very unlikely, as you reported that the most simple example doesn't work as expected -- but still could you please add these two lines [here](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/softAP/main/softap_example_main.c#L82)
```cpp
    ESP_ERROR_CHECK( mdns_init() );
    ESP_ERROR_CHECK( mdns_hostname_set("lampvest") );
```
* the resolver: What kind of service you use for mdns name and service resolution? (avahi, dns-sd, dig, some android/ios app?) Do you see the query packets in the logs? I'd suggest sending one shot queries as they are not cached and easily debugged and clearly seen in the debug log. you can run this command to post one such query:
```
dig -p 5353 @224.0.0.251 lampvest.local
```

Also, it would be nice to post a wireshark log to see if the mdns packets are present on the network if not received by the device.

stdenits, 2021-09-27:
@**david-cermak**,
Hello. 

>What kind of service you use for mdns name and service resolution? (avahi, dns-sd, dig, some android/ios app?) Do you see the query packets in the logs? I'd suggest sending one shot queries as they are not cached and easily debugged and clearly seen in the debug log.

Dig command output:
```
$ dig -p 5353 @224.0.0.251 lampvest.local

; <<>> DiG 9.16.21 <<>> -p 5353 @224.0.0.251 lampvest.local
; (1 server found)
;; global options: +cmd
;; Got answer:
;; WARNING: .local is reserved for Multicast DNS
;; You are currently testing what happens when an mDNS query is leaked to DNS
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 35528
;; flags: qr aa; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;lampvest.local.                        IN      A

;; ANSWER SECTION:
lampvest.local.         120     IN      A       192.168.4.1

;; Query time: 126 msec
;; SERVER: 192.168.4.1#5353(224.0.0.251)
;; WHEN: Mon Sep 27 17:16:40 Russia TZ 4 Standard Time 2021
;; MSG SIZE  rcvd: 57
```

Trying to ping \ nslookup.
Can not to ping lampvest.local
```
C:\Users\User>ping 192.168.4.1

Pinging 192.168.4.1 with 32 bytes of data:
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Reply from 192.168.4.1: bytes=32 time=4ms TTL=255
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Reply from 192.168.4.1: bytes=32 time=4ms TTL=255

Ping statistics for 192.168.4.1:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 4ms, Average = 2ms

C:\Users\User>ping lampvest

Pinging lampvest.local [192.168.4.1] with 32 bytes of data:
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255

Ping statistics for 192.168.4.1:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 1ms, Average = 1ms

C:\Users\User>ping lampvest.local
^C
C:\Users\User>ping lampvest

Pinging lampvest.local [192.168.4.1] with 32 bytes of data:
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Reply from 192.168.4.1: bytes=32 time=4ms TTL=255
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255

Ping statistics for 192.168.4.1:
[... 285 lines cut ...]
    A: lampvest.local. A IN FLUSH 120 [4] 192.168.4.1

TX[1][0]: To: 224.0.0.251:5353, Packet[268100]: AUTHORITATIVE
    A: lampvest.local. A IN FLUSH 120 [4] 192.168.4.1

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[268890]: 
    Q: lampvest.local. AAAA IN

TX[1][0]: To: 224.0.0.251:5353, Packet[269000]: AUTHORITATIVE
    A: lampvest.local. A IN FLUSH 120 [4] 192.168.4.1

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[270660]: 
    Q: ._lxi._tcp.local. PTR IN
    Q: ._scpi-raw._tcp.local. PTR IN
    Q: ._vxi-11._tcp.local. PTR IN
    Q: ._hislip._tcp.local. PTR IN
```

>the device itself: Very unlikely as the networking part works okay and you're also receiving mdns queries (based on the log above) -- but could you please try another board/chip?

Can not do it right now.

>Also, it would be nice to post a wireshark log to see if the mdns packets are present on the network if not received by the device.

Sorry, Wireshark is still a dark forest for me. I can't figure out what and how it should be... I'll try to do something.

david-cermak, 2021-09-27:
This looks like the hostname is resolved correctly
```
RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[268890]: 
    Q: lampvest.local. AAAA IN

TX[1][0]: To: 224.0.0.251:5353, Packet[269000]: AUTHORITATIVE
    A: lampvest.local. A IN FLUSH 120 [4] 192.168.4.1
```
The query is received, response is sent and received back by the sender of the query.

Or, are you reporting that pinging the hostname doesn't work on your platform, since it's mentioned in the example [readme](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mdns#build-and-flash) ?
Could you please check some standard mDNS querier? You can try to use "Bonjour Print Services for Windows", for example.

stdenits, 2021-09-28:
@**david-cermak** ,
>The query is received, response is sent and received back by the sender of the query.

Time to time.
Here is the saved Wireshark session, if it tells you anything.
[mDNS_problem.zip](https://github.com/espressif/esp-idf/files/7240282/mDNS_problem.zip)

>Or, are you reporting that pinging the hostname doesn't work on your platform, since it's mentioned in the example readme?
Everything worked before, but I'll check.

david-cermak, 2021-09-29:
@stdenits Thanks for sharing the packet capture. I thought initially that the device responded to one-shot queries (e.g. packets 200, 207, 2011) but didn't answer standard queries (like 309, 310), until checked the query content in detail. Please double check the spelling when resolving addresses, as the packets `309`, `310`, `323` and `324` which were not answered were actually trying to resolve the name of `lampbest`, while the answered packets (everything else) were querying `lampvest` hostname.
So in that sense I don't see any issue with the attached packet capture, all packets that are supposed to be answered were answered.

>  I still cannot PING any instances such as "lampvest" or "lampvest.local" 

I'm sorry, we cannot support `PING` in general. We could support a standard mDNS querrier or a simplified one-shot querrier. I suggest removing all references of `ping` in the example docs and listing all standard tooling that users can use to send mDNS queries (though ping _usually_ works on most platforms).

stdenits, 2021-10-01:
>Please double check the spelling when resolving addresses

Sorry, probably it is mistyping.

>So in that sense I don't see any issue with the attached packet capture, all packets that are supposed to be answered were answered.

Ok, is there any way to check why ESP32 device can not be accessed via mdns-name or netbios name?

david-cermak, 2021-10-01:
> is there any way to check why ESP32 device can not be accessed via mdns-name or netbios name?

As said above, your packet capture actually shows that the ESP32 device mdns-name gets *resolved correctly*.

if you're asking why you were not able access the device, I can see two reasons:
* typos in the hostname (as seen in the wireshark log)
* tools you use for mdns resolution don't send mdns queries (didn't see any other query in you wireshark log that wasn't answered)

I'd suggest you start using
* either some standard mdns tools for resolving names/services. This could be `dns-sd`, `avahi`, `bonjour service` or some mobile apps work good as well (I'm using `Service Browser`)
* or send one-shot mdns query (you can use any standard DNS resolver for that, e.g. `dig` as mentioned before)

ESP32 cannot a reply if the tool sends no queries. The device advertises its name and services, though. So you can try to re-init the mdns every time a new station connects to your AP -- this would be a *very* non-standard workaround.

From your logs and reports, I could only conclude that the mDNS library works as expected with your AP so I could only ask you to close this issue.

stdenits, 2021-10-02:
Thank you for your help and your time.

closed 2021-10-02 by stdenits