espressif/esp-idf #7585: adding third sensor to sensor_server example: BLE_MESH: Configuration Server only allowed in primary element

ChrisIdema, 2021-09-20:
I modified the sensor_server example: esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_sensor_model/sensor_server

If I add a third sensor to the sensor server example  it crashes with the following error:

`BLE_MESH: Configuration Server only allowed in primary element`

What is the cause of this?

How many octets do I need to set in ESP_BLE_MESH_MODEL_PUB_DEFINE?

ChrisIdema, 2021-09-21:
I solved the issue. I added several models to the code. I had to have the sensor server in the root model and that fixed the issue.

JiaLWang, 2021-09-22:
Hi ChrisIdema,

Of cause you can set the `msg_len` to 40 for 4 sensors.

The value of  `msg_len` is depend on the maximum length of all the message that the model can publish. About which message can publish  and the length of the message, please refer to `mesh model specification` document.

Chris-Idema-WizzDev, 2021-09-29:
Do you mean `Mesh_Model_Specification v1.0.pdf`?
If so I see the following messages for sensors send by the device (TX):

- Sensor Descriptor Status "8*N" bytes (I assume N is number of sensors)
- Sensor Status 5 bytes for 16-bit sensors in format B per sensor*
- Sensor Column Status 8 bytes per sensor for 16-bit sensors
- Sensor Series Status 2+6*N bytes for 16-bit sensors

*I assume sensor format B since I want to use custom sensors which have a custom property ID >= 0x800 in the future.

If my calculations are correct with 16-bit sensors you do not need more than 8 bytes per sensor, with 24-bit sensors 11 bytes and with 32-bit sensors you need 14 bytes per sensor.  This makes me wonder what the original example based 20 bytes for 2 8-bit sensors on.

ChrisIdema, 2021-09-29:
Do you mean Mesh_Model_Specification v1.0.pdf?
If so I see the following messages for sensors send by the device (TX):

    Sensor Descriptor Status "8*N" bytes (I assume N is number of sensors)
    Sensor Status 5 bytes for 16-bit sensors in format B per sensor*
    Sensor Column Status 8 bytes per sensor for 16-bit sensors
    Sensor Series Status 2+6*N bytes for 16-bit sensors

*I assume sensor format B since I want to use custom sensors which have a custom property ID >= 0x800 in the future.

If my calculations are correct with 16-bit sensors you do not need more than 8 bytes per sensor, with 24-bit sensors 11 bytes and with 32-bit sensors you need 14 bytes per sensor. This makes me wonder what the original example based 20 bytes for 2 8-bit sensors on. Maybe it was rounded up?