espressif/esp-idf #7587: Watchdog Reset Issue after OTA Update

codspeed, 2021-09-22:
## Environment

- Module or chip used:  ESP32-WROOM-32 
- IDF version (run ``git describe --tags`` to find it):
 v4.3 and v3.1
- Build System:        
idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:   
macOS
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
Yes, MS Visual Studio
- Power Supply:         [USB|external 5V|external 3.3V|Battery]
USB
## Problem Description
OTA upgrade triggering Watchdog Reset
//Detailed problem description goes here.
I have 2 applications v1.1 built on IDF v3.1 and V1.2 built on IDF v4.3. When I OTA upgrade v1.2 over v1.1 the device logs 2 resets. One is the reboot from the OTA (SW_CPU_RESET), then followed by the Watchdog reset (RTCWDT_RTC_RESET).

[image]

The same behaviour is observed on each factory reset.

I do not see the watchdog reset when doing an OTA upgrade using an application built with the same IDF version.
On debug I found that the v1.1 (IDF v3.1) does not have any configuration for 
CONFIG_BOOTLOADER_WDT_ENABLE 
CONFIG_BOOTLOADER_WDT_TIME_MS 
CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE
V1.2 (IDF v4.3) has configurations for 
CONFIG_BOOTLOADER_WDT_ENABLE=y 
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
 
Bootloader options for IDF v3.1

[image]

Bootloader options for IDF v4.3

[image]

Please see attached logs (esp_ticket_watchdog_reset.rtf). Once it is upgraded to IDF v4.3 compiled image see log for SW_CPU_RESET followed by RTCWDT_RTC_RESET and again SW_CPU_RESET.

Could anyone help me understand why an application built with IDF v4.3 logs a RTCWDT_RTC_RESET when upgraded on a device running an application built with IDF v3.1 
How can I avoid this ?
### Expected Behavior
After OTA or factory reset a SW_CPU_RESET
### Actual Behavior
After OTA or factory reset a SW_CPU_RESET followed by RTCWDT_RTC_RESET
### Steps to reproduce

1. Compile an application image with ESP-IDF v3.1
2. Flash bootloader and application
3. OTA upgrade the device using application image build on ESP IDF v4.3
4. After upgrade the device resets with reason SW_CPU_RESET followed by RTCWDT_RTC_RESET

// If possible, attach a picture of your setup/wiring here.
Not applicable

### Code to reproduce this issue
Not Applicable

## Debug Logs
See attached file esp_ticket_watchdog_reset.rtf
```
I (10:07:46.057) esp_insights: Next cloud reporting is scheduled after 300 seconds
ets Jun 8 2016 00:22:57
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:944
ho 0 tail 12 room 4
load:0x40078000,len:9860
load:0x40080000,len:5284
0x40080000: _WindowOverflow4 at /Users/user/esp/esp-idf-v4.3/components/freertos/port/xtensa/xtensa_vectors.S:1730
entry 0x40080274
0x40080274: _Level5Vector at ??:?
I (624) cpu_start: Pro cpu up.
I (624) cpu_start: Starting app cpu, entry point is 0x40081528
0x40081528: call_start_cpu1 at /Users/user/esp/esp-idf-v4.3/components/esp_system/port/cpu_start.c:179

```
## Other items if possible

- [ ✓] sdkconfig file (attach the sdkconfig file from your project folder)
See attached sdkconfig_IDF_v3.1 and sdkconfig_IDF_v4.3
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

[drive-download-20210922T045729Z-001.zip](https://github.com/espressif/esp-idf/files/7207891/drive-download-20210922T045729Z-001.zip)

laukik-hase, 2021-09-27:
Hello, @codspeed. Do you get the same behavior when -
1. Security features (Flash Encryption & Secure Boot) are disabled
2. Running with IDF examples (e.g for v3.1, the [native_ota](https://github.com/espressif/esp-idf/tree/release/v3.1/examples/system/ota/native_ota_example) and [simple_ota](https://github.com/espressif/esp-idf/tree/release/v3.1/examples/system/ota/simple_ota_example) examples)

closed 2021-10-08 by codspeed

laukik-hase, 2021-10-11:
Hello, @codspeed 

Could you share if this issue was resolved and how? It would be helpful for other users.