espressif/esp-idf #7588: Failed build after change main.c name

Szymono, 2021-09-22:
For exampe: 

I changed file name main.c to test.c. Made clean and try to make build, but build is failed.
`undefined reference to `app_main'`

I'm new in esp, maybe I dont understand whole idf system, but I found bug or not perfect action.
IMHO if I change mian.c name, esp-idf should change file name in `main/CMakeList.txt`, but **the name is missed**
``` txt
idf_component_register(SRCS
                    INCLUDE_DIRS ".")
```

After I added the new name, all works good. 

``` txt
idf_component_register(SRCS "test.c"
                    INCLUDE_DIRS ".")
```
If u can, make this automatic please :)

chegewara, 2021-09-22:
https://github.com/espressif/esp-idf/blob/master/examples/get-started/hello_world/main/CMakeLists.txt#L1

chegewara, 2021-09-22:
You can do it automatically, just you have to find cmake code that will add all files with .c and/or .cpp to SRC.

Something like this:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/CMakeLists.txt#L11

Dazza0, 2021-09-29:
@Szymono 

- The `SRCS` argument requires that you list each source file in your component, in this case the component is the `main` component of your project. The source file names **must match exactly** for them to be compiled.
- However, it can be inconvenient t have to modify `SRCS` if you plan to add/change a lot of source files in `main`. Therefore, you can use `SRC_DIRS` instead. This allows you to list directories that contain source files, and the build system will automatically glob all `.c`, `.cpp`, and `.S` files in those directories.

The following will add your `main` folder as a source directory of the component. Refer to the [build system documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#idf-component-commands) for more details. 

```
idf_component_register(SRC_DIRS "."
                    INCLUDE_DIRS ".")
```

I'm closing this issue as it's not a bug with the build system. Feel free to reopen

closed 2021-09-29 by Dazza0

Japczy, 2021-09-29:
Thx, but still I don't know why `main.c` missed in cmake file after rename `main.c`

Second of all, I think esp-idf can give information when file `*.c` in cmake not found e.g. `"Problem in main/CMakeList.txt file"`

Dazza0, 2021-09-29:
> Thx, but still I don't know why `main.c` missed in cmake file after rename `main.c`

Run `idf.py fullclean` and build again. Remember that CMake is not the build system itself but a build system generator (e.g., generating Makefile files), so it won't be able to detect the addition of new files unless you run CMake again to generate those build system files again.

> Second of all, I think esp-idf can give information when file `*.c` in cmake not found e.g. `"Problem in main/CMakeList.txt file"`

It already does, if your `SRC` lists a file that does not exist, CMake will generate an error such as the following:

```
Cannot find source file:
    /home/user/my_project/main/some_file.c
```

Szymono, 2021-10-01:
> It already does, if your `SRC` lists a file that does not exist, CMake will generate an error such as the following:
> 
> ```
> Cannot find source file:
>     /home/user/my_project/main/some_file.c
> ```

Ofcourse it works. Sorry, my mistake. Unfortunately no similar error when no any path in `main/CMakeList.txt`. (as describen in [first post](https://github.com/espressif/esp-idf/issues/7588#issue-1004484080)