espressif/esp-idf #7589: Debugger: Debugger is not stopping on Breakpoints after "esp_restart()"

carlessole, 2021-09-22:
- Module or chip used:  ESP32-WROVER (With ESP_PROG)
- IDF version: v4.3
- Build System: CMake/idf.py (usingEclipse Plugin)
- Compiler version: esp-2020r3-8.4.0
- Operating System: Windows
- (Windows only) environment type: -.
- Using an IDE?: Yes. Eclipse + ESP-IDF plugin
- Power Supply: USB

When I run the deubgger on my project, it is working fine and it is stopping and the breawkpoints that I set.
If I reset the program using the IDE button:
[image]

I have no problems.

But, if in somewhere on my code I have decided to restart the program using code:

`esp_restart()`

Device is rebooting ( :) ), I can see that taking a look on the debug console output **but debugger is not stopping on the breakpoints.**

Does anyone knows why that is happening? It should not be happening right?

Thanks

Carles

igrr, 2021-09-23:
Hi @carlessole, unfortunately at the moment esp_restart resets that CPU, which causes hardware breakpoints to be reset as well. The debugger doesn't get notified about the reset and therefore doesn't re-insert the breakpoints. We'll think about some way to solve this issue.

carlessole, 2021-09-23:
Thanks @igrr .
As you have talked about **Hardware** Breakpoints...

> which causes hardware breakpoints to be reset as well
So, the same is happening with Software breakpoints?

And, how I can distinguish between a Hardware or a Software Breakpoint? I mean, how can I decide to set a HW or SW breakpoint when setting a breakpoint? I can not find different settings for each one.

[image]

[image]

Thanks,

Carles

carlessole, 2022-02-04:
> Thanks @igrr . As you have talked about **Hardware** Breakpoints...
>
>
> So, the same is happening with Software breakpoints?
>
> And, how I can distinguish between a Hardware or a Software Breakpoint? I [...]

Probably a dummy question, but @igrr , any info about that?

Thanks