espressif/esp-idf #7590: Could not find a version that satisfies the requirement bidict>=0.21.0

b1tr4t, 2021-09-23:
```
esp-idf$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Installing Python environment and packages
Installing Python packages from /home/sean/develop/dad-phone/pub/esp32/esp-idf/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
.
.
.
.
Collecting greenlet>=0.4.14; platform_python_implementation == "CPython"
  Using cached greenlet-1.1.1-cp27-cp27mu-manylinux2010_x86_64.whl (149 kB)
Collecting MarkupSafe>=0.23
  Using cached MarkupSafe-1.1.1-cp27-cp27mu-manylinux1_x86_64.whl (24 kB)
ERROR: Could not find a version that satisfies the requirement bidict>=0.21.0 (from python-socketio>=1.6.1->Flask-SocketIO<3.0,>=2.9->gdbgui==0.13.2.0->-r /home/sean/develop/dad-phone/pub/esp32/esp-idf/requirements.txt (line 14)) (from versions: 0.1.5, 0.2.1, 0.3.0, 0.3.1, 0.9.0rc0, 0.9.0.post1, 0.10.0, 0.10.0.post1, 0.11.0, 0.12.0.post1, 0.13.0, 0.13.1, 0.14.0, 0.14.1, 0.14.2, 0.15.0.dev0, 0.15.0.dev1, 0.15.0rc1, 0.15.0, 0.16.0, 0.17.0, 0.17.1, 0.17.2, 0.17.3, 0.17.4, 0.17.5, 0.18.0, 0.18.1, 0.18.2, 0.18.3, 0.18.4)
ERROR: No matching distribution found for bidict>=0.21.0 (from python-socketio>=1.6.1->Flask-SocketIO<3.0,>=2.9->gdbgui==0.13.2.0->-r /home/sean/develop/dad-phone/pub/esp32/esp-idf/requirements.txt (line 14))
Traceback (most recent call last):
  File "/home/sean/develop/dad-phone/pub/esp32/esp-idf/tools/idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "/home/sean/develop/dad-phone/pub/esp32/esp-idf/tools/idf_tools.py", line 1488, in main
    action_func(args)
  File "/home/sean/develop/dad-phone/pub/esp32/esp-idf/tools/idf_tools.py", line 1215, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/sean/.espressif/python_env/idf4.2_py2.7_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/sean/develop/dad-phone/pub/esp32/esp-idf/requirements.txt']' returned non-zero exit status 1

```

dobairoland, 2021-09-23:
@b1tr4t Are you using Ubuntu? Which version?

The use of Python 2 is not recommended. 

If you install the package dependencies from [this page](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/get-started/linux-setup.html) then the latest version of ESP-IDF v4.2.x will pick up Python 3.

Please also note that there is a known issue with [Ubuntu 16.04](
https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/get-started/linux-setup.html#fixing-broken-pip-on-ubuntu-16-04) which can be solved by following the steps of the guide.

b1tr4t, 2021-09-23:
Sorry about that.  It turns out I missed this step:

sudo apt-get install git wget flex bison gperf python3 python3-pip python3-setuptools cmake ninja-build ccache libffi-dev libssl-dev dfu-util

All good now, thanks!  

:-)

dobairoland, 2021-09-23:
I'm happy that it got solved!

closed 2021-09-23 by dobairoland