espressif/esp-idf #7593: WIFI_EVENT_STA_DISCONNECTED with reason=34 (not part of the err enum)

0xFEEDC0DE64, 2021-09-23:
Hi,
a customer has a home wifi setup where the esp32 is not able to connect to his homenetwork.
I tried debugging remotely and got the conclusion, that esp-idf is emitting a WIFI_EVENT with WIFI_EVENT_STA_DISCONNECTED with reason = 34.

Reason 34 is not part of the `wifi_err_reason_t`.

What does reason 34 mean? Is it coming from the closed-source binary blobs which do all the wifi magic? Can someone fill out the missing error reasons?

0xFEEDC0DE64, 2021-09-23:
Something to add here: If we don't use dhcpc to acquire an ip but set a static ip address, everything works fine. So error 34 might mean "no dhcp found on the network" ?

sagb2015, 2021-09-28:
Hi @0xFEEDC0DE64, the reason code 34 is "Association denied due to excessive frame loss rates and/or poor conditions on current operating channel". The reason codes < 200 are defined by IEEE standard and we covered most common ones in the enum. We will update the structure to cover all the defined ones.

AxelLin, 2021-09-28:
> Hi @0xFEEDC0DE64, the reason code 34 is "Association denied due to excessive frame loss rates and/or poor conditions on current operating channel". 

But this does not explain that it works if using static ip.
Do you need more information about the AP brand/model/configurations?

nishanth-radja, 2021-09-28:
@0xFEEDC0DE64 Can we have the sniffer capture and SDK config.Also can you enable the supplicant logs and MBEDTLS logs and try the connection once.What is the AP security config.
If it is wpa2,Pls start the sniffer before association so that the 4 way handshake is captured and pls provide the password of the network,so that we can decrypt the packet. 
Also pls mention the IDF version you are using.
Are other clients able to successfully connected to the AP and get the IP address?
Pls provide the DHCP server configuration on the AP.

0xFEEDC0DE64, 2021-09-30:
Thanks for all your answers so far :)

I will try my best to get the esp32 log output remotely (by modifying the esp-idf so that they are sent over websocket to me) but doing a remote wifi capture will be impossible.

We are using an esp-idf branch/fork based on 4.4-dev 

Other clients are able to talk to the AP and also they get an ip address. I don't know the exact dhcp configuration on the customers wifi router (but if that is really the issue here, i will try to get at least that)

AxelLin, 2021-11-09:
> Hi @0xFEEDC0DE64, the reason code 34 is "Association denied due to excessive frame loss rates and/or poor conditions on current operating channel". The reason codes < 200 are defined by IEEE [...]

@sagb2015 
Just check the master tree, still no update regarding the enum.

AxelLin, 2021-11-09:
> Thanks for all your answers so far :)
> 
> I will try my best to get the esp32 log output remotely (by modifying the esp-idf so that they are sent over websocket to me) but doing a remote wifi capture will be impossible.

@0xFEEDC0DE64 
Any update?

AxelLin, 2022-01-14:
@0xFEEDC0DE64 

Does this issue still happen or already fixed?
I check the whole discussions but couldn't find the exactly esp-idf version,
what is the "git describe --tags" output of this issue?

BTW, since you are using v4.4, does it work with latest v4.4 code (e.g v4.4-beta1-308-gf3e0c8bc41)?

0xFEEDC0DE64, 2022-01-14:
i just checked with the newest esp-idf master, i cannot reproduce this error 34 with the same access point anymore, seems like a random binary blob update fixed the issues somehow.
I think its about time to start up the decompilers and start paralell development of an open source wifi blob :)

closed 2022-01-14 by 0xFEEDC0DE64

AxelLin, 2022-01-27:
> Hi @0xFEEDC0DE64, the reason code 34 is "Association denied due to excessive frame loss rates and/or poor conditions on current operating channel". The reason codes < 200 are defined by IEEE [...]

@sagb2015 I don't see any update for the wifi_err_reason_t.