espressif/esp-idf #7594: GPIO interrupts don't work if `gpio_install_isr_service` executes on APP cpu

KaeLL, 2021-09-23:
## Environment

- IDF version: v4.4-dev-3042-g220590d599

## Problem Description

Title. Peripheral raises the GPIOs but the interrupt doesn't fire or isn't serviced. On the other hand, if everything but `gpio_install_isr_service` executes on APP cpu, it all works fine.
It's something related to running `gpio_isr_register_on_core_static` on the APP cpu.

chegewara, 2021-11-02:
Sorry for offtopic
@pawel-soja can i contact you somehow please?

pawel-soja, 2021-11-02:
> Sorry for offtopic @pawel-soja can i contact you somehow please?

Please see my github profile.

AxelLin, 2021-11-06:
It was working in v4.3, this was broken by 874a720286b78e866684129ce18a6c948c8fa7fe.

nonoo, 2021-11-08:
I'm using ESP32-S3-DevKitC-1 v1.0 with the latest ESP-IDF (d0dd9ce1b9825c29ae2b462556c4bf3965ab774b) and by comparing the SOC `gpio_struct.h` header files between ESP32 and ESP32S3 (esp-idf/components/soc/esp32s3/include/soc/gpio_struct.h). It seems that the ESP32S3 does not have the `acpu_int` structure, so it can't receive GPIO interrupts on the app CPU.

Is this true?

I can't find any documentation about this, and the `gpio_install_isr_service()` and `gpio_isr_handler_add()` calls complete successfully, they don't give any warning about this.

AxelLin, 2021-11-11:
> It was working in v4.3, this was broken by [874a720](https://github.com/espressif/esp-idf/commit/874a720286b78e866684129ce18a6c948c8fa7fe).

@o-marshmallow 
Any follow up?

o-marshmallow, 2021-11-11:
Sorry for the delay, 
Thanks @AxelLin for the commit that broke everything and @pawel-soja for finding the exact line! 
I am going to handle this now

closed 2021-11-12 by espressif-bot