espressif/esp-idf #7596: RF calibration not skipped after deep sleep wake up on ESP32C3

glitch911, 2021-09-23:
-## Environment

- Module or chip used: ESP32-C3-MINI-1
- IDF version (run ``git describe --tags`` to find it): 4.3.1
- Build System:   default vscode + Esp-idf extension 
- Compiler version): default vscode + Esp-idf extension 
- Operating System:    Windows
- (Windows only) environment type: default vscode + Esp-idf extension .
- Using an IDE?: default vscode + Esp-idf extension 
- Power Supply:       external regulated 3.3V

## Problem Description

Hello,

### Expected Behavior
From documentation (https://docs.espressif.com/projects/esp-idf/en/v4.3.1/esp32c3/api-guides/RF_calibration.html) RF calibration is not supposed to occur after wake-up from deep sleep as long as phy calibration data are available.

I use same code with former ESP32 module and I can see on current measurement the RF calibration at power-up and no calibration after wakeup from deep sleep (= expected behavior).

### Actual Behavior
But with my new ESP32-C3 setup I still observe RF calibration (big current spike) after wake-up from deep sleep :(

### Steps to reproduce
Take wifi scan example, set target to esp32c3, add headers:

```
#include "driver/rtc_io.h"
#include "esp_sleep.h"
```
and 
after wifi_scan(); 

```
    const gpio_config_t config = {
        .pin_bit_mask = BIT(1),
        .mode = GPIO_MODE_INPUT,
    };
    gpio_config(&config);

   esp_deep_sleep_enable_gpio_wakeup(BIT(1), ESP_GPIO_WAKEUP_GPIO_HIGH);

   esp_deep_sleep_start();`
```

Then after first scan module go to deep sleep and you can wake it up by pulling GPIO1 to VDD and observe RF calibration that should not occur.

I'm I missing something or is this a bug ?

Thank you,

Thomas

### Code to reproduce this issue

https://gist.github.com/glitch911/accfba844a938ffd9387c27b807dafc4

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7219809/sdkconfig.txt)

zhangyanjiaoesp, 2022-03-29:
Thanks for reporting, we will look into this issue.

zhangyanjiaoesp, 2022-04-20:
@glitch911 According to the feedback from my colleagues, RF calibration skipped after deep sleep wake up on ESP32C3 has not been implemented yet.

glitch911, 2022-04-20:
Thank you for feedback,
Can you confirm it will be in the future?
Any timeline?
Thank you

zhangyanjiaoesp, 2022-04-20:
It will be in the future, but the exact timeline is uncertain, we will feedback to you as soon as possible.

Alvin1Zhang, 2022-07-13:
Thanks for reporting, feel free to reopen.

closed 2022-07-13 by Alvin1Zhang

AxelLin, 2022-08-04:
> @glitch911 [44989a9](https://github.com/espressif/esp-idf/commit/44989a9ea0e21520b31a3f5e03b000d538ca0ab0) for master support this feature, you can have a try if you want. It will be backported to other branches later.

Just remind that the fix is not yet in v4.3, v4.4 branches.

AxelLin, 2022-09-20:
>  It will be backported to other branches later.

@jack0c After waiting for 3 months, I still cannot find the backport fix in stable branches. What's wrong.

jack0c, 2022-11-14:
@AxelLin Sorry, you can find it in 1adb0b6b943610bab8b8eca85f4aad7233fab3e3