espressif/esp-idf #7597: rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

matthuszagh, 2021-09-24:
I'm using the [ESP32-PICO-DevKitM-2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-pico-devkitm-2.html) to prototype applications for the esp32. In the process of building/flashing/monitoring applications, I've now twice (two different dev boards) encountered errors when attempting to monitor the running application. Once the error manifests itself, I've been unable to fix it. That is, all applications exhibit the same error and no simple fixes such as power cycling or waiting seem to fix the error.

The error I'm seeing is:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6716
ho 0 tail 12 room 4
load:0x07800000,len:3649583
ets Jul 29 2019 12:21:46
```

which continues indefinitely.

I've also tested the [hello world example application](https://github.com/espressif/esp-idf/tree/master/examples/get-started/hello_world) (previously worked fine), which fails with the exact same error.

The error first manifest from a different application that only used GPIOs 22, 19 and 21, and all device GPIOs were left floating and fully insulated from one another. It did not manifest immediately. Rather, it only manifest after I'd already run the same application many times on the device. The dev board is powered by a USB 3.2 connector on my laptop. I've probed this voltage and it's about 3.33V. I have not attempted to break the circuit to measure the current, but the USB port should be able to provide enough current for the device.

This seems like a hardware failure to me. Is that correct? Is there any way to recover from it? How can I avoid this in the future with the same board? Other thoughts/suggestions?

---

Related: https://github.com/espressif/arduino-esp32/issues/4278

zekoli123, 2021-09-24:
Your ESP kit appears in programming mode. Did you try to disconnect the connection between I0 and GND after the program?

matthuszagh, 2021-09-24:
> Your ESP kit appears in programming mode. Did you try to disconnect the connection between I0 and GND after the program?

@zekoli123 Can you clarify what you mean by this? If by IO you're referring to GPIO, I haven't made any connections to any of the dev board GPIOs.

If IO means the programmer, I can't remove the programmer since it's built into the dev board. I could probably remove the USB monitoring connection from the device after flashing, but would need to provide external power to do so since the device is currently powered through USB. Though I'm not sure I understand what this would do.

tuupola, 2021-09-25:
`I0` instead of `IO`. You can force ESP32 to go to download mode by `GPIO0` to `GND`.  I do not know if it is related but I have randomly had similar problems (https://github.com/espressif/esp-idf/issues/1857) with different dev boards. Erasing and flashing the boards couple of times usually fixed the problem. I haven't however been able to figure out the root cause.

KavehSons, 2021-11-26:
hi All, I have the same problem with my wroom 32 dev board , I've used a 2 A adaptors and have tried to select different combinations such as speed, flash speed, flash mode etc. however  can not run the sketch that uses wifi library such as basicsOTA example, while simple sketch such as toggle the LED is work fine 
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:12984
load:0x40080400,len:3648
entry 0x400805f8
ets Jun  8 2016 00:22:57

```

S-A-L-M-A-N-A-N-S-A-R-I, 2022-05-02:
Hi , i also suffered from same problem in my case i have led at gpio12 i removed that and problem is solved

ginkgm, 2022-08-01:
I guess you are not able to flash due to some reason, it always goes into flash boot mode.

This may be caused by some starpping are not set to correct level. You may check the IO state of IO0, IO2 and IO12 first. We will also debug on our side to see if there is issue when those pins are not used.

S-A-L-M-A-N-A-N-S-A-R-I, 2022-08-01:
The issue is solved the problem was boot button and reset button has the same capacitor these capacitor takes some time to charge up , when power on both boot and reset button is at low level till the capacitor is charging up
As both are at low level esp goes in download mode and then we cant upload the code 
Solution is remove capacitors connected with boot and reset button

ginkgm, 2022-08-01:
So you are using switches to control the VDD of [ESP32-PICO-DevKitM-2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-pico-devkitm-2.html), instead of the EN pin?

oyidi, 2023-08-15:
I have same issue with ESP32-PICO-D4, I make GPIO16 and GPIO17 empty and it works, seems GPIO16, GPIO17  connect some SPI (maybe it was the flash rom) device in the chip.

tharakapasan, 2024-01-17:
hi I'm using a custom PCB using ESP32-PICO-d4. GPIO16 and GPIO17 are left out.But i got logic high level in the GPIO16 is it normal when nothing is connected to it?.GPIO6 and 11 are also not connected they has logic low.  Also power is good solid 3.3v with more than 1.5A of current draw from the supply. I'm encountering the same problem.Im flashing using RX and TX pins of UART via a FTDI. 

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7080
load:0x40078000,len:15584
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c 

any thoughts/suggestions?