espressif/esp-idf #7598: boot: Disabling RNG early entropy source?

zekoli123, 2021-09-24:
Hello. I am using the ESP32S board. I've used this board 3-5 times and had no problems programming it before. I'm trying to use a new esp32s. I made all the connections. But I couldn't assign any program. I can't switch it to programming mode, I can't delete or update from the flash download program. As seen below. I can't fix it no matter what I did. Please help me

10:55:27.929 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:55:27.929 -> configsip: 0, SPIWP:0xee
10:55:27.929 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:55:27.929 -> mode:DIO, clock div:2
10:55:27.929 -> load:0x3fff0018,len:4
10:55:27.929 -> load:0x3fff001c,len:6916
10:55:27.929 -> load:0x40078000,len:14368
10:55:27.929 -> load:0x40080400,len:4248
10:55:27.929 -> entry 0x400806e0
10:55:27.975 -> [0;32mI (71) boot: Chip Revision: 1[0m
10:55:27.975 -> [0;32mI (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
10:55:27.975 -> [0;32mI (39) boot: ESP-IDF v4.0-beta2-174-g99fb9a3f7 2nd stage bootloader[0m
10:55:27.975 -> [0;32mI (39) boot: compile time 07:55:04[0m
10:55:27.975 -> [0;32mI (40) boot: Enabling RNG early entropy source...[0m
10:55:27.975 -> [0;32mI (46) boot: SPI Speed      : 40MHz[0m
10:55:27.975 -> [0;32mI (50) boot: SPI Mode       : DIO[0m
10:55:27.975 -> [0;32mI (54) boot: SPI Flash Size : 4MB[0m
10:55:27.975 -> [0;32mI (58) boot: Partition Table:[0m
10:55:27.975 -> [0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
10:55:28.020 -> [0;32mI (69) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
10:55:28.020 -> [0;32mI (76) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
10:55:28.020 -> [0;32mI (84) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
10:55:28.020 -> [0;32mI (91) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
10:55:28.020 -> [0;32mI (99) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
10:55:28.020 -> [0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
10:55:28.066 -> [0;32mI (114) boot: End of partition table[0m
10:55:28.066 -> [0;32mI (118) boot_comm: chip revision: 1, min. application chip revision: 0[0m
10:55:28.066 -> [0;32mI (125) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x28178 (164216) map[0m
10:55:28.111 -> [0;32mI (192) esp_image: segment 1: paddr=0x001281a0 vaddr=0x3ffbdb60 size=0x032ec ( 13036) load[0m
10:55:28.111 -> [0;32mI (198) esp_image: segment 2: paddr=0x0012b494 vaddr=0x40080000 size=0x00400 (  1024) load[0m
10:55:28.156 -> [0;32mI (200) esp_image: segment 3: paddr=0x0012b89c vaddr=0x40080400 size=0x04774 ( 18292) load[0m
10:55:28.156 -> [0;32mI (216) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x102acc (1059532) map[0m
10:55:28.520 -> [0;32mI (593) esp_image: segment 5: paddr=0x00232aec vaddr=0x40084b74 size=0x134c0 ( 79040) load[0m
10:55:28.566 -> [0;32mI (626) esp_image: segment 6: paddr=0x00245fb4 vaddr=0x400c0000 size=0x00064 (   100) load[0m
10:55:28.566 -> [0;32mI (641) boot: Loaded app from partition at offset 0x100000[0m
10:55:28.566 -> [0;32mI (641) boot: Disabling RNG early entropy source...[0m
10:55:28.746 -> 2.0.0

10:55:28.746 -> max tx power=78,ret=0

I have read the forms opened on this subject but have not found any answers.

As I said above, this is the part that says when I put it in programming mode:
11:01:55.822 -> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
11:01:55.822 -> waiting for download

It seems to go into programming mode but it doesn't throw any code.

ESP-Marius, 2022-08-31:
@zekoli123 Sorry for the late reply, seems like we missed this issue.

Are you still experiencing any issues with regards to this?

ESP-Marius, 2022-09-08:
Closing this one due to lack of activity, but feel free to reopen if you are still having issues.

closed 2022-09-08 by ESP-Marius