espressif/esp-idf #7599: Mesh time authority re-sends last time message

IvanPeredriy, 2021-09-24:
## Environment

- Module or chip used:  ESP32-WROOM-32S and TTGO T5 v2.3
- IDF version v4.4-dev-2487-g8131d6f46
- Build System:  idf.py
- Operating System:    Ubuntu 18.04
- Using an IDE?: CLion only for code edit, idf.py for build process.
- Power Supply:      USB | external

## Problem Description

Every publish event time server publish previous data and then publish actual data.  For example, it send:

1 package: publish time 20 seconds.
                   publish time 40 seconds.
~20 seconds delay~
2 package: publish time 40 seconds.
                   publish time 60 seconds.

### Expected Behavior

update time and send data once.

### Actual Behavior

2 data messages per package.

### Steps to reproduce

1. Set publish address in time server;
2. Set publish interval, retransmits = 0;
3. Watch wrong packages.

[image]

## Debug Logs

2021-09-24 15:19:32 I (87444) BLE_MESH: Publish period 20000 ms
2021-09-24 15:19:32 I (87445) BLE_MESH: Publish Retransmit Count 0 Interval 100ms
2021-09-24 15:19:32 I (87445) BLE_MESH: send, app_idx 0x0000 src 0x00b8 dst 0xc000
2021-09-24 15:19:32 I (87445) BLE_MESH: send, len 13: 5d4200000000cc000100000000   <------ THIS IS INCORRECT PACKAGE!
2021-09-24 15:19:32 I (87447) BLE_MESH: Settings store, waiting 0 seconds
2021-09-24 15:19:32 I (87457) BLE_MESH: Settings store, waiting 0 seconds
2021-09-24 15:19:32 I (87461) ble_mesh: esp_ble_mesh_model_callback 5
2021-09-24 15:19:32 I (87461) ble_mesh: esp_ble_mesh_model_callback: ESP_BLE_MESH_MODEL_PUBLISH_UPDATE_EVT
2021-09-24 15:19:32 
2021-09-24 15:19:32  Sending time EET-3: 1970-01-01 03:01:26
2021-09-24 15:19:32  TAI TIME 
2021-09-24 15:19:32  Time raw TAI:86
2021-09-24 15:19:32 I (87462) ble_mesh: process_publish_update_event: ESP_BLE_MESH_MODEL_ID_TIME_SRV0m
2021-09-24 15:19:32 I (87463) BLE_MESH: Publish Retransmit Count 0 Interval 100ms
2021-09-24 15:19:32 I (87464) BLE_MESH: send, app_idx 0x0000 src 0x00b8 dst 0xc000
2021-09-24 15:19:32 I (87464) BLE_MESH: send len 13: 5d5600000000ff000100000000  <----- THIS ONE IS CORRECT!
2021-09-24 15:19:32 I (87466) BLE_MESH: Settings store, waiting 0 seconds
2021-09-24 15:19:32 I (87470) BLE_MESH: Settings store, waiting 0 seconds
2021-09-24 15:19:32 I (87474) ble_mesh: esp_ble_mesh_model_callback 2
2021-09-24 15:19:32 I (87474) ble_mesh: esp_ble_mesh_model_callback: ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT, model 4608
2021-09-24 15:19:32 I (87474) ble_mesh: Publish complete, error coe is: 0
2021-09-24 15:19:32 I (87641) BLE_MESH: Complete tx sdu to group
2021-09-24 15:19:33 I (87641) BLE_MESH: Publishing took 184ms
2021-09-24 15:19:33 I (87641) BLE_MESH: Publishing next time in 19816ms
2021-09-24 15:19:33 I (87829) BLE_MESH: Complete tx sdu to group
2021-09-24 15:19:33 I (87830) BLE_MESH: Publishing took 185ms
2021-09-24 15:19:33 I (87830) BLE_MESH: Publishing next time in 19815ms

InfiniteYuan, 2021-09-26:
Sorry for the late reply.

I don't understood completed. Could you describe in detail your usage scenarios and the problems encountered during usage?

Alvin1Zhang, 2021-11-22:
@IvanPeredriy Thanks for reporting, would you please help share more details per https://github.com/espressif/esp-idf/issues/7599#issuecomment-927247610? Thanks.

closed 2022-03-08 by Alvin1Zhang