espressif/esp-idf #7601: IRAM0 doesnt fit, please correct the IDF.PY SIZE

filzek, 2021-09-24:
Latest SDK as of 24/09

please correct the instruction

IDF.PY size

so, even if it doesnt FIT in the IRAM0 region it must show all the tables with all information to try to solve this at once!

dobairoland, 2021-09-24:
@filzek Please specify what do you mean by "latest SDK". Please state the exact version of ESP-IDF as it is suggested by the issue template. This has been recently fixed. If you already have that change then we will need more information to reproduce it.

Alvin1Zhang, 2021-10-13:
@filzek Thanks for reporting, would you please help provide more information suggested in https://github.com/espressif/esp-idf/issues/7601#issuecomment-926885926? Thanks.