espressif/esp-idf #7603: Incorrect TWAI macros

kdschlosser, 2021-09-25:
## Environment

- Development Kit:
- Kit version
- Module or chip used:  ESP32S2
- IDF version 4.2
- Build System: MicroPython
- Compiler version 8.4.0
- Operating System:  Ubuntu (Linux)
- (Windows only) environment type: 
- Using an IDE?: No
- Power Supply:  USB

## Problem Description
In the header file `components\soc\include\hal\twai_types.h` the following macros are defined.
```c
#define TWAI_TIMING_CONFIG_1KBITS()     {.brp = 4000, .tseg_1 = 15, .tseg_2 = 8, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_5KBITS()     {.brp = 800, .tseg_1 = 15, .tseg_2 = 8, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_10KBITS()    {.brp = 400, .tseg_1 = 15, .tseg_2 = 8, .sjw = 3, .triple_sampling = false}
```

I believe there is an error in these macros and the `tseg_2` field should be a 4 and not an 8. I wrote a script that calculates the brp, tseg1, tseg2 matches all of the rest of the macros in the header file except for these 3. The script returns a 4 for the `tseg_2` field. All of the rest of the fields are correct... 

If for some reason the 66.6% sample point must stay then the bitrates for TWAI_TIMING_CONFIG_5KBITS and TWAI_TIMING_CONFIG_10KBITS cannot be used as there is no match. I am gathering that an 80% sample point is fine as most other macros are set to an 80% sample point. And if that is the case then a change of the `tseg_2` field for each of those would need to be a 4.

I do not have any kind of an example that shows they are incorrect other then the Python script that I wrote to calculate the needed parameters. I have attached the Python script so you can run it and also check if my calculations are correct if you like. I am pretty sure the tseg_2 is incorrect.

[can_timing.py.txt](https://github.com/espressif/esp-idf/files/7228985/can_timing.py.txt)

negativekelvin, 2021-09-25:
https://github.com/espressif/esp-idf/commit/d604bf05affcbd80e85afde17624bfdec8f0ff9d#diff-0e889d130b589e061b63dc49b8b37ab2df59e0287b8025c1343c7be707297460

kdschlosser, 2021-09-25:
Heyyyyy so I wasn't wrong.  It was caught earlier so this issue can be closed. 

Maybe someone might consider porting the attached script to C code and adding it to esp-idf. It has the values needed to calculate timing registers for a bunch of different interface IC's and it can be added onto really easily.

It makes it a whole lot easier to be able to plug in the bit rate you want along with a tolerance and have it give the timing registers needed to get to that bit rate. I even went to the extent of automatically calculating what the SJW/RJW should be set to using the transceiver processing delay and the bus physical length. It also takes into consideration oscillator differences and how much of a difference that would be tolerated for the provided bit rate, bus length and transceiver delay.

I also added in CIA compliant sample points for those that do not want to deal with them.

It would be a useful tool to have and maybe add it as a separate module/library called twai_tools or some such thing.

Dazza0, 2021-09-27:
@kdschlosser Auto bit timing calculation is in our future road map. So is hot-plugging bit rate detection. But it's planned for v5.0 because both features will require a change to the TWAI driver's API/programming model.

kdschlosser, 2021-09-27:
OK cool. I see there is a kind of add-on repository that has been made. Is that functional? is it public, meaning can I write something and have it added to it?

Bit rate detection is also good to have but isn't it slow?  I thought the only way to do it is the interface has to be set to listen only and then set the bit rate one at a time and check for errors. That is a time consuming process  and you have the issue of how busy the bus is. If it's only a couple of nodes and those nodes don't broadcast then the bit rate detection would fail.  If using an S2 there are 16384 prescalars and if 1/4 of those are unique bit rates you are talking 4096 different bit rates. If it was something that was built into the interface it's self that would be the better way to go about doing it.

I don't know how else an automatic bit rate feature could be done. 

I am glad I was correct about those macros being wrong. It means my calculations are working correctly.  Out of question what is the tolerance of the oscillator on the ESP32's? knowing this would help to select a proper sample point and SJW.

Another really nice feature would be interrupts for when a message gets received and also an interrupt if there is no traffic on the bus at all. The interface on the ESP32 seems to be directed at the automotive market and the modules in the vehicle go to sleep when there is no more data being passed on the bus and it wakes when it sees data again. I am pretty sure the wake and sleep re probably pointing to specific frame id's for this. If I put the ESP32 into deep sleep does the interface also get powered down?.. I see there is a bus-off interrupt, can this be used to put the ESP32 into deep sleep and can it be used to wake the ESP32??

I know I am straying from the issue topic figured it is easier to ask those questions here then to try and get a correct answer from the forum.

Dazza0, 2021-09-27:
@kdschlosser 

> Bit rate detection is also good to have but isn't it slow? I thought the only way to do it is the interface has to be set to listen only and then set the bit rate one at a time and check for errors.

Yes, it would be slow to scan every single bit rate. The more likely scenario would be to only scan the commonly used bit rates (e.g., 10K, 20K, 25K, 50K, 100K, ... up to 100M).

> If it's only a couple of nodes and those nodes don't broadcast then the bit rate detection would fail.
Yes that would be true. But it's not uncommon for the network layer protocol (such as CanOpen) to have a Heartbeat/keep-alive message that is transmitted periodically.

> I am glad I was correct about those macros being wrong. It means my calculations are working correctly. Out of question what is the tolerance of the oscillator on the ESP32's? knowing this would help to select a proper sample point and SJW.

The peripheral is run off the APB clock which usually operates at 80MHz. I'm not sure about the exact drift tolerance (more details in this [thread](https://www.esp32.com/viewtopic.php?t=16778)),  but I think its drift be too small to affect the sample point selection. The propagation delay of your bus (i..e, the length) will probably have a much larger effect on sample point selection.

> Another really nice feature would be interrupts for when a message gets received and also an interrupt if there is no traffic on the bus at all. The interface on the ESP32 seems to be directed at [...]

- It wouldn't be too use a timer that triggers an alert if no messages are received within a fixed period of time. For the ESP32S3, IIRC a bus idle interrupt was added so this should make things easier.
- I don't think you can put it into light/deep sleep as that will shut down the peripheral. Adding a light/deep sleep wake up from the TWAI peripheral will require a hardware change. Can't use DFS either as DFS will affect the APB clock of the peripheral (thus affect the bit timing).

Note that we've also had similar request regarding TWAI power saving (such as https://github.com/espressif/esp-idf/issues/5883), so we are open to some form of power saving to TWAI in future hardware.
- Adding an extra GPIO form the TWAI controller to automatically put the connected PHY to a lower power mode should be trivial
- But having the TWAI wake up the chip from deep sleep will be a non-trivial change as the TWAI controller needs to have some form of operation in the RTC domain.
- But if there's a significant user base that wants power saving, we are open to adding some form of TWAI power saving feature to HW/SW.

kdschlosser, 2021-09-28:
OK the issue you linked to has someone reply to a comment I made and that caused me to do some research. The MCP2562 has the ability to wake the MCU. YAY!!!!!

I will give more information on how to do this once I mess about with it.