espressif/esp-idf #7604: Can esp32c3 be used for PDM?

xpdrycry, 2021-09-25:
Can esp32c3 be used for PDM? We can’t get the output using SPH0644, don’t know why? The program runs normally, but there is no sound。。by using example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s/i2s_audio_recorder_sdcard,this example.

L-KAYA, 2021-09-26:
Sorry, ESP32-C3 only support PDM ouput, not support PDM input. This example should not not able to run on C3. 
Thanks for reporting this issue, we will add warning in driver and list the target chip in example `README.md` later.

xpdrycry, 2021-09-26:
Can s3 be used for PDM?

L-KAYA, 2021-09-26:
S3 have both PDM TX and RX, so it should be. But I tried to run this example on exp32s3 just now, and find it has no output. There should be something wrong with the example or driver. If any clues found, I'll post here.

L-KAYA, 2021-09-28:
This example will support S3 soon and it will also get some bugfix.

closed 2021-10-12 by igrr

dzanis, 2021-12-15:
Can esp32c3 be used for i2s adc mode input ?

4rthurmonteiro, 2024-07-25:
Yes .

> Can esp32c3 be used for i2s adc mode input ?