espressif/esp-idf #7606: esp_hid doesn't accept valid hid reports

sqrtroot, 2021-09-26:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.4-dev-3235-g3e370c4296
- Build System:        idf.py/cmake
- Compiler version 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: Yes (VSCode not part of the problem)
- Power Supply:         external 5V|

## Problem Description

Valid HID Descriptors are not parsed correctly by the esp_hid component.
The valid HID Descriptors come from https://usb.org/sites/default/files/hutrr39b_0.pdf but it's a more general problem where non application collections are not accepted https://github.com/espressif/esp-idf/blob/97e2465439eaadd7cbaec644b71919f6d56c17b2/components/esp_hid/src/esp_hid_common.c#L229 is where it goes wrong in the parser state machine

### Expected Behavior
HID Descriptors be parsed correctly

### Actual Behavior
Getting error `E (1133) hid_parser: expected APPLICATION, but got 0x00`

### Steps to reproduce

1. Copy esp_hid_device example
2. Add the hid descriptor helpers from the paper
3. Add one of the descriptors from the paper
4. Compile, run and see the error

boblane1, 2021-09-27:
Hi @sqrtroot,

Thanks for your reporting. According to the `Device Class Definition for HID 1.11`, application collection shall be the top level collection. So, I think the the parser state machine is right. How do you think?
[image]

boblane1, 2021-09-27:
Hi @sqrtroot,

Thanks for your reporting! According to the `Device Class Definition for HID 1.11`, application collection shall be the top level collection. So, I think the handle of the parser state machine is right. How do you think?

[image]

sqrtroot, 2021-09-27:
Aah, I think I know where the confusion started. Microsoft's drivers allow non application top-level collections https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/top-level-collections they even admit here that this deviates from the USB hid spec. Would be nice if we have a define that allows us to toggle to this behaviour, but I see now that's not a bug.

closed 2021-10-19 by Alvin1Zhang