espressif/esp-idf #7608: ESP IDF 4.3 PPPOS example crashes at hangup / changing from PPP mode to command mode

MartinPedersenpp, 2021-09-27:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used: ESP32-WROOM-32D
- IDF version V4.3
- Build System:         [VS code extension Cmake]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [Yes VS Code]
- Power Supply:         [USB|]

## Problem Description

I have been trying for the past few days to get the pppos example to run. After a lot of error when setting up (RX errors, wrong apn / auth) I finally got to the step where it works.
The example detects my sim800 module and connects to my provider using my sim card. It publishes and reads the commands from the MQTT, but when it comes to setting the modem back in command mode it fails and throws an error like this:
I (49811) esp-modem: In change mode
I (49821) esp-modem: Changing to ppp mode
E (49901) esp-modem: esp_dte_handle_line(113): handle line failed
W (49911) pppos_example: Unknow line received: ~!E
E (50081) esp-modem: esp_dte_handle_line(113): handle line failed
W (50081) pppos_example: Unknow line received: ~!E
E (50911) esp-modem: esp_dte_handle_line(113): handle line failed
W (50911) pppos_example: Unknow line received: ~!E
E (53141) esp-modem: esp_dte_handle_line(113): handle line failed
W (53151) pppos_example: Unknow line received: ~!E
I (55811) esp-modem: Changed mode succesfully
I (55811) esp-modem: Posted to event
I (55911) pppos_example: Modem PPP Stopped
I (55911) pppos_example: PPP state changed event 267
I (55911) pppos_example: PPP state changed event 265
I (55911) pppos_example: PPP state changed event 262
I (67911) pppos_example: PPP state changed event 268
I (67911) pppos_example: PPP state changed event 256
I (67911) esp-netif_lwip-ppp: User interrupt
I (67911) pppos_example: PPP state changed event 5
I (67921) pppos_example: User interrupted event from netif:0x3ffbd988
I (67921) esp-modem-netif: PPP state changed event 5
E (145811) esp-modem: esp_modem_dte_send_cmd(304): process command timeout
E (145811) dce_service: esp_modem_dce_hang_up(375): send command failed
E (145811) esp-modem: esp_modem_stop_ppp(615): hang up failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40084928
0x40084928: _esp_error_check_failed at C:/Users/praktikant/esp/esp-idf/components/esp_common/src/esp_err.c:41

file: "../main/pppos_client_main.c" line 310
func: app_main
expression: esp_modem_stop_ppp(dte)

I have been digging and located the handle line error happening in the esp_modem.c library at the function "esp_modem_dte_change_mode" and as the error reports the sending of the ATH command times out and fails the hang up.

It might just be me being stupid and missing something obvious, but I can't find a way to solve this issue. Any help is appreciated.

### Expected Behavior
It is expected that the modem succesfully changes into command mode to send the ATH command to the modem.
### Actual Behavior
Something goes wrong when changing from PPP mode to command mode and it causes the program to break and reboot the ESP32
### Steps to reproduce

1. Connect SIM800 module to ESP32 and run the 4.3 idf PPPOS client example
2. A failure is expected

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

https://github.com/MartinPedersenpp/ESP32PPPOS

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

The elf and sdkconfig is available in the repository

MartinPedersenpp, 2021-09-27:
Update:
Adding a 1sec delay in the esp_modem_dce_service.c hang_up function solved the timeout issue.
Should this be necessary or is it just the behavior of the "outdated" sim800module?
I hope for someone clever to answer this, but I can close the issue if requested.

chegewara, 2021-09-27:
I am using SIM800L and usually it didnt crash. 
You could change all `ESP_ERROR_CHECK` with `ESP_ERROR_CHECK_WITHOUT_ABORT`.

MartinPedersenpp, 2021-09-27:
> 
> 
> I am using SIM800L and usually it didnt crash.
> You could change all `ESP_ERROR_CHECK` with `ESP_ERROR_CHECK_WITHOUT_ABORT`.

Thanks, but wouldn't that just "ignore" the error causing the crash and make the program not work anyway?

chegewara, 2021-09-27:
For now i am using this approach, and if this will crash i will have more info why. I believe it is not critical error which does not have to abort code, but i may be wrong. An temporary i have to move to BG96, which finally is working for me with pppos.

MartinPedersenpp, 2021-09-28:
> 
> 
> For now i am using this approach, and if this will crash i will have more info why. I believe it is not critical error which does not have to abort code, but i may be wrong. An temporary i have to move to BG96, which finally is working for me with pppos, but code stays the same actually.

I am glad to hear it is working for you that way and as mentioned the added delay also made it work, I am just hoping for clarification about why this is necessary.

MartinPedersenpp, 2021-09-29:
Looks like I can just close this issues before I forget it.

closed 2021-09-29 by MartinPedersenpp

david-cermak, 2021-10-05:
Hi @MartinPedersenpp 

Thanks for the report! It seems like your problem might be an actual bug in IDF, which however never hits other devices, like `7600`, `SIM7000E`, BG96. This even used to work with my old `SIM800L` but couldn't connect to cellular network anymore so not able to test it.

The hang-up command fails after 90 seconds (quite a timeout), so the command itself was sent just after switch to command (`+++`) completed successfully, so assuming this

https://github.com/espressif/esp-idf/blob/e493a4c30e1b2f7c54a376327388b9e12262cbbf/examples/protocols/pppos_client/components/modem/src/sim800.c#L301

returned `ESP_OK` and simply continued to finish the `esp_modem_stop_ppp()` (no `Sending "+++" command failed` in your logs and the warnings indicate the PPP mode still in progress)

From [sim800l at reference](https://www.elecrow.com/wiki/images/2/20/SIM800_Series_AT_Command_Manual_V1.09.pdf):
```
2.2.12 +++ Switch from Data Mode or PPP Online Mode to Command Mode 
```
* No characters entered for T1 time (1 second)
* "+++" characters entered with no characters in between (1 second)
* No characters entered for T1 timer (1 second)

Switch to Command mode, otherwise go to step 1.

So it seems that we're missing the last point `no characters for 1 second` and send a hang-up request immediately. This apparently works for many devices (probably because we also wait for the `OK` reply, which arrives a bit later...)

This patch should work better than ignoring failures and adding delays (the same delay as you proposed [here](https://github.com/espressif/esp-idf/issues/7608#issuecomment-927656792), though)
```patch
diff --git a/examples/protocols/pppos_client/components/modem/src/sim800.c b/examples/protocols/pppos_client/components/modem/src/sim800.c
@@ -92,6 +92,7 @@ static esp_err_t sim800_set_working_mode(modem_dce_t *dce, modem_mode_t mode)
             DCE_CHECK(dce->state == MODEM_STATE_SUCCESS, "sync failed", err);
         } else {
             DCE_CHECK(dce->state == MODEM_STATE_SUCCESS, "enter command mode failed", err);
+            vTaskDelay(pdMS_TO_TICKS(1000)); // spec: 1s delay after `+++` command
         }
         ESP_LOGD(DCE_TAG, "enter command mode ok");
         dce->mode = MODEM_COMMAND_MODE;
```

vil-nakul, 2022-06-26:
I am still getting this error on my sim800 module. I enabled the verbose logs and saw that it gives error on "ATO" AT command(resume data command). I am using ap_to_pppos example. Please help.