espressif/esp-idf #7611: Build with MBEDTLS with CONFIG_MBEDTLS_DYNAMIC_BUFFER breaks compiler (Linker errors)

alorbach, 2021-09-28:
I am building esp-idf with esp32-arduino version 2.0.0 based on "esp32-arduino-lib-builder" toolchain - which works so far.
No I am trying to reduce memory usage as 2.0.0 sucked about 40-50kb more memory than 1.0.4 / 1.0.6. On way was trying to use 
**CONFIG_MBEDTLS_DYNAMIC_BUFFER** for mbedtls. When I do that I cannot link anymore, the build process fails with these linker errors: 

```
"C:\\code\\WiFiClientSecureUdp\\ssl_client_udp.cpp.o" "C:\\code\\_build\\m5alit\\core\\core.a" -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -llwip -llog -lheap -lsoc -lesp_hw_support -lxtensa -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lasio -lbt -lcbor -lunity -lcmock -lcoap -lconsole -lnghttp -lesp-tls -lesp_adc_cal -lesp_hid -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lesp_lcd -lprotobuf-c -lprotocomm -lmdns -lesp_local_ctrl -lsdmmc -lesp_serial_slave_link -lesp_websocket_client -lexpat -lwear_levelling -lfatfs -lfreemodbus -ljsmn -ljson -llibsodium -lmqtt -lopenssl -lperfmon -lspiffs -lulp -lwifi_provisioning -lbutton -ljson_parser -ljson_generator -lesp_schedule -lesp_rainmaker -lqrcode -lws2812_led -lesp-dsp -lesp32-camera -lesp_littlefs -lfb_gfx -lesp_ringbuf -lefuse -lesp_ipc -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -llwip -llog -lheap -lsoc -lesp_hw_support -lxtensa -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lasio -lbt -lcbor -lunity -lcmock -lunity -lcoap -lconsole -lnghttp -lesp-tls -lesp_adc_cal -lesp_hid -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lesp_lcd -lprotobuf-c -lprotocomm -lmdns -lesp_local_ctrl -lsdmmc -lesp_serial_slave_link -lesp_websocket_client -lexpat -lwear_levelling -lfatfs -lfreemodbus -ljsmn -ljson -llibsodium -lmqtt -lopenssl -lperfmon -lspiffs -lulp -lwifi_provisioning -lesp_adc_cal -lfatfs -lwear_levelling -lspiffs -lopenssl -lesp_hid -lesp_rainmaker -ljson_parser -ljson_generator -lwifi_provisioning -lprotocomm -lbt -lbtdm_app -lprotobuf-c -lmdns -lconsole -ljson -lmqtt -lesp_schedule -lqrcode -lesp_littlefs -lcat_face_detect -lhuman_face_detect -ldl -ldriver -lcxx -lnewlib -lfreertos -lesp_hw_support -lheap -llog -llwip -lsoc -lhal -lesp_rom -lesp_common -lesp_system -lbootloader_support -lspi_flash -lesp_pm -lesp_ringbuf -lefuse -lesp_timer -lesp_ipc -ldriver -lmbedtls -lmbedcrypto -lmbedx509 -lmbedtls -lapp_update -lnvs_flash -lpthread -lapp_trace -lesp_gdbstub -lespcoredump -lesp_phy -lesp_event -lesp_netif -lesp_eth -ltcpip_adapter -lwpa_supplicant -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lvfs -lesp_wifi -lxtensa -lulp -lesp_http_client -lesp_http_server -lesp_serial_slave_link -lesp-tls -lesp_https_ota -lnghttp -ltcp_transport -lsdmmc -ldriver -lcxx -lnewlib -lfreertos -lesp_hw_support -lheap -llog -llwip -lsoc -lhal -lesp_rom -lesp_common -lesp_system -lbootloader_support -lspi_flash -lesp_pm -lesp_ringbuf -lefuse -lesp_timer -lesp_ipc -ldriver -lmbedtls -lmbedcrypto -lmbedx509 -lmbedtls -lapp_update -lnvs_flash -lpthread -lapp_trace -lesp_gdbstub -lespcoredump -lesp_phy -lesp_event -lesp_netif -lesp_eth -ltcpip_adapter -lwpa_supplicant -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lvfs -lesp_wifi -lxtensa -lulp -lesp_http_client -lesp_http_server -lesp_serial_slave_link -lesp-tls -lesp_https_ota -lnghttp -ltcp_transport -lsdmmc -ldriver -lcxx -lnewlib -lfreertos -lesp_hw_support -lheap -llog -llwip -lsoc -lhal -lesp_rom -lesp_common -lesp_system -lbootloader_support -lspi_flash -lesp_pm -lesp_ringbuf -lefuse -lesp_timer -lesp_ipc -ldriver -lmbedtls -lmbedcrypto -lmbedx509 -lmbedtls -lapp_update -lnvs_flash -lpthread -lapp_trace -lesp_gdbstub -lespcoredump -lesp_phy -lesp_event -lesp_netif -lesp_eth -ltcpip_adapter -lwpa_supplicant -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lvfs -lesp_wifi -lxtensa -lulp -lesp_http_client -lesp_http_server -lesp_serial_slave_link -lesp-tls -lesp_https_ota -lnghttp -ltcp_transport -lsdmmc -ldriver -lcxx -lnewlib -lfreertos -lesp_hw_support -lheap -llog -llwip -lsoc -lhal -lesp_rom -lesp_common -lesp_system -lbootloader_support -lspi_flash -lesp_pm -lesp_ringbuf -lefuse -lesp_timer -lesp_ipc -ldriver -lmbedtls -lmbedcrypto -lmbedx509 -lmbedtls -lapp_update -lnvs_flash -lpthread -lapp_trace -lesp_gdbstub -lespcoredump -lesp_phy -lesp_event -lesp_netif -lesp_eth -ltcpip_adapter -lwpa_supplicant -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lvfs -lesp_wifi -lxtensa -lulp -lesp_http_client -lesp_http_server -lesp_serial_slave_link -lesp-tls -lesp_https_ota -lnghttp -ltcp_transport -lsdmmc -ldriver -lcxx -lnewlib -lfreertos -lesp_hw_support -lheap -llog -llwip -lsoc -lhal -lesp_rom -lesp_common -lesp_system -lbootloader_support -lspi_flash -lesp_pm -lesp_ringbuf -lefuse -lesp_timer -lesp_ipc -ldriver -lmbedtls -lmbedcrypto -lmbedx509 -lmbedtls -lapp_update -lnvs_flash -lpthread -lapp_trace -lesp_gdbstub -lespcoredump -lesp_phy -lesp_event -lesp_netif -lesp_eth -ltcpip_adapter -lwpa_supplicant -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lvfs -lesp_wifi -lxtensa -lulp -lesp_http_client -lesp_http_server -lesp_serial_slave_link -lesp-tls -lesp_https_ota -lnghttp -ltcp_transport -lsdmmc -lstdc++ -lpthread -lcxx -lm -lgcc -lnewlib -lapp_trace -lgcov -lapp_trace -lgcov -lc -lphy -lrtc -lesp_phy -lphy -lrtc -lesp_phy -lphy -lrtc -lxt_hal -Wl,--end-group -Wl,-EL -o "C:\\code\\_build\\m5alit/AmbiBack.Controller.ino.elf"
ERROR: C:/code/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\code\source\\libraries\WiFiClientSecure\ssl_client.cpp.o:(.literal._Z15stop_ssl_socketP17sslclient_contextPKcS2_S2_+0x0): undefined reference to `__wrap_mbedtls_ssl_free'
ERROR: C:/code/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\code\source\\libraries\WiFiClientSecure\ssl_client.cpp.o:(.literal._Z16start_ssl_clientP17sslclient_contextPKcjiS2_S2_S2_S2_S2_b+0x30): undefined reference to `__wrap_mbedtls_ssl_setup'
ERROR: C:/code/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\code\source\\libraries\WiFiClientSecure\ssl_client.cpp.o:(.literal._Z12data_to_readP17sslclient_context+0x0): undefined reference to `__wrap_mbedtls_ssl_read'
ERROR: C:/code/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\code\source\\libraries\WiFiClientSecure\ssl_client.cpp.o:(.literal._Z13send_ssl_dataP17sslclient_contextPKhj+0x0): undefined reference to `__wrap_mbedtls_ssl_write'
ERROR: C:/code/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\code\source\\libraries\WiFiClientSecure\ssl_client.cpp.o: in function `stop_ssl_socket(sslclient_context*, char const*, char const*, char const*)':
ERROR: C:\code\\buildenv\2.0.0.custom\packages\esp32\hardware\esp32\2.0.0\libraries\WiFiClientSecure\src/ssl_client.cpp:309: undefined reference to `__wrap_mbedtls_ssl_free'
ERROR: C:/code/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\code\source\\libraries\WiFiClientSecure\ssl_client.cpp.o: in function `start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, char const*, char const*, char const*, char const*, bool)':
ERROR: C:\code\\buildenv\2.0.0.custom\packages\esp32\hardware\esp32\2.0.0\libraries\WiFiClientSecure\src/ssl_client.cpp:225: undefined reference to `__wrap_mbedtls_ssl_setup'
ERROR: C:/code/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\code\source\\libraries\WiFiClientSecure\ssl_client.cpp.o: in function `data_to_read(sslclient_context*)':
ERROR: C:\code\\buildenv\2.0.0.custom\packages\esp32\hardware\esp32\2.0.0\libraries\WiFiClientSecure\src/ssl_client.cpp:276: undefined reference to `__wrap_mbedtls_ssl_read'
ERROR: C:/code/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\code\source\\libraries\WiFiClientSecure\ssl_client.cpp.o: in function `send_ssl_data(sslclient_context*, unsigned char const*, unsigned int)':
[... cut ...]
[95/1531] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/__/__/port/sha/parallel_engine/esp_sha512.c.obj
[96/1531] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/__/__/port/dynamic/esp_mbedtls_dynamic_impl.c.obj
[97/1531] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/ssl_ticket.c.obj
[98/1531] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/__/__/port/dynamic/esp_ssl_tls.c.obj
```

Help please

mahavirj, 2021-09-29:
@alorbach 

Can you please try to build IDF example https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_request with this feature once?

alorbach, 2021-09-29:
How do I do that? 
Till now I have always used esp32-arduino environment for building my code.

I have build my environment using https://github.com/espressif/esp32-arduino-lib-builder , so I can use idf utilities.

alorbach, 2021-09-29:
These are my mbedtls options
[image]

And here are BUILD logs, one with Stock 2.0.0 and one with CONFIG_MBEDTLS_DYNAMIC_BUFFER=Y which FAILS with linker errors

[HTTPS_OTA_Update.ino-2.0.0-dynamictxrx.log](https://github.com/espressif/esp-idf/files/7251218/HTTPS_OTA_Update.ino-2.0.0-dynamictxrx.log)
[HTTPS_OTA_Update.ino-2.0.0.log](https://github.com/espressif/esp-idf/files/7251219/HTTPS_OTA_Update.ino-2.0.0.log)