espressif/esp-idf #7613: Windows: special characters in build of idf.py are not displayed correctly in the last stage of the build

georgik, 2021-09-29:
## Environment

- Development Kit: none
- Module or chip used:  ESP32
- IDF version : v4.4-dev-2883-gb1f851b8f8
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: No
- Power Supply:   USB

## Problem Description

The last step of build log contains garbled characters in build path. Other log entries contains correct path displayed on the terminal.

### Expected Behavior

The build log displayed on the screen contains path displayed correctly in all log entries e.g.:
```
...
[96/96] C:\Users\Ján\Desktop\esp-idf\examples\get-started\blink\build\bootloader...
...
```

### Actual Behavior

The last build step contains garbled characters `��` in the path:
```
[1001/1001] cmd.exe /C "cd /D C:\Users\J��n\De...ktop/esp-idf/examples/get-started/blink/build/blink.bin"
blink.bin binary size 0x2b160 bytes. Smallest app partition is 0x100000 bytes. 0xd4ea0 bytes (83%) free.
```

### Steps to reproduce

* Install ESP-IDF on path with special character. e.g. C:\Users\Ján\Desktop\esp-idf
* Run the build: `cd "C:\Users\Ján\Desktop\esp-idf\examples\get-started\blink"; idf.py build` 
* Observe the log

zikalino, 2023-05-31:
This problem occurs due to how CMake's **execute_process()** is handling stdout and stderr from Windows process.
Most likely it redirects output to file which subsequently reads. In such case default coding is set to CP1251 instead of UTF-8.

To change default encoding use:

    chcp 65001
    set PYTHONIOENCODING=utf-8

zikalino, 2023-11-28:
This issue is related to 3rd party tools we are using (ninja). I am closing it as it has low impact.

closed 2023-11-28 by zikalino