espressif/esp-idf #7615: UART_DMA

99003662, 2021-09-29:
Hi team, we are working on UART_DMA . we need support on this 
1. Do we have example for UART_DMA in ESP-IDF ?
2. We have eof interrupt (end of frame) clearly mentioned in Technical reference .But do we have inter char interrupt(byte by byte)
[image]
 ?

99003662, 2021-10-01:
Thanks for the reply , you clearly mentioned that no to use DMA with ESP32 fine. I have some other doubts also related to ESP-idf
1. Does the IDF-MODBUS RTU example is works fine , 
2. In MODBUS slave.c file , where exactly interrupt is getting triggered and how can we test that file with external tool (like MODBUS POLL, which is act as master ).

alisitsyn, 2021-10-08:
See the answers [here](https://github.com/espressif/esp-idf/issues/7649)