espressif/esp-idf #7617: Can't set GPIO17 HIGH during deep sleep.. unexpected behaviour when forced.

GajosTheFajos, 2021-09-29:
- Development Kit:      [none]
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):  v4.3.1
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Windows
- Using an IDE?: Yes (VSCODE)
- Power Supply:        external 3.3V

## Problem Description

Trying to keep GPIO 17 on the module high while in deep_sleep. Tried all the possible combinations of the deep sleep functions.
Setting the RTC_CNTL_DG_PAD_FORCE_HOLD bit makes the gpio stay high (EXAMPLE 2) but if i try to unhold it (i cant have all the pins locked when restarted) it makes the pin not hold high when in deep sleep (EXAMPLE 3). on top of that i managed to get various watchdog resets out of deep sleep when using this code in more complicated scenarios (EXAMPLE 4). migrated from 4.2 hoping this would work but no joy.

What am i missing here? Please help

### Code to reproduce this issue

```cpp

//EXAMPLE 1 
// no panic but no result as well
void app_main(void)
{
    gpio_deep_sleep_hold_dis();
    esp_sleep_enable_timer_wakeup(1000ul*5000);
    gpio_reset_pin(17);
    gpio_set_direction(17,GPIO_MODE_OUTPUT);
    gpio_set_level(17,1);
    vTaskDelay(500 / portTICK_PERIOD_MS);
    gpio_sleep_set_direction(17,GPIO_MODE_OUTPUT);
    gpio_hold_en(17);
    gpio_deep_sleep_hold_en();
    esp_deep_sleep_start(); 
}

//EXAMPLE 2
// all the pins hold forever (no uart post)
void app_main(void)
{
    gpio_deep_sleep_hold_dis();
    esp_sleep_enable_timer_wakeup(1000ul * 5000);
    gpio_reset_pin(17);
    gpio_set_direction(17, GPIO_MODE_OUTPUT);
    gpio_set_level(17, 1);
    vTaskDelay(500 / portTICK_PERIOD_MS);
    gpio_sleep_set_direction(17, GPIO_MODE_OUTPUT);
    gpio_hold_en(17);
    REG_SET_BIT(RTC_CNTL_DIG_ISO_REG, RTC_CNTL_DG_PAD_FORCE_HOLD);
    esp_deep_sleep_start();
}

//EXAMPLE 3
// pin 17 doesn't stay high
void app_main(void)
{
    gpio_deep_sleep_hold_dis();
    REG_SET_BIT(RTC_CNTL_DIG_ISO_REG, RTC_CNTL_DG_PAD_FORCE_UNHOLD);
    esp_sleep_enable_timer_wakeup(1000ul * 5000);
    gpio_reset_pin(17);
    gpio_set_direction(17, GPIO_MODE_OUTPUT);
    gpio_set_level(17, 1);
    vTaskDelay(500 / portTICK_PERIOD_MS);
    gpio_sleep_set_direction(17, GPIO_MODE_OUTPUT);
    gpio_hold_en(17);
    REG_SET_BIT(RTC_CNTL_DIG_ISO_REG, RTC_CNTL_DG_PAD_FORCE_HOLD);
    esp_deep_sleep_start();
}
//EXAMPLE 4
// getting various watchdog and system resets out of deepsleep but pin manages to stay high
void app_main(void)
{
    static portMUX_TYPE sleep_spinlock = portMUX_INITIALIZER_UNLOCKED;
    gpio_deep_sleep_hold_dis();
    REG_SET_BIT(RTC_CNTL_DIG_ISO_REG, RTC_CNTL_DG_PAD_FORCE_UNHOLD);
    esp_sleep_enable_timer_wakeup(1000ul * 5000);
    gpio_reset_pin(17);
    gpio_set_direction(17, GPIO_MODE_OUTPUT);
    gpio_set_level(17, 1);
    vTaskDelay(500 / portTICK_PERIOD_MS);
    gpio_sleep_set_direction(17, GPIO_MODE_OUTPUT);
    gpio_hold_en(17);
    portENTER_CRITICAL(&sleep_spinlock);
    REG_CLR_BIT(RTC_CNTL_DIG_ISO_REG,RTC_CNTL_DG_PAD_FORCE_UNHOLD);
    REG_SET_BIT(RTC_CNTL_DIG_ISO_REG, RTC_CNTL_DG_PAD_FORCE_HOLD);
    portEXIT_CRITICAL(&sleep_spinlock);
    esp_deep_sleep_start();
}

//this is the log for example 4
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7028
load:0x40078000,len:14308
load:0x40080400,len:3708
0x40080400: _init at ??:?

entry 0x40080680
I (27) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (27) boot: compile time 15:12:01
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=07e58h ( 32344) map
I (122) esp_image: segment 1: paddr=00017e80 vaddr=3ffb0000 size=02954h ( 10580) load
I (126) esp_image: segment 2: paddr=0001a7dc vaddr=40080000 size=0583ch ( 22588) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=151cch ( 86476) map
I (170) esp_image: segment 4: paddr=000351f4 vaddr=4008583c size=05ec0h ( 24256) load
I (180) esp_image: segment 5: paddr=0003b0bc vaddr=400c0000 size=00064h (   100) load
I (180) esp_image: segment 6: paddr=0003b128 vaddr=50000000 size=00010h (    16) load
I (192) boot: Loaded app from partition at offset 0x10000
I (192) boot: Disabling RNG early entropy source...
I (209) cpu_start: Pro cpu up.
I (209) cpu_start: Starting app cpu, entry point is 0x400810e8
0x400810e8: call_start_cpu1 at C:/Users/Dariusz/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (224) cpu_start: Pro cpu start user code
I (224) cpu_start: cpu freq: 80000000
I (224) cpu_start: Application information:
I (228) cpu_start: Project name:     BATTERY_GAUGE
I (233) cpu_start: App version:      228ed77-dirty
I (239) cpu_start: Compile time:     Sep 29 2021 15:11:47
I (245) cpu_start: ELF file SHA256:  34ed756149deb357...
I (251) cpu_start: ESP-IDF:          v4.3.1-dirty
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (270) heap_init: At 3FFB3218 len 0002CDE8 (179 KiB): DRAM
I (276) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (282) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (289) heap_init: At 4008B6FC len 00014904 (82 KiB): IRAM
I (296) spi_flash: detected chip: generic
I (300) spi_flash: flash io: dio
I (305) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (320) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7028
load:0x40078000,len:14308
load:0x40080400,len:3708
0x40080400: _init at ??:?

entry 0x40080680
I (54) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (54) boot: compile time 15:12:01
I (54) boot: chip revision: 1
I (58) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (65) boot.esp32: SPI Speed      : 40MHz
I (70) boot.esp32: SPI Mode       : DIO
I (74) boot.esp32: SPI Flash Size : 4MB
W (79) boot.esp32: PRO CPU has been reset by WDT.
W (84) boot.esp32: WDT reset info: PRO CPU PC=0x400803c0
0x400803c0: _DoubleExceptionVector at C:/Users/Dariusz/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:455

W (90) boot.esp32: WDT reset info: APP CPU PC=0x400896a1
0x400896a1: wdt_hal_write_protect_enable at C:/Users/Dariusz/esp/esp-idf/components/hal/wdt_hal_iram.c:138

```

negativekelvin, 2021-09-29:
https://github.com/espressif/esp-idf/issues/6422

GajosTheFajos, 2021-09-29:
> 
> 
> #6422

Thank you for this.
1. Can we keep the domain powered up somehow ?
2. if the domain is unpowered why can't you pull it up externally. I tried adding external resistor (4.7k) and i remember it only managed to climb to  ~2V.
3. Any workarounds to keep it high ?

GajosTheFajos, 2021-09-30:
> 
> 
> To anwer my own questions..
> 
> esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_ON);
> 
> will see if that works tomorrow and close the issue if sorted.

Unfortunately this doesnt work..