espressif/esp-idf #7620: Not getting esp client disconnection event at ESP SoftAP

jigneshpatel47, 2021-09-30:
Hi,

I have used the default [SoftAp](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/softAP/main/softap_example_main.c) and [station](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c) example from ESP-IDF v4.3.1 on ESP32C3.
We have configured one ESP as SoftAp and the other as a client. The client can successfully be connected to SoftAP and also get the event of connection.

My issue is that: when the ESP client is disconnect from ESP SoftAp, It doesn't give a disconnect event in SoftAp.

Note: ESP SoftAP is giving disconnection event on mobile/laptop disconnection, It doesn't give disconnection event on ESP32C3.

Thank you,
Jignesh Patel

jigneshpatel47, 2021-10-09:
Thank you for reply @ESP-YJM, 

Yes, I am already doing this. 
But my device is portable battery-operated, so in case of removing the battery and also moving the device out of network coverage of AP, I want notification as soon as possible in AP.

Is there any way to get this?

Best Regards,
Jignesh Patel

ESP-YJM, 2021-10-09:
@jigneshpatel47 I think if station not send wifi disassociate packet to softap, softap itself need a mechanism to detect the station disconnect and send event to upper level. I’m not an expert in wifi, so I don’t know if there is such a mechanism. I can seek help for my wifi colleague and reply you as soon as possible.