espressif/esp-idf #7621: Generating x509_crt_bundle

Typoception, 2021-09-30:
Hi. This issue keeps popping up. I'm no sure how to fix it. I've just installed VSCode and ESP-IDF extension. Tried the project examples (blink) and this appears. 

Issue:
Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle 
cmd.exe /C "cd /D C:\Users\yahsa\Desktop\NTU\FYP\ESP_Workspace\blink\build\esp-idf\mbedtls && C:\Users\yahsa\esp\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:/Users/yahsa/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input C:/Users/yahsa/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q"
gen_crt_bundle.py: Invalid certificate in C:/Users/yahsa/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
ninja: build stopped: subcommand failed.

Anyone can guide me on how to solve this?

AdityaHPatwardhan, 2021-09-30:
Hi @Typoception Thank you for the issue. The issue is because one of the certificates in the [cacrt_all.pem](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/esp_crt_bundle/cacrt_all.pem) file has been expired recently ( 30 sep ). That is causing the failure in build.
Can you please disable the following option and try again if it works.
`(Top) > Component config > mbedTLS > Certificate Bundle->Enable trusted root certificate bundle`

Typoception, 2021-09-30:
@AdityaHPatwardhan how do i do that? I'm new to vscode. cant seem to find the option.
[image]

Typoception, 2021-09-30:
@AdityaHPatwardhan found it. thanks. finally works.

ghada-sellami, 2021-09-30:
> Hi @Typoception Thank you for the issue. The issue is because one of the certificates in the [...]

could you please tell me how to do this? I'm building on terminal on Linux

AStripsTR, 2021-09-30:
@AdityaHPatwardhan Thank you a lot!

@ghada-sellami in your sdkconfig file you can comment these lines : 
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=y
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=y

closed 2021-09-30 by Typoception

italocjs, 2021-09-30:
> @AdityaHPatwardhan found it. thanks. finally works.

Where is it?

italocjs, 2021-09-30:
> > @AdityaHPatwardhan found it. thanks. finally works.
> 
> Where is it?
If anyone is blind like me, it can be found here :)

[image]

felmue, 2021-10-01:
Hello guys

I also thought the issue stems from the recently expired Let's Encrypt root certificate (DST Root CA X3), but removing it from `cacrt_all.pem` did not solve the issue for me.

Doing some debugging I found that the EC-ACC certificate (which has not yet expired) seem to be the culprit. Removing that certificate from `cacrt_all.pem` allows me to compile successfully again.

I have not dug further so I don't know exactly what is wrong with that certificate.

BTW: there are multiple certificates inside `cacrt_all.pem` which have expired.

Thanks
Felix

felmue, 2021-10-02:
Hi guys

the issue seems to be that the EC-ACC certificate is using a negative serial number...

[7631](https://github.com/espressif/esp-idf/issues/7631)

Thanks
Felix

DwangDNguyen, 2021-10-15:
> Hi @DwangDNguyen , can you try adding `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=` line in your project's `sdkconfig.defaults` or `sdkconfig` file. That will disable the `esp_crt_bundle` by default. Just to note here- The [fix](https://github.com/espressif/esp-idf/commit/4e45f13e2df72a4cb4dc875942e95775198db85c) is already part of the `release/v4.2` branch of the esp-idf. You may just update your `esp-idf` and it should be fixed. Thanks

Thanks

askdesigners, 2021-10-19:
For anyone else getting stuck on this, you need to disable the option in the UI menu if you're in VSCode, as the `sdkconfig` file gets written automatically on build and will nuke your changes. 
Also, I just installed this all from scratch today and hit the same error, so I don't think it's fixed in v4.2

olsonap, 2021-11-01:
What worked for me was:
idf.py menuconfig
Component config --->
mbedTLS --->
Certificate Bundle ---> (directly below [ ] Enable mbedTLS debudding)
Uncheck [ ] Enable trusted root certificate bundle

askdesigners, 2021-11-07:
Yes this issue still persists in the latest version, but following the directions above to disable the cert works to get builds working again.

AdityaHPatwardhan, 2021-11-08:
Hi @askdesigners, Can you please mention which branch and which commit id you are using, thanks.

rtheil, 2021-11-09:
This issue still exists in 4.3.1. Brand new install. None of the above suggestions work.

AdityaHPatwardhan, 2021-11-10:
Yes, the release `v4.3.1` was released on `9 sep` and the issue came around 30 sep ( after a newer version of cryptography module was released). The fix should be a part of the `v4.3.2` release.
To fix the problem:
* I would recommend you to checkout to the `release/v4.3` branch which contains the respective fix. 

If that is not possible then as a temporary workaround you can do one of the following steps. 
( Please note that below option are temporary and is not recommended for any production env)
Temp Workaround:
1) You can revert back to earlier version of cryptography (<35) for the time being, refer [this](https://github.com/espressif/esp-idf/commit/9944c8acbe4f7f5bb9d5748387d8618fbfd20f6e) change.
2) You can remove the `EC-ACC` manually from `esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem`.
you can refer [this](https://github.com/espressif/esp-idf/commit/4e45f13e2df72a4cb4dc875942e95775198db85c#diff-77db583030591ad0d11e4f737094569581117b228d50511c8b0437d4e42ff431) change.

askdesigners, 2021-11-10:
@AdityaHPatwardhan TBH I'm not really sure as I just installed it all according to the tutorial related to the VSCode extension.

AdityaHPatwardhan, 2021-11-10:
@askdesigners No issues, I think you are using v4.3.1 as it is the latest release. Please refer to my earlier [comment](https://github.com/espressif/esp-idf/issues/7621#issuecomment-964777267) for possible solutions.
Thanks

rtheil, 2021-11-10:
Thanks for the suggestions.

> * I would recommend you to checkout to the `release/v4.3` branch which contains the respective fix.

I've gone through the automated setup in VS Code, and while I attempted to force check out release/v4.3, that results in an error when building
[image]

>You can revert back to earlier version of cryptography (<35) for the time being, refer this change.

Reverting to cryptography <35 still results in the error, and the compiler seems to continue, but in the end it does not generate the blink.bin file as expected.
[image]

>You can remove the EC-ACC manually from esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem.
you can refer this change.

After making this change, the project does build, so it seems like you must do both, not just one.

>( Please note that below option are temporary and is not recommended for any production env)

I am working on projects that will go into production, so building in this way isn't acceptable. What am I to do until a fix is in place?

AdityaHPatwardhan, 2021-11-10:
Hi @rtheil 
While on `release/v4.3`
1) Please make sure that you have removed `build`, `sdkconfig`, `sdkconfig.old` from the `blink` example folder.
2) After checking out to the `release/v4.3` branch, please ensure that you have updated all the submodules ( There must be some option in VS code to do that).

To remove the last error that you mentioned.
Can you please run the necessary `install.bat` command to install required dependancies, the part of the fix is to lock the `cryptography` version.

If you are still facing some issue after this then as a last resort you can do following.

* perform a clean install of the latest idf version. That should be `v4.3.1` ( which still contains the bug ).

a)If you are not using `crt_bundle` in your project, then please follow this [comment](https://github.com/espressif/esp-idf/issues/7621#issuecomment-956501988) to disable `crt_bundle` for your project. That should fix the issue.
Alternatively,
b) Follow [this](https://github.com/espressif/esp-idf/issues/7621#issuecomment-950135638) comment to remove the invalid certificate.

I hope this helps, thanks

paullaffitte, 2021-11-15:
Why is this issue closed? It's still not fixed.

rtheil, 2021-11-16:
> Why is this issue closed? It's still not fixed.

I believe it's referenced here: https://github.com/UncleRus/esp-idf-lib/issues/251