espressif/esp-idf #7622: Generating binary image from built executable

Typoception, 2021-09-30:
Hi, anyone knows how to solve this or what i did wrong?

Issue:
Generating binary image from built executable
FAILED: .bin_timestamp 
cmd.exe /C "cd /D C:\Users\yahsa\espdrone-nh\build && C:\Users\yahsa\esp\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:/Users/yahsa/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB --elf-sha256-offset 0xb0 -o C:/Users/yahsa/espdrone-nh/build/ESPDrone.bin C:/Users/yahsa/espdrone-nh/build/ESPDrone.elf && C:\Users\yahsa\esp\.espressif\tools\cmake\3.16.4\bin\cmake.exe -E echo "Generated C:/Users/yahsa/espdrone-nh/build/ESPDrone.bin" && C:\Users\yahsa\esp\.espressif\tools\cmake\3.16.4\bin\cmake.exe -E md5sum C:/Users/yahsa/espdrone-nh/build/ESPDrone.bin > C:/Users/yahsa/espdrone-nh/build/.bin_timestamp"
esptool.py v3.1-dev
Merged 1 ELF section

A fatal error occurred: Segment loaded at 0x3f4019e0 lands in same 64KB flash mapping as segment loaded at 0x3f400020. Can't generate binary. Suggest changing linker script or ELF to merge sections.
ninja: build stopped: subcommand failed.

laukik-hase, 2021-10-01:
Could you take a look at https://github.com/espressif/esp-idf/issues/1425 and https://www.esp32.com/viewtopic.php?t=435 ?

Alvin1Zhang, 2021-11-22:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details.

closed 2021-11-22 by Alvin1Zhang

zackees, 2024-09-02:
I work on the FastLED library and support a lot of ESP devices. What we've figured out is that when there are two data symbols with different linkage, for example one is const and the other is not, and both these same symbols end up in the same segment then this error is generated.

For example this is what we use in our source code to work around this:

```
// Implementation for a unique custom section
//
// This prevents gcc producing "x causes a section type conflict with y"
// errors if two variables in the same source file have different linkage (maybe const & non-const) but are placed in the same custom section
//
// Using unique sections also means --gc-sections can remove unused
// data with a custom section type set
#ifndef CONFIG_IDF_TARGET_LINUX
#define _SECTION_ATTR_IMPL(SECTION, COUNTER) __attribute__((section(SECTION "." _COUNTER_STRINGIFY(COUNTER))))
#define _COUNTER_STRINGIFY(COUNTER) #COUNTER
#else
// Custom section attributes are generally not used in the port files for Linux target, but may be found
// in the common header files. Don't declare custom sections in that case.
#define _SECTION_ATTR_IMPL(SECTION, COUNTER)
#endif
```

And then do things like this:

```
// Forces code into IRAM instead of flash
#define IRAM_ATTR _SECTION_ATTR_IMPL(".iram1", __COUNTER__)

// Forces data into DRAM instead of flash
#define DRAM_ATTR _SECTION_ATTR_IMPL(".dram1", __COUNTER__)
```

Esp32 core devs are aware of this and this is actually their code that we used in our project.

zackees, 2024-09-02:
To add more to this: espressif has a tendency to use const char *TAG = "module-tag"

My current theory is that when porting to C++ some library (ours was led_strip) that multiple TAGs get included in the linker and this causes a library segment error.