espressif/esp-idf #7625: lock_acquire_generic: Recursive mutexes make no sense in ISR context

baldhead69, 2021-09-30:
Hi,

I'm having a similar problem( "https://github.com/espressif/esp-idf/issues/7212" ) .

I  increased the "httpd stack_size" from 10240 to 20480 and the problem persists. 
```c
httpd_ssl_config_t config = HTTPD_SSL_CONFIG_DEFAULT(); 
config.httpd.stack_size = 20480;
```
idf version: v4.4-dev-2359-g58022f859

```c
abort() was called at PC 0x40085a07 on core 1
0x40085a07: lock_acquire_generic at C:/esp-idf/components/newlib/locks.c:139

Backtrace:0x400d3a3e:0x3fff03100x4008eb31:0x3fff0330 0x400967fa:0x3fff0350 0x40085a07:0x3fff03c0 0x40085b65:0x3fff03f0 0x40085c01:0x3fff0410 0x40170241:0x3fff0440 0x4016babf:0x3fff0700 0x4016ba3a:0x3fff0750 0x4009484a:0x3fff0780 0x400951c2:0x3fff07a0 0x4008218a:0x3fff07c0 0x40096825:0x3fff07e0 0x4011a3a1:0x3fff0800 0x4011adaa:0x3fff0820 0x4011a385:0x3fff0840 0x400ebf54:0x3fff0860 0x4011c2c1:0x3fff0880 0x4011c305:0x3fff08a0 0x4011c3f4:0x3fff08c0 0x4011c636:0x3fff08e0 0x4011ae95:0x3fff0900 0x4011b062:0x3fff0940 0x4011b0e6:0x3fff0990 0x40092b8e:0x3fff09b0
0x400d3a3e: panic_abort at C:/esp-idf/components/esp_system/panic.c:391

0x4008eb31: esp_system_abort at C:/esp-idf/components/esp_system/esp_system.c:129

0x400967fa: abort at C:/esp-idf/components/newlib/abort.c:46

0x40085a07: lock_acquire_generic at C:/esp-idf/components/newlib/locks.c:139

0x40085b65: _lock_acquire_recursive at C:/esp-idf/components/newlib/locks.c:167

0x4011c305: httpd_sess_clear_ctx at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:251

0x4011c3f4: httpd_sess_delete at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:365

0x4011c636: httpd_sess_close at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:145

0x4011ae95: httpd_process_ctrl_msg at C:/esp-idf/components/esp_http_server/src/httpd_main.c:150 (discriminator 15)

0x4011b062: httpd_server at C:/esp-idf/components/esp_http_server/src/httpd_main.c:215 (discriminator 15)

0x4011b0e6: httpd_thread at C:/esp-idf/components/esp_http_server/src/httpd_main.c:250 (discriminator 15)

0x40092b8e: vPortTaskWrapper at C:/esp-idf/components/freertos/port/xtensa/port.c:168
```

```c
abort() was called at PC 0x40085a07 on core 0
0x40085a07: lock_acquire_generic at C:/esp-idf/components/newlib/locks.c:139

Backtrace:0x400d3a3e:0x3fff03100x4008eb31:0x3fff0330 0x400967fa:0x3fff0350 0x40085a07:0x3fff03c0 0x40085b65:0x3fff03f0 0x40085c01:0x3fff0410 0x40170241:0x3fff0440 0x4016babf:0x3fff0700 0x4016ba3a:0x3fff0750 0x4009484a:0x3fff0780 0x400951c2:0x3fff07a0 0x4008218a:0x3fff07c0 0x40096825:0x3fff07e0 0x4011a3a1:0x3fff0800 0x4011adaa:0x3fff0820 0x4011a385:0x3fff0840 0x400ebf54:0x3fff0860 0x4011c2c1:0x3fff0880 0x4011c305:0x3fff08a0 0x4011c3f4:0x3fff08c0 0x4011c636:0x3fff08e0 0x4011ae95:0x3fff0900 0x4011b062:0x3fff0940 0x4011b0e6:0x3fff0990 0x40092b8e:0x3fff09b0
0x400d3a3e: panic_abort at C:/esp-idf/components/esp_system/panic.c:391

0x4008eb31: esp_system_abort at C:/esp-idf/components/esp_system/esp_system.c:129

0x400967fa: abort at C:/esp-idf/components/newlib/abort.c:46

0x40085a07: lock_acquire_generic at C:/esp-idf/components/newlib/locks.c:139

0x40085b65: _lock_acquire_recursive at C:/esp-idf/components/newlib/locks.c:167

0x40085c01: __retarget_lock_acquire_recursive at C:/esp-idf/components/newlib/locks.c:323

0x40170241: _vfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x4016babf: fiprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fiprintf.c:48

0x4016ba3a: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:58 (discriminator 8)

0x4009484a: tlsf_free at C:/esp-idf/components/heap/heap_tlsf.c:866 (discriminator 1)

0x400951c2: multi_heap_free_impl at C:/esp-idf/components/heap/multi_heap.c:220

0x4008218a: heap_caps_free at C:/esp-idf/components/heap/heap_caps.c:305

0x40096825: free at C:/esp-idf/components/newlib/heap.c:46

0x4011a3a1: esp_tls_internal_event_tracker_destroy at C:/esp-idf/components/esp-tls/esp_tls_error_capture.c:46

0x4011adaa: esp_mbedtls_server_session_delete at C:/esp-idf/components/esp-tls/esp_tls_mbedtls.c:629

0x4011a385: esp_tls_server_session_delete at C:/esp-idf/components/esp-tls/esp_tls.c:584

0x400ebf54: httpd_ssl_close at C:/esp-idf/components/esp_https_server/src/https_server.c:28

0x4011c2c1: httpd_sess_free_ctx at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:230

0x4011c305: httpd_sess_clear_ctx at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:251

0x4011c3f4: httpd_sess_delete at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:365

0x4011c636: httpd_sess_close at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:145

0x4011ae95: httpd_process_ctrl_msg at C:/esp-idf/components/esp_http_server/src/httpd_main.c:150 (discriminator 15)

0x4011b062: httpd_server at C:/esp-idf/components/esp_http_server/src/httpd_main.c:215 (discriminator 15)

0x4011b0e6: httpd_thread at C:/esp-idf/components/esp_http_server/src/httpd_main.c:250 (discriminator 15)

0x40092b8e: vPortTaskWrapper at C:/esp-idf/components/freertos/port/xtensa/port.c:168
```
```c
abort() was called at PC 0x40085a07 on core 1
0x40085a07: lock_acquire_generic at C:/esp-idf/components/newlib/locks.c:139

Backtrace:0x400d3a3e:0x3fff03300x4008eb31:0x3fff0350 0x400967fa:0x3fff0370 0x40085a07:0x3fff03e0 0x40085b65:0x3fff0410 0x40085c01:0x3fff0430 0x40170231:0x3fff0460 0x4016baaf:0x3fff0720 0x4016ba2a:0x3fff0770 0x4009484a:0x3fff07a0 0x400951c2:0x3fff07c0 0x4008218a:0x3fff07e0 0x40096825:0x3fff0800 0x4011a391:0x3fff0820 0x4011ad9a:0x3fff0840 0x4011a375:0x3fff0860 0x400ebf44:0x3fff0880 0x4011c2b1:0x3fff08a0 0x4011c2f5:0x3fff08c0 0x4011c3e4:0x3fff08e0 0x4011adff:0x3fff0900 0x4017e823:0x3fff0920 0x4011b06d:0x3fff0940 0x4011b0d6:0x3fff0990 0x40092b8e:0x3fff09b0
0x400d3a3e: panic_abort at C:/esp-idf/components/esp_system/panic.c:391

0x4008eb31: esp_system_abort at C:/esp-idf/components/esp_system/esp_system.c:129

0x400967fa: abort at C:/esp-idf/components/newlib/abort.c:46

0x40085a07: lock_acquire_generic at C:/esp-idf/components/newlib/locks.c:139

0x40085b65: _lock_acquire_recursive at C:/esp-idf/components/newlib/locks.c:167

0x40085c01: __retarget_lock_acquire_recursive at C:/esp-idf/components/newlib/locks.c:323

0x40170231: _vfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x4016baaf: fiprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fiprintf.c:48

0x4016ba2a: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:58 (discriminator 8)

0x4009484a: tlsf_free at C:/esp-idf/components/heap/heap_tlsf.c:866 (discriminator 1)

0x400951c2: multi_heap_free_impl at C:/esp-idf/components/heap/multi_heap.c:220

0x4008218a: heap_caps_free at C:/esp-idf/components/heap/heap_caps.c:305

0x40096825: free at C:/esp-idf/components/newlib/heap.c:46

0x4011a391: esp_tls_internal_event_tracker_destroy at C:/esp-idf/components/esp-tls/esp_tls_error_capture.c:46

0x4011ad9a: esp_mbedtls_server_session_delete at C:/esp-idf/components/esp-tls/esp_tls_mbedtls.c:629

0x4011a375: esp_tls_server_session_delete at C:/esp-idf/components/esp-tls/esp_tls.c:584

0x400ebf44: httpd_ssl_close at C:/esp-idf/components/esp_https_server/src/https_server.c:28

0x4011c2b1: httpd_sess_free_ctx at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:230

0x4011c2f5: httpd_sess_clear_ctx at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:251

0x4011c3e4: httpd_sess_delete at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:365

0x4011adff: httpd_process_session at C:/esp-idf/components/esp_http_server/src/httpd_main.c:179

0x4017e823: httpd_sess_enum at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:50 (discriminator 1)

0x4011b06d: httpd_server at C:/esp-idf/components/esp_http_server/src/httpd_main.c:228

0x4011b0d6: httpd_thread at C:/esp-idf/components/esp_http_server/src/httpd_main.c:250 (discriminator 15)

0x40092b8e: vPortTaskWrapper at C:/esp-idf/components/freertos/port/xtensa/port.c:168
```

chipweinberger, 2021-09-30:
most likely is heap corruption. 

Turn on the comprehensive heap debugger, and call "check heap integrity" in many places in your code.

baldhead69, 2021-10-01:
@chipweinberger 

if the problem is heap corruption, i think the problem is in some component of esp-idf, however, i need to debug the code.

baldhead69, 2021-10-03:
hi,

When i close my wss client in android app, the esp32 crash.

```c
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Appears that mbedtls are the problem.

Core  1 register dump:
PC      : 0x4012fbdd  PS      : 0x00060c30  A0      : 0x8013a159  A1      : 0x3fff1af0  
0x4012fbdd: mbedtls_pk_free at C:/esp-idf/components/mbedtls/mbedtls/library/pk.c:97

A2      : 0xfefeffba  A3      : 0x3ffeed58  A4      : 0x0000000c  A5      : 0x3ffd0a60
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3fff1bf0  A9      : 0x3fff1bd0
A10     : 0x3ffeed58  A11     : 0x00000037  A12     : 0x00000000  A13     : 0x3ffb1ecc

0x4011aebd: esp_mbedtls_server_session_delete at C:/esp-idf/components/esp-tls/esp_tls_mbedtls.c:628

0x4011a4a1: esp_tls_server_session_delete at C:/esp-idf/components/esp-tls/esp_tls.c:584

0x400ec070: httpd_ssl_close at C:/esp-idf/components/esp_https_server/src/https_server.c:28

0x4011c3dd: httpd_sess_free_ctx at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:230
0x4011c421: httpd_sess_clear_ctx at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:251

0x4011c510: httpd_sess_delete at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:365

0x4011c752: httpd_sess_close at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:145

0x4011afb1: httpd_process_ctrl_msg at C:/esp-idf/components/esp_http_server/src/httpd_main.c:150 (discriminator 15)

0x4011b17e: httpd_server at C:/esp-idf/components/esp_http_server/src/httpd_main.c:215 (discriminator 15)

0x4011b202: httpd_thread at C:/esp-idf/components/esp_http_server/src/httpd_main.c:250 (discriminator 15)

0x40092bb6: vPortTaskWrapper at C:/esp-idf/components/freertos/port/xtensa/port.c:168
```

baldhead69, 2021-10-03:
@ESP-YJM 

hi,

I think that i encouter the problem.
After i comment the "closes_socket_instantly(hd, sockfd);", the problem disappeared.

How can i solve this ?

I still want to release server resources instantly.

```c
// typedef void (*httpd_close_func_t)(httpd_handle_t hd, int sockfd);

static void closed_socket_handler(httpd_handle_t hd, int sockfd)
{    
    ESP_LOGI(WSS_SERVER_TAG, LOG_USER("Closed socket handler:"));
    ESP_LOGI(WSS_SERVER_TAG, LOG_USER("hd = %p"), hd);
    ESP_LOGI(WSS_SERVER_TAG, LOG_USER("Closed socket = %d\n"), sockfd);
      
    remove_ws_client_from_list(sockfd);
    
    //closes_socket_instantly(hd, sockfd);
}

// Closes the socket instantly, freeing up server resources.

static inline void closes_socket_instantly(httpd_handle_t hd, int sockfd)
{
    esp_tls_t *tls = httpd_sess_get_transport_ctx(hd, sockfd);
    esp_tls_conn_delete(tls);
    struct linger linger;
    linger.l_onoff = 1;
    linger.l_linger = 0;
    if( setsockopt(sockfd, SOL_SOCKET, SO_LINGER, &linger, sizeof(linger) == 0) )
    {
        ESP_LOGI(WSS_SERVER_TAG, "Set SO_LINGER success");
        close(sockfd); // Now call close will send RST to peer and free pcb immediately.
    }
 
    ESP_LOGI(WSS_SERVER_TAG, "Closing Socket %d, Heap:%d\n", sockfd, esp_get_free_heap_size());    
}
```

baldhead69, 2021-10-04:
@ESP-YJM,

I found an error in the code that i copied from here:
https://esp32.com/viewtopic.php?f=13&t=20705&p=84642#p84642

I already commented in the forum, if you want to fix the code.

I think that this line of code are incorrect "if( setsockopt(sockfd, SOL_SOCKET, SO_LINGER, &linger, sizeof(linger) == 0 )".

Should be:
```c
if( setsockopt( sockfd, SOL_SOCKET, SO_LINGER, &linger, sizeof(linger) ) == 0 )
or
int ret = setsockopt(sockfd, SOL_SOCKET, SO_LINGER, &linger, sizeof(linger) );
if( ret == 0 )
.....
```

The strange thing is that neither the compiler nor the visual studio code showed the error.

```c
static inline void closes_socket_instantly(httpd_handle_t hd, int sockfd)
{
    esp_tls_t* tls = httpd_sess_get_transport_ctx(hd, sockfd);  
    
    printf("tls: %p\n\n", tls);
    
    if( tls == NULL ) 
        return;  
    
    esp_tls_conn_delete(tls);
    struct linger linger;
    linger.l_onoff = 1;
    linger.l_linger = 0;

    int ret = setsockopt(sockfd, SOL_SOCKET, SO_LINGER, &linger, sizeof(linger) ); 
    if( ret == 0 )
    {
        ESP_LOGI(WSS_SERVER_TAG, "Set SO_LINGER success");
        close(sockfd); // Now call close will send RST to peer and free pcb immediately.
    }
    
    ESP_LOGI(WSS_SERVER_TAG, "Closing Socket %d, Heap:%d\n", sockfd, esp_get_free_heap_size());    
}
```

I changed the code, but the problem still persists.

ESP-YJM, 2021-10-08:
@baldhead69 Maybe in your `closes_socket_instantly` function, we no need free `esp_tls_conn_delete` and only need close socket fd. See like https://github.com/espressif/esp-idf/blob/58022f8599401740979ac7657a57348285a70374/components/esp_http_server/src/httpd_sess.c#L358-L361. You can remove logic about `esp_tls_conn_delete` and try.

ESP-YJM, 2021-10-09:
@baldhead69 You can test it and from code logic, if the callback function `close_fn` not defined, it only close the fd, also will not call `esp_tls_conn_delete `. So i think you can only follow the code to close the socket by `SO_LINGER` option.

baldhead69, 2021-10-10:
@ESP-YJM 

WORKED. Thank's.

```c
static inline void closes_socket_instantly(httpd_handle_t hd, int sockfd)
{ 
    struct linger linger;
    linger.l_onoff = 1;
    linger.l_linger = 0;

    int ret = setsockopt(sockfd, SOL_SOCKET, SO_LINGER, &linger, sizeof(linger) ); 
    if( ret == 0 )
    {
        ESP_LOGI(WSS_SERVER_TAG, "Set SO_LINGER success");
        close(sockfd); // Now call close will send RST to peer and free pcb immediately.
    }
    
    ESP_LOGI(WSS_SERVER_TAG, "Heap: %d", esp_get_free_heap_size());   
    ESP_LOGI(WSS_SERVER_TAG, "MALLOC_CAP_SPIRAM: %d", heap_caps_get_free_size(MALLOC_CAP_SPIRAM) );
    ESP_LOGI(WSS_SERVER_TAG, "MALLOC_CAP_INTERNAL: %d\n", heap_caps_get_free_size(MALLOC_CAP_INTERNAL) );    
}
```

closed 2021-10-10 by baldhead69

luzidchris, 2023-06-26:
Hi,

I'd like to come back to this issue because I'm facing a similar effect. In my case, I'm calling a function to dump a string to stdout, i.e., eventually, `_fwrite_r` is called. This happens in context of an Ada function which represents some kind of destructor (`Finalize`) - I know this is not officially supported, but I couldn't find any ISR context in my backtrace, similar to what's described here.

I could work-around the problem by modifying functions `lock_acquire_generic` and `lock_released generic` in `locks.c` in component `newlib`. After trying to find out more using gdb, I ended up comparing implementation of `newlib` of `esp-idf` with the one of `ESP8266_RTOS_SDK` (with this SDK, I didn't observe the issue). And I in fact found a difference exactly in the checks which determine whether to abort or not:

 * `esp-idf` uses `xPortCanYield` to determine wheter we're in ISR context
 *  `ESP8266_RTOS_SDK` uses `xPortInIsrContext`

I changed `esp-idf` code to use `xPortInIsrContext` and my problem disappeared, and since I doubt that `xPortInIsrContext` is completely non-sense, and because it seems to be appropriate (the check is about "ISR context"), I'd like to kindly ask for some clarification on this because it would be really interesting to have an explanation for this behavior and the different result of `xPortInIsrContext` and `xPortInIsrContext`.

While debugging, I could observe PS (processor status) register being 0x....3 and in another case it was 0x....5. `xPortCanYield` basically checks for `PS & 0xF` to identify the INT level and if the result is not 0, it returns FALSE. So, the function in fact returned the truth from its standpoint, but I simply couldn't identify the ISR context.

 * Does anybody know why `esp-idf`'s `newlib` version make use of `xPortCanYield` instead of `xPortInIsrContext` to check for ISR contex?
 * What would be the impact of using `xPortInIsrContext`? Are there any negative side-effects?
 * Does PS refer to both cores? Where can I find some more documentation of this register to identify the exact meaning of its content?
 * Some comments here mention some kind of heap check but I couldn't find more information on this. Could anyone point me in the right direction?

Thanks in advance for any remarks or hints you could provide!

Best regards!
Chris

igrr, 2023-06-26:
Hi @luzidchris,

When newlib locks implementation attempts to acquire a mutex, it tries to ensure that this is safe to do. When that code was first written, we thought that only acquiring the mutex in an ISR is not safe, so we added a check for this. Indeed, trying to use a blocking FreeRTOS API from an ISR won't work.

Later, we found that checking for an ISR is not sufficient, and we have to also ensure that if a FreeRTOS API tries to block, it can successfully switch context to another task (i.e. yield). This lead to `xPortInIsrContext` being replaced with `xPortCanYield` (in https://github.com/espressif/esp-idf/commit/4987a5ad904d7ab469853f28bc2345f9248bc872). Unfortunately, we didn't notice that the comments surrounding the code were not updated to match the new implementation, so now it does look confusing!

> What would be the impact of using xPortInIsrContext? Are there any negative side-effects?

The negative side effect fixed by the above mentioned commit is that if a newlib lock is taken while the scheduler can't yield, FreeRTOS internal data structures may get corrupted, most likely resulting in a watchdog reset. This would happen, for instance, if a newlib lock is acquired inside a critical section (between portENTER_CRITICAL and portEXIT_CRITICAL.)

This restriction (no yielding in critical sections) wasn't present in the ESP8266_RTOS_SDK, if i recall correctly, so the code there is still using `xPortInIsrContext`.

---

If you aren't using critical sections in your code, I think most likely you are running into some kind of heap corruption. This could explain why the PS register gets changed. You can try enabling some of the [heap corruption detection](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-corruption-detection) features. I would also recommend enabling [`CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-freertos-watchpoint-end-of-stack) option, just in case — it helps detect stack overflows.

> Does PS refer to both cores? Where can I find some more documentation of this register to identify the exact meaning of its content?

There is a separate PS register for each CPU. You can find the description on page 132 of the [Xtensa ISA summary](https://www.cadence.com/content/dam/cadence-www/global/en_US/documents/tools/ip/tensilica-ip/isa-summary.pdf)

> How could a heap corruption cause the register to change its content?

For example, an out-of-bounds write to a buffer located immediately before the task stack could overwrite the register context of a task, if the task is blocked and its context is saved to the stack.

luzidchris, 2023-06-26:
Hi,

thanks so much for the quick reply and for the detailed explanation! I now understand why `xPortInIsrContext` was changed and will search the Ada runtime code to check for critical sections. I remember that `portENTER_CRITICAL`/`portEXIT_CRITICAL` were used somewhere in context of the "finalization" - so, it would perfectly make sense that I ended up in `abort`. It seems I focused too much in the ISR context^^.

The heap corruption detection and the stack watchpoint both seem to be very valuable features. I'm sure I'll benefit from that, if not now, then certainly sometime in the future.

I really appreciate your help! I'm sure I'll be able to track down the root cause and find a way to resolve the problem! Thank you very, very much!