espressif/esp-idf #7626: 【ESP32-C3/BLE Feature】Some info about Extended ADV, the max number of connections

xiaolongba, 2021-10-01:
Hi Espressif Experts,

Regarding the ESP32-C3, I would like to inquire something about BLE features, the details are as below:

1. As we all know, the ESP32-C3 can support full features of BLE 5.0, including the Coded PHY, 1M PHY, 2M PHY, Extended ADV, but I don't know if all of them are available for users? I mean that the current SDK can support all of BLE 5.0 features already? I tried to consult the ESP-IDF and found that Coded PHY,1M PHY, 2M PHY will be possible, but I am not sure if all Extended Adv features can be supported as the Extended Adv includes period extended adv, chained extended adv, etc.
2. I also looked into the lots of materials online, I still cannot confirm the max number of BLE connections that can be supported. a lot of answers told me the number is 8 or 9, is it possible to implement more than 8 connections, such as 20,30 or more connections. can the Espressif guys help me double confirm?
3. I tried to find something about BLE stack, it told me that the ESP32-C3 will use Bluedroid host or Nimble host, but the controller is developed by espressif based on CEVA IP, Am I correct?

Reason:

I am preparing to write some BLE tutorials about ESP32-C3, the info above is important for me as I need to make the outline for this BLE series tutorials.

could you give me your hands, thanks in advance

chegewara, 2021-10-01:
I can only answer on point (1):
- all 3 PHY are supported,
- extended ADV - supported,
- periodic advertising with sync scan supported,
- multi-advertising - supported

Basically all API with `_ext_` in name are BLE 5, plus few functions that are without `_ext_`:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/bluetooth/esp_gap_ble.html

xiaolongba, 2021-10-01:
> I can only answer on point (1):
> 
> * all 3 PHY are supported,
> * extended ADV - supported,
> * periodic advertising with sync scan supported,
> * multi-advertising - supported
> 
> Basically all API with `_ext_` in name are BLE 5, plus few functions that are without `_ext_`: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/bluetooth/esp_gap_ble.html
> 
> PS if you would like to test or add some info in your tutorials, then some or most BLE 5 from bludroid are added in arduino-esp32 https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE/examples

Hi @chegewara Bro,

Many thanks for your detailed responses. regarding point1, some of them are clear, but the chained advertisement is supported or not? also, the max advertising length can be up to 1650 bytes for extended adv?

as for the other two points, could you help me check with your teammates even though you can only answer the point1?

thanks again and waiting for your news.

chegewara, 2021-10-01:
Sorry, but i have no knowledge about chained advertising, just didnt test it yet.
I also have to disappoint you, but i dont have teammates, i guess espressif have policy to not hire chegewaras.

Here is one more info, i hope it can help you:
https://esp32.com/viewtopic.php?f=19&t=21011#p77540

xiaolongba, 2021-10-01:
Hi @chegewara Bro,

it doesn't matter, Actually, I have implemented the BLE 5.0 new features with other platforms. the purpose is to know if the esp32-c3 can support them as well, but I still cannot confirm as the lack of materials.

so let's waiting for the feedback from espressif guys together at this moment.

at last, many thanks your kindly help.

xiaolongba, 2021-10-09:
any updates about it?

xiewenxiang, 2021-10-09:
@xiaolongba 

 The max number of BLE connections of ESP32C3 is 9.

xiaolongba, 2021-10-09:
@xiewenxiang 

thanks. where can I find this info? other two doubts, could you help me double-check?

xiaolongba, 2021-10-11:
appended more info here, the last doubt I mentioned, it seems that it is the new feature from BLE spec 5.2, not BLE spec 5.0. so you can ignore this doubt, other doubts above can help me check again?

thanks in advance.

xiaolongba, 2021-10-12:
Hi @Sushant-Espressif Bro,

any updates about it? I need your help, pls.

xiaolongba, 2021-10-13:
Hi @Sushant-Espressif Bro,

you are a good man.

> Answer for both the questions is same :-)
> Support in controller as well as host stack is needed. Both Bluedroid and Nimble stack support these features. For Nimble stack, make sure that required features are enabled from menuconfig (component config --> Bluetooth --> Nimble options ). For Bluedroid stack it is enabled by default. In BLE controller as well these features are enabled by default.

great, clear. but could you help me check the following doubt:

1. at this moment, the max connections of esp32-c3 BLE controller have been fixed to 9? it is impossible to increase the number of connections? even if the host stack can support more than 9 or more, the final connections are still 9?  correct?

waiting for your feedback, I promise that this is the last question about it😂

closed 2021-10-14 by xiaolongba

iamcalledusman, 2022-03-23:
Hi All ,
Very nice discussion and very helpfull thank you!

I have a question if you could answer it , i tried Nimble spp_client and server . But i am unable to connect more than 1 device (esp32c3) .

All the detail is mentioned in that link it includes all the testing and results 

https://github.com/espressif/esp-idf/issues/8619

Thanks in advance , if you could help me in this regard,

xiaolongba, 2022-03-24:
@iamcalledusman 

Hello Bro,

ESP32-C3 can only support Bluetooth LE only, but SPP is the feature that is from Bluetooth classic. I think you should implement these two examples with ESP32

iamcalledusman, 2022-03-24:
Hello Bro,
Thank you so much for the response! Right I will try to implement it with esp32.

Thank You!