espressif/esp-idf #7627: Getting error for esp-idf modbus  slave example testing

dimpalsharm, 2021-10-01:
Hello,

I am trying to run the esp-idf modbus slave examplewith modbus master/poll tool
[image]
getting read/write error in poll tool like above.

alisitsyn, 2021-10-02:
Hello @dimpalsharm,

The Break condition means that the transmitter is holding the line in zero state when transmittion is done for a long time. This often heppens when RS485 line drivers do not work correctly. 
1. Please use the [rs485 echo example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo_rs485) to check interface connection first.
2. Flash the Modbus slave example `idf.py -p PORT flash monitor'.
3. Start modbus communication then use "Display communication traffic" option in the tool bar or menu of your "Modbus Master Poll" software. This will show communication log which will allow to recognize the problem. Please include the log here.
4. Include the [mb_slave example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/modbus/serial/mb_slave) log here.

Thanks.

alisitsyn, 2021-10-08:
@dimpalsharm,

Could you give the update for this issue? 
Also, you are creating several test issues for Modbus that are related to each other that prevents to track them appropriately.

closed 2021-11-22 by Alvin1Zhang