espressif/esp-idf #7628: Classic bluetooth reconnect bug

SimonPVS, 2021-10-01:
## Environment

- Development Kit:  ESP-32 WROOM 
- Kit version V3
- Module or chip used:  ESP32-WROOM-32
- IDF version: esp-idf-v4.2.1
- Build System: idf.py
- Compiler version: esp-2020r3-8.4.0
- Operating System: Windows 10
- Using an IDE?: Eclipse V4.19.0.v20210303-1800
- Power Supply:  External 3.3

## Problem Description

I am using classic bluetooth on my device and pairing, disconnect,.. all seem to work fine. For my use case i should be able to remove bonded devices. After removing the bonded devices using: 'esp_bt_gap_remove_bond_device' , the ESP returns 0 as number of bonded devices.
But when i set the bluetooth discovery mode to 0, my mobile phone (android) is still able to reconnect to the ESP. It will get a new key and is able to reconnect. 

### Expected Behavior

I would expect when removing a bonded bluetooth device, that it can not reconect to the ESP as long as the ESP is not discoverable.

### Actual Behavior

The removed bonded device is able to reconnect to the ESP. 

### Steps to reproduce

1. Bond a mobile phone to the ESP (SSP)
2. Remove the bonded device from the ESP using: esp_bt_gap_remove_bond_device
3. Make sure the ESP is now set to: 'ESP_BT_NON_DISCOVERABLE'
4. Try to reconnect using a mobile phone

### Code to reproduce this issue

I am using the a2dp example provided by esspresif.

SimonPVS, 2021-10-06:
To fix this, you should set BTM_ALLOW_CONN_IF_NONDISCOVER to FALSE
(components/bt/host/bluedroid/common/include/common/bt_target.h

[image]

BetterJincheng, 2021-10-08:
Hi, @SimonPVS,

Based on your description about the connection behavior, I am afraid it is just correct.

When your android phone connected once successfully to ESP32, the phone will record some information including peer bluetooth device address. When your phone tried reconnecting to ESP32, it will just connect to the recorded address without an INQUIRY process, so it just connected successfully even if the ESP32 is non-discoverable. 

So, removing bonded device locally on ESP32 and set mode NON-DISCOVERABLE will not affect the connection behavior of peer device.

Turn to us anytime.

Thanks

SimonPVS, 2021-10-20:
Okay thanks!

closed 2021-11-08 by Alvin1Zhang