espressif/esp-idf #7629: Build fails due to unused static variable "LOG_TAG" in esp-idf/components/esp_adc_cal/esp_adc_cal_esp32.c

zinke-ct-video, 2021-10-01:
Hello everyone,

I am trying to compile my project using `idf.py build` on a Debian machine with latest ESP-IDF (commit 3e370c4296247b349aa3b9a0076c05b9946d47dc).
But I get the following error:

> /home/user-xy/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_IDF_LWIP_HOOK_FILENAME=\"global_lwip_hooks.h\" -I/home/user-xy/esp/projects/project-xy/webinterface/main/global-include -Iconfig -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_adc_cal/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/newlib/platform_include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/freertos/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/freertos/include/esp_additions/freertos -I/home/user-xy/esp/projects/project-xy/esp-idf/components/freertos/port/xtensa/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/freertos/include/esp_additions -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_hw_support/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_hw_support/include/soc -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_hw_support/port/esp32/. -I/home/user-xy/esp/projects/project-xy/esp-idf/components/heap/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/log/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/lwip/include/apps -I/home/user-xy/esp/projects/project-xy/esp-idf/components/lwip/include/apps/sntp -I/home/user-xy/esp/projects/project-xy/esp-idf/components/lwip/lwip/src/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/lwip/port/esp32/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/lwip/port/esp32/include/arch -I/home/user-xy/esp/projects/project-xy/esp-idf/components/soc/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/soc/esp32/. -I/home/user-xy/esp/projects/project-xy/esp-idf/components/soc/esp32/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/hal/esp32/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/hal/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/hal/platform_port/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_rom/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_rom/include/esp32 -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_rom/esp32 -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_common/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_system/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_system/port/soc -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_system/port/public_compat -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp32/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/xtensa/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/xtensa/esp32/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/driver/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/driver/esp32/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_pm/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_ringbuf/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/efuse/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/efuse/esp32/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/vfs/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_wifi/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_event/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_netif/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_eth/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/tcpip_adapter/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_phy/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_phy/esp32/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_ipc/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/app_trace/include -I/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_timer/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -O2 -fmacro-prefix-map=/home/user-xy/esp/projects/project-xy/webinterface=. -fmacro-prefix-map=/home/user-xy/esp/projects/project-xy/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4-dev-3235-g3e370c4296\" -DESP_PLATFORM -DNDEBUG -MD -MT esp-idf/esp_adc_cal/CMakeFiles/__idf_esp_adc_cal.dir/esp_adc_cal_esp32.c.obj -MF esp-idf/esp_adc_cal/CMakeFiles/__idf_esp_adc_cal.dir/esp_adc_cal_esp32.c.obj.d -o esp-idf/esp_adc_cal/CMakeFiles/__idf_esp_adc_cal.dir/esp_adc_cal_esp32.c.obj   -c /home/user-xy/esp/projects/project-xy/esp-idf/components/esp_adc_cal/esp_adc_cal_esp32.c
**/home/user-xy/esp/projects/project-xy/esp-idf/components/esp_adc_cal/esp_adc_cal_esp32.c:86:19: error: 'LOG_TAG' defined but not used [-Werror=unused-const-variable=]
 const static char LOG_TAG[] = "ADC_CALI";
                   ^~~~~~~
cc1: some warnings being treated as errors**
[747/1085] Building C object esp-idf/console/CMakeFiles/__idf_console.dir/esp_console_repl.c.obj
[748/1085] Building C object esp-idf/esp_hid/CMakeFiles/__idf_esp_hid.dir/src/esp_hidd.c.obj
[749/1085] Building C object esp-idf/console/CMakeFiles/__idf_console.dir/linenoise/linenoise.c.obj
[750/1085] Building C object esp-idf/console/CMakeFiles/__idf_console.dir/argtable3/argtable3.c.obj
[751/1085] Building CXX object esp-idf/asio/CMakeFiles/__idf_asio.dir/asio/asio/src/asio.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
Executing action: all (aliases: build)
Running cmake in directory /home/user-xy/esp/projects/project-xy/webinterface/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/user-xy/esp/projects/project-xy/webinterface"...
Running ninja in directory /home/user-xy/esp/projects/project-xy/webinterface/build
Executing "ninja all"...

I noticed during compilation that this warning (here treated as an error) also occurs in other components of the ESP-IDF but there it is not treated as an error.
Very interesting is that the error does **NOT** occur with the same project, same ESP-IDF commit on my windows machine.
Any ideas why this error pops up on linux (and interestingly not on windows) ?

Don't know if it's important, but I have customized the CMakeLists.txt in the project folder (not in the main folder), because I need a global LwIP hook:
>cmake_minimum_required(VERSION 3.5)
> include($ENV{IDF_PATH}/tools/cmake/project.cmake)
> add_compile_definitions(ESP_IDF_LWIP_HOOK_FILENAME="global_lwip_hooks.h")
> include_directories("main/global-include")
> project(My-Project)

Thank you!

igrr, 2021-10-01:
Hi @zinke-ct-video,
are the sdkconfig files in your project directory the same between the two computers?
It looks like this warning may occur if the log level is set to "None" in sdkconfig.

zinke-ct-video, 2022-01-12:
I have tested compilation again with commit df1f9e of the master branch. The issue does not appear anymore. Seems as if someone has solved this. Thanks a lot!

closed 2022-01-12 by zinke-ct-video