espressif/esp-idf #7630: Error in mbedtls_base64_decode() --> buffer not fully translated

MartyMcFlyInTheSky, 2021-10-01:
It seems that the base64 decoding functionality in mbedtls is partially broken as the last bits are just cut off if t hey don't match up to a four base64 characters which can then be concatenated to 3 bytes.

Try decoding the following:

eyJzdWIiOiJGNDpDRjpBMjo5RjpGNjpCQyIsImp0aSI6IjY1OGRhOGI1LWE2MjktNDFkYS04ZjRjLWRmMmExZDc0NTYwOCIsImlhdCI6MTYzMzA5MzM2NiwiZXhwIjoxNjMzMDkzNDI2fQ

Using a decoder from a website (like https://www.base64decode.org/) this yields:

{"sub":"F4:CF:A2:9F:F6:BC","jti":"658da8b5-a629-41da-8f4c-df2a1d745608","iat":1633093366,"exp":1633093426}

Using mbedtls_base64_decode():

{"sub":"F4:CF:A2:9F:F6:BC","jti":"658da8b5-a629-41da-8f4c-df2a1d745608","iat":1633093366,"exp":1633093426W

Notice the last character: In the former correct version the tail would translate to a closing curly brace } but it doesn't for the mbedtls implementation.

I think I've pinpointed the location of the error. These are the important bits from the decode function:

```
 for( j = 3, n = x = 0, p = dst; i > 0; i--, src++ )
   {
        if( *src == '\r' || *src == '\n' || *src == ' ' )
            continue;

        printf("%c/n=%d, ", *src, n);

        j -= ( base64_dec_map[*src] == 64 );
        x  = ( x << 6 ) | ( base64_dec_map[*src] & 0x3F );

        if( ++n == 4 )
        {
            n = 0;
            if( j > 0 ) *p++ = (unsigned char)( x >> 16 );
            if( j > 1 ) *p++ = (unsigned char)( x >>  8 );
            if( j > 2 ) *p++ = (unsigned char)( x       );
        }
    }
```
i was formerly set to the end of the src string, so the loop would terminate if all characters of src have been walked through. You can see that n is increased every fourth time. only if n is equal to four, the destination string character *p is set to the calculated value. Which leaves me wondering what happens if the source string doesn't match up to a multiple of four? Indeed it appears that the byte is simply not written in this case. As you can see I placed a log in the above sample which tracks n for each character:

`e/n=0, y/n=1, J/n=2, z/n=3, d/n=0, W/n=1, I/n=2, i/n=3, O/n=0, i/n=1, J/n=2, G/n=3, N/n=0, D/n=1, p/n=2, D/n=3, R/n=0, j/n=1, p/n=2, B/n=3, M/n=0, j/n=1, o/n=2, 5/n=3, R/n=0, j/n=1, p/n=2, G/n=3, N/n=0, j/n=1, p/n=2, C/n=3, Q/n=0, y/n=1, I/n=2, s/n=3, I/n=0, m/n=1, p/n=2, 0/n=3, a/n=0, S/n=1, I/n=2, 6/n=3, I/n=0, j/n=1, Y/n=2, 1/n=3, O/n=0, G/n=1, R/n=2, h/n=3, O/n=0, G/n=1, I/n=2, 1/n=3, L/n=0, W/n=1, E/n=2, 2/n=3, M/n=0, j/n=1, k/n=2, t/n=3, N/n=0, D/n=1, F/n=2, k/n=3, Y/n=0, S/n=1, 0/n=2, 4/n=3, Z/n=0, j/n=1, R/n=2, j/n=3, L/n=0, W/n=1, R/n=2, m/n=3, M/n=0, m/n=1, E/n=2, x/n=3, Z/n=0, D/n=1, c/n=2, 0/n=3, N/n=0, T/n=1, Y/n=2, w/n=3, O/n=0, C/n=1, I/n=2, s/n=3, I/n=0, m/n=1, l/n=2, h/n=3, d/n=0, C/n=1, I/n=2, 6/n=3, M/n=0, T/n=1, Y/n=2, z/n=3, M/n=0, z/n=1, A/n=2, 5/n=3, M/n=0, z/n=1, M/n=2, 2/n=3, N/n=0, i/n=1, w/n=2, i/n=3, Z/n=0, X/n=1, h/n=2, w/n=3, I/n=0, j/n=1, o/n=2, x/n=3, N/n=0, j/n=1, M/n=2, z/n=3, M/n=0, D/n=1, k/n=2, z/n=3, N/n=0, D/n=1, I/n=2, 2/n=3, f/n=0, Q/n=1,`

You can see that the last base64 characters f and Q don't add up to n==4, so they are just dropped/ignored. I think this is where the irregularity stems from?

Or am I supposed to fill in characters so the base64 string length is always % 4 divisible?

negativekelvin, 2021-10-02:
> In some implementations, the padding character is mandatory, while for others it is not used.

I guess in this case it is needed

MartyMcFlyInTheSky, 2021-10-04:
> > In some implementations, the padding character is mandatory, while for others it is not used.
> 
> I guess in this case it is needed

Ok so this is a design choice then. This should be documented at least.

atanisoft, 2021-10-04:
Usually b64 strings are padded as part of encoding with "=" characters.

AdityaHPatwardhan, 2021-10-14:
Hi @MartyMcFlyInTheSky If this is an issue related with `mbedtls_base64_decode` then can you please raise the issue in https://github.com/ARMmbed/mbedtls.
Thanks