espressif/esp-idf #7631: Can't load some certificates with cryptography v35.0.0

electrocucaracha, 2021-10-01:
## Problem Description

The latest cryptography python module doesn't allow to [parse X.509 certificates with negative serial numbers]( https://github.com/pyca/cryptography/blob/35.0.0/CHANGELOG.rst). This new behavior affects the [EC-ACC certificate](https://github.com/espressif/esp-idf/blob/v4.3/components/mbedtls/esp_crt_bundle/cacrt_all.pem#L1648-L1676).

```bash
$ openssl x509 -in ec-acc.pem -text                                                                                               
Certificate:                                                    
    Data:                                                       
        Version: 3 (0x2)                                        
        Serial Number:                                          
             (Negative)11:d4:c2:14:2b:de:21:eb:57:9d:53:fb:0c:22:3b:ff
        Signature Algorithm: sha1WithRSAEncryption
```

### Expected Behavior

```bash
$ python components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input components/mbedtls/esp_crt_bundle/cacrt_all.pem -q
```

### Actual Behavior

```bash
$ python components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input components/mbedtls/esp_crt_bundle/cacrt_all.pem -q
gen_crt_bundle.py: Invalid certificate in components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
```

### Steps to reproduce

```bash
$ virtualenv .venv
$ source .venv/bin/activate
$ IDF_PATH=$(pwd) pip install -r requirements.txt
$ python components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input components/mbedtls/esp_crt_bundle/cacrt_all.pem -q
```

manuelbl, 2021-10-03:
As this issue correctly points out, the problem was triggered by the [Python cryptography package](https://pypi.org/project/cryptography/). The [change log](https://cryptography.io/en/latest/changelog/) says:

> **BACKWARDS INCOMPATIBLE**: The X.509 certificate parser no longer allows negative serial numbers. RFC 5280 has always prohibited these

Whether you are affected by this bug or not depends on the date when the ESP-IDF tools were installed and – as part of it – the virtual Python environment in `~/.espressif/python_env/idfx.x_pyx.x_env was created. So this is a very peculiar bug:
- It is not triggered by a change of the ESP-IDF framework.
- It will affect several ESP-IDF versions.
- Users with a CI/CD pipeline, which reinstalls the ESP-IDF tools for each build, will be suddenly affected without making changes to their software. So will all software engineers who setup a new computer for developing ESP-IDF software.

The bug likely affects all ESP-IDF versions using Python 3, at least v4.3, v4.3.1 and v4.4. Thus the fix should probably be back ported to all of those.

It likely does not affect versions using Python 2 such as ESP-IDF v4.2 and earlier.

It might make sense to investigate whether ESP-IDF should create a virtual Python environment with fixed package versions.

electrocucaracha, 2021-10-03:
> It might make sense to investigate whether ESP-IDF should create a virtual Python environment with fixed package versions.

Adding a constraint [here](https://github.com/espressif/esp-idf/blob/master/requirements.txt#L11), like `cryptography>=2.1.4,35.0.0`, can solves the issue but it will lose some library improvements and/or fixes.

francois-berder, 2021-10-04:
> The bug likely affects all ESP-IDF versions using Python 3, at least v4.3, v4.3.1 and v4.4. Thus the fix should probably be back ported to all of those.

I confirm that this bug also affects v4.2

dmezh, 2021-10-05:
This is a pretty big problem that totally breaks builds for (presumably) everyone with updated packages and the default sdkconfig (at least in PlatformIO). Maybe resolving the problem in the immediate term should take priority (hold back `cryptography` below 35.0.0?) - this is already a few days old at this point.

igrr, 2021-10-05:
Hi all, sorry for the issue and the delayed response.

Thanks to @electrocucaracha for submitting #7621 with a temporary fix this issue. We have merged this PR to `master`, and will try to get the branch tested and published to Github soon. There will be a follow-up commit to update the list of certificates in esp_crt_bundle. These fixes will be backported to the affected release branches and will be included into the next bugfix releases.

Release tags are permanent, so we will not be able to fix this issue for past releases. There are several workarounds for this issue you can apply, depending on the situation:

* If you aren't using the Certificate Bundle feature, disable `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE` option. You can do this in menuconfig (under Component Config, mbedTLS, Certificate Bundle) or by appending `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=n` to your `sdkconfig` and/or `sdkconfig.defaults` files.
* If you are using the Certificate Bundle feature, choose "Use only the most common certificates from the default bundles" option (`CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN`) instead of "Use the full default certificate bundle". This can be done in menuconfig or by appending `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y` to your `sdkconfig` and/or `sdkconfig.defaults` files. This reduces the number of trusted root certificates, and as a side effect removes the EC-ACC certificate.

  If you do need the full list of certificates, enable `CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE`, set `CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE_PATH`, and disable the default bundle using `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_NONE`.
* Downgrade `cryptography` package to version 3.4.8.
  If you are using ESP-IDF from the command line, make sure you have the environment set up, then run `pip install --upgrade cryptography==3.4.8`.
  If you are using ESP-IDF from the official VS Code or Eclipse plugins, open the ESP-IDF terminal from within the IDE and execute the same command there. (In VS Code, press F1 then search for `Open ESP-IDF Terminal`; in Eclipse, follow [these instructions](https://github.com/espressif/idf-eclipse-plugin#idfterminal))

We realize that this issue has serious impact, and will look for ways to avoid similar problems in the future.

We have avoided version-locking `cryptography` because ESP-IDF releases are maintained for a rather long time (30 months). If there is a fix for security issue in a new `cryptography` release, keeping users stuck with an outdated version (an no indication that it is outdated) is a worse issue than meeting a breaking change like this one. However there are clear downsides to the latter as well, as we see in this case, so we will re-evaluate this decision.

mspang, 2021-10-05:
> Hi all, sorry for the issue and the delayed response.
>
> Thanks to @electrocucaracha for submitting #7621 with a temporary fix this issue. We have merged this PR to `master`, and will try to get the [...]

If you are going to lock the version, please do it via pip --constraint in your virtualenv setup and avoid doing it directly in requirements.txt. Otherwise projects that include your requirements.txt won't be able to update their dependencies, and furthermore the combined requirements are likely to be unsatisfiable.

jinliangli, 2021-12-06:
Update:

Allow parsing of negative serial numbers in certificates. Negative serial numbers are prohibited by RFC 5280 so a deprecation warning will be raised whenever they are encountered. A future version of cryptography will drop support for parsing them.

jinliangli, 2021-12-06:
Upgrade cryptography to 36.0.0 ( pip install --upgrade cryptography==36.0.0 ) can also resolve this issue.

The change log of cryptography 36.0.0:

Allow parsing of negative serial numbers in certificates. Negative serial numbers are prohibited by RFC 5280 so a deprecation warning will be raised whenever they are encountered. A future version of cryptography will drop support for parsing them.  For details, please refer to https://cryptography.io/en/latest/changelog/#v36-0-0

PR #7632 (2021-10-12): esp_crt_bundle: remove EC-ACC certificate (IDFGH-5944)
Removing the EC-ACC certificate given it has Negative serial value invalid for [RFC 5280](https://datatracker.ietf.org/doc/html/rfc5280#section-4.1.2.2). 

Fix #7631