espressif/esp-idf #7634: Unable to setup ESP-IDF extension (Certificate has exprired)

Typoception, 2021-10-02:
Just reinstalled esp-idf extension. have done it before without problems but now there's an error stating certificate has expired.

Error: certificate has expired
	at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)
	at TLSSocket.emit (events.js:315:20)
	at TLSSocket._finishInit (_tls_wrap.js:932:8)
	at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)
Error opening esp-idf version list file. HTTP/HTTPS Request error https://dl.espressif.com/dl/esp-idf/idf_versions.txt
Downloading C:\Users\yahsa\esp\.espressif\dist\idf-git-2.30.1-win64.zip...
Error: certificate has expired
	at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)
	at TLSSocket.emit (events.js:315:20)
	at TLSSocket._finishInit (_tls_wrap.js:932:8)
	at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)
Failed download. Retrying...
Error: Error: HTTP/HTTPS Request error https://dl.espressif.com/dl/idf-git/idf-git-2.30.1-win64.zip
Waiting 8 seconds...
Error: certificate has expired
	at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)
	at TLSSocket.emit (events.js:315:20)
	at TLSSocket._finishInit (_tls_wrap.js:932:8)
	at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)
Failed download. Retrying...
Error: Error: HTTP/HTTPS Request error https://dl.espressif.com/dl/idf-git/idf-git-2.30.1-win64.zip
Waiting 16 seconds...
Error: certificate has expired
	at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)
	at TLSSocket.emit (events.js:315:20)
	at TLSSocket._finishInit (_tls_wrap.js:932:8)
	at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)
Failed download. Retrying...
Error: Error: HTTP/HTTPS Request error https://dl.espressif.com/dl/idf-git/idf-git-2.30.1-win64.zip
Installing C:\Users\yahsa\esp\.espressif\tools\idf-git\2.30.1 ...
Error during ESP-IDF setup

chegewara, 2021-10-02:
I believe you may have this error, because its not first report, but it is very strange. Certificate is valid for 6 more weeks.
[image]

Typoception, 2021-10-02:
@chegewara what do you mean not first report? is it because i uninstall and reinstall?

I even reset my pc just to try again but same issue.

italocjs, 2021-10-02:
Same as #7624 .
A workaround is also there.

I'm not 100% sure its the same , but a similar issue is happening all over internet, as the letsencript certificate expired in 30/sep.  s

chegewara, 2021-10-02:
I mean that ive seen some issue about certificate expired recently from other people, i think.

italocjs, 2021-10-02:
> I mean that ive seen some issue about certificate expired recently from other people, i think.

What i've read is that the Letsencrypt certificate expired (which is "one authority level above" the website certificate, if one cant be authenticated, the other also cant).  But as i said, i'm not 100% sure is the same issue.

Typoception, 2021-10-02:
@italocjs thanks man. managed to finally get it working. I didn't know esp-idf can be installed separately, thought only through vscode extension :)

closed 2021-10-02 by Typoception