espressif/esp-idf #7636: certificate has expired [again]

dine909, 2021-10-03:
Guys come on, this isn't the first time this has caught you out.

```
Error: certificate has expired
	at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)
	at TLSSocket.emit (events.js:315:20)
	at TLSSocket._finishInit (_tls_wrap.js:932:8)
	at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)
Error opening esp-idf version list file. HTTP/HTTPS Request error https://dl.espressif.com/dl/esp-idf/idf_versions.txt

```

Cambeltashburton, 2021-10-04:
Hi!
Same issue here. It gets up to loading Ninja in the tools section and then fails. I've had the problem for a couple of days while trying to set up VSCode with esp-idf for the first time. I thought it was just me messing things up.

stekolos-dev, 2021-10-04:
Same problem

rathnakumar1999, 2021-10-04:
Facing the Same Issue

igrr, 2021-10-04:
Hi all, thanks for the reports and apologies for the slow reaction to this issue.

dl.espressif.com was using the following certificate chain, which is [recommended by letsencrypt for compatibility](https://docs.certifytheweb.com/docs/kb/kb-202109-letsencrypt/#chain-2-legacy--your-cert--r3--isrg-root-x1--dst-root-ca-x3):

```
 0 s:/CN=dl.espressif.com
   i:/C=US/O=Let's Encrypt/CN=R3
 1 s:/C=US/O=Let's Encrypt/CN=R3
   i:/C=US/O=Internet Security Research Group/CN=ISRG Root X1
 2 s:/C=US/O=Internet Security Research Group/CN=ISRG Root X1
   i:/O=Digital Signature Trust Co./CN=DST Root CA X3
```

Most operating systems which have "ISRG Root X1" certificate in the certificate store will recognize certificate "1" in the chain as a valid root certificate, and will ignore the fact that certificate "2" is signed by an expired root ("DST Root CA X3").

However, as the above letsencrypt page notes,

> In some cases, the expiry of the root (and its related expiring R3 intermediate certificate) may causes certificates to be considered untrusted or invalid.

Which seems to be what is happening in your case.

We have now updated the certificate chain served by dl.espressif.com to the following:

```
 0 s:/CN=dl.espressif.com
   i:/C=US/O=Let's Encrypt/CN=R3
 1 s:/C=US/O=Let's Encrypt/CN=R3
   i:/C=US/O=Internet Security Research Group/CN=ISRG Root X1
 2 s:/C=US/O=Internet Security Research Group/CN=ISRG Root X1
   i:/C=US/O=Internet Security Research Group/CN=ISRG Root X1
```

which is the ["modern" chain recommended by letsencrypt](https://docs.certifytheweb.com/docs/kb/kb-202109-letsencrypt/#chain-1-modern--your-cert--r3--isrg-root-x1).

Please try the installation process again. If you meet an issue related to the ISRG Root X1 certificate not being trusted, please follow the instructions [here, section "Clients (browsers etc)"](https://docs.certifytheweb.com/docs/kb/kb-202109-letsencrypt/#clients-browsers-etc) to add ISRG Root X1 to your OS certificate store.

dine909, 2021-10-04:
Awesome, thanks for the quick update.

closed 2021-10-04 by dine909