espressif/esp-idf #7638: [mDNS] service announced in _hap._tcp gets removed after a few minutes

albrownwood, 2021-10-03:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-3235-g3e370c4296
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:    macOS
- Using an IDE?: Visual
- Power Supply:  USB

## Problem Description
mdns service gets removed after a few minutes.

### Expected Behavior
Service should stay announced permanently.

### Actual Behavior
Service gets removed after a few minutes.

### Steps to reproduce
I am running the mdns example provided with esp-idf  
in file mdns_example_main.c I changes occurrences of "_http" by "_hap"  
When running the example the service ESP32-WebServer is announced in _hap category as expected, but after a few minutes the service is removed.
[image]

The issue appears with modem/router BGW320-505 provided by ATT for Fiber and very common in US. The issue does not appear with Linksys WRT3200ACM wifi router.

### Code to reproduce this issue
 mdns_example_main.c with changes on occurrences of "_http" replaced by "_hap" 

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6740
load:0x40078000,len:14812
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dev-3235-g3e370c4296-dirty 2nd stage bootloader
I (27) boot: compile time 11:56:55
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=14dfch ( 85500) map
I (142) esp_image: segment 1: paddr=00024e24 vaddr=3ffb0000 size=0382ch ( 14380) load
I (148) esp_image: segment 2: paddr=00028658 vaddr=40080000 size=079c0h ( 31168) load
I (161) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=75ec8h (483016) map
I (336) esp_image: segment 4: paddr=000a5ef0 vaddr=400879c0 size=0ccdch ( 52444) load
I (358) esp_image: segment 5: paddr=000b2bd4 vaddr=50000000 size=00010h (    16) load
I (368) boot: Loaded app from partition at offset 0x10000
I (368) boot: Disabling RNG early entropy source...
I (380) cpu_start: Pro cpu up.
I (381) cpu_start: Starting app cpu, entry point is 0x40081174
0x40081174: call_start_cpu1 at /Users/floriangaldo/dev/esp-idf/components/esp_system/port/cpu_start.c:155

I (0) cpu_start: App cpu up.
I (394) cpu_start: Pro cpu start user code
I (394) cpu_start: cpu freq: 160000000
I (395) cpu_start: Application information:
I (399) cpu_start: Project name:     mdns-test
I (404) cpu_start: App version:      v4.4-dev-3235-g3e370c4296-dirty
I (411) cpu_start: Compile time:     Oct  3 2021 11:56:39
I (417) cpu_start: ELF file SHA256:  cbb987a6e53fac1f...
I (423) cpu_start: ESP-IDF:          v4.4-dev-3235-g3e370c4296-dirty
I (431) heap_init: Initializing. RAM available for dynamic allocation:
I (438) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (444) heap_init: At 3FFB7BC8 len 00028438 (161 KiB): DRAM
I (450) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (456) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (463) heap_init: At 4009469C len 0000B964 (46 KiB): IRAM
I (470) spi_flash: detected chip: generic
I (474) spi_flash: flash io: dio
W (477) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (492) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (538) mdns-test: mdns hostname set to: [esp32-mdns]
I (548) wifi:wifi driver task: 3ffc1838, prio:23, stack:6656, core=0
I (548) system_api: Base MAC address is not set
I (548) system_api: read default base MAC address from EFUSE
I (558) wifi:wifi firmware version: d683787
I (558) wifi:wifi certification version: v7.0
I (558) wifi:config NVS flash: enabled
I (568) wifi:config nano formating: disabled
I (568) wifi:Init data frame dynamic rx buffer num: 32
I (568) wifi:Init management frame dynamic rx buffer num: 32
I (578) wifi:Init management short buffer num: 32
I (578) wifi:Init dynamic tx buffer num: 32
I (588) wifi:Init static rx buffer size: 1600
I (588) wifi:Init static rx buffer num: 10
I (598) wifi:Init dynamic rx buffer num: 32
I (598) wifi_init: rx ba win: 6
I (598) wifi_init: tcpip mbox: 32
I (608) wifi_init: udp mbox: 6
I (608) wifi_init: tcp mbox: 6
I (608) wifi_init: tcp tx win: 5744
I (618) wifi_init: tcp rx win: 5744
I (618) wifi_init: tcp mss: 1440
I (628) wifi_init: WiFi IRAM OP enabled
I (628) wifi_init: WiFi RX IRAM OP enabled
I (638) example_connect: Connecting to ATT2frcbIf...
I (638) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (748) wifi:mode : sta (9c:9c:1f:c9:4e:2c)
I (748) wifi:enable tsf
I (758) example_connect: Waiting for IP(s)
I (2798) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (3538) wifi:state: init -> auth (b0)
I (3548) wifi:state: auth -> assoc (0)
I (3548) wifi:state: assoc -> run (10)
W (3558) wifi:<ba-add>idx:0 (ifx:0, 0c:7c:28:e3:3c:d4), tid:5, ssn:0, winSize:64
I (3578) wifi:connected with ATT2frcbIf, aid = 29, channel 6, BW20, bssid = 0c:7c:28:e3:3c:d4
I (3578) wifi:security: WPA2-PSK, phy: bgn, rssi: -61
I (3578) wifi:pm start, type: 1

W (3598) wifi:<ba-add>idx:1 (ifx:0, 0c:7c:28:e3:3c:d4), tid:0, ssn:0, winSize:64
I (3618) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (5538) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5538) esp_netif_handlers: example_connect: sta ip: 192.168.1.223, mask: 255.255.255.0, gw: 192.168.1.254
I (5548) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.1.223
I (5558) example_connect: Connected to example_connect: sta
I (5568) example_connect: - IPv4 address: 192.168.1.223
I (5578) example_connect: - IPv6 address: fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5588) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
```

david-cermak, 2021-10-05:
@albrownwood I think I would need that router (and maybe other devices) to reproduce this issue.

Is the querier (`Florian’s MacBook`) and the device (`192.168.1.223`) on the same subnet? How do you query for the service?
```
dns-sd -B _services._dns-sd._udp
```
or simply
```
dns-sd -B _hap._tcp
```
I didn't find any `buy` packet in your capture (Standard query response with TTL=0), so let's assume the removal of the service happen due to TTL expiry, which is weird as the default TTL is set to [4500 seconds](https://github.com/espressif/esp-idf/blob/3e370c4296247b349aa3b9a0076c05b9946d47dc/components/mdns/private_include/mdns_private.h#L51), but from your screenshot it was removed after ~3 minutes. 
Could your router somehow alter the mdns TTL of *some* packets? (this is not happening for `_http`, but only for `_hap` services)
But even if it does, the querier would send additional queries (simply one `QM` packets asking for `Standard query 0x0000 PTR _hap._tcp`) and the device should (*must*) reply. No such packets found in your capture, but that would probably happen ~1 minute before you started capturing (if the TTL changed to 120s)

Could you please capture the entire session from advertising the `ESP32-Webserver` service up to its removal and share the wireshark log (please attach as an archive file), so that we can see how the announcement packets and potential re-queries after service expiry?

albrownwood, 2021-10-05:
Hi David,
Thanks for your help!
I verified that the device and the querier are on the same subnet.
I am using the following command to list the services:
`dns-sd -B _hap._tcp`

Here is a new capture:
[image]

david-cermak, 2021-10-06:
@albrownwood Thanks for sharing more details.
I think the problem starts from packet 42183:
```
Multicast Domain Name System (query)
    Transaction ID: 0x0000
    Flags: 0x0000 Standard query
    Questions: 29
    Answer RRs: 2
    Authority RRs: 0
    Additional RRs: 1
    Queries
        ...
        ...
        _hap._tcp.local: type PTR, class IN, "QM" question
        ...
    Answers
    Additional records
```
Which queried (among other things) `_hap._tcp` protocols.
Interestingly this packet is from `192.168.1.204`, i.e. `Florian’s MacBook Pro`, so not the querier (`192.168.1.208`), a.k.a. `Florian’s MacBook Pro NW`!

This packet should be answered by the ESP device, but it wasn't.
I tried to *replay* this same packet on my local network to verify if it was actually answered and it was.
(here's this the replay script 
[test.py.txt](https://github.com/espressif/esp-idf/files/7293323/test.py.txt) for completeness)

Let's assume that this packet didn't reach the device (for some reason, blocked/discarded by the router, the third MacBook a different segment, use of DNSSEC?). Could you please enable mdns debug prints, so we can prove this theory?
Please uncomment this line:

https://github.com/espressif/esp-idf/blob/3e370c4296247b349aa3b9a0076c05b9946d47dc/components/mdns/private_include/mdns_private.h#L22

And share the debug log from the device. It should print something like:
```
RX[0][0]: From: 192.168.1.204:5353, To: 224.0.0.251, Packet[18470]: 
    Q: ._uscans._tcp.local. PTR IN
    Q: ._uscan._tcp.local. PTR IN
    Q: ._ippusb._tcp.local. PTR IN
    Q: ._ipp._tcp.local. PTR IN
    Q: ._scanner._tcp.local. PTR IN
    Q: ._ipps._tcp.local. PTR IN
    Q: ._printer._tcp.local. PTR IN
    Q: ._pdl-datastream._tcp.local. PTR IN
    Q: ._ptp._tcp.local. PTR IN
    Q: ._hap._tcp.local. PTR IN
    Q: ._homekit._tcp.local. PTR IN
    Q: ._airport._tcp.local. PTR IN
    Q: ._rdlink._tcp.local. PTR IN
    Q: ._hap._udp.local. PTR IN
    Q: ._apple-mobdev2._tcp.local. PTR IN
    Q: ._remotepairing-manual-pairing._tcp.local. PTR IN
    Q: 344C68EC-CBAC-5517-BB60-F9608F9AA3D8._homekit._tcp.local. TXT IN
    Q: 77FA02A5-97B4-5FFE-B041-ED242B960329._homekit._tcp.local. TXT IN
    Q: ._googlecast._tcp.local. PTR IN
    Q: ._raop._tcp.local. PTR IN
    Q: ._airplay._tcp.local. PTR IN
    Q: ._raop._tcp.local. PTR IN
    Q: ._airplay._tcp.local. PTR IN
    Q: ._afpovertcp._tcp.local. PTR IN
    Q: ._smb._tcp.local. PTR IN
    Q: ._rfb._tcp.local. PTR IN
    Q: ._adisk._tcp.local. PTR IN
    Q: ._sleep-proxy._udp.local. PTR IN
    Q: ._companion-link._tcp.local. PTR IN
    A: _companion-link._tcp.local. PTR IN 4500 [29] Florian’s MacBook Pro NW._companion-link._tcp.local.
    A: _companion-link._tcp.local. PTR IN 4500 [26] Florian’s MacBook Pro._companion-link._tcp.local.
    X: . OPT 05A0 4500 [12]  Code: 0004 Data[8]: 00 3c f4 0f 24 1e 5d 95
```
and is immediately answered with:
```
TX[0][0]: To: 224.0.0.251:5353, Packet[18640]: AUTHORITATIVE
    A: _hap._tcp.local. PTR IN 120 [18] ESP32-WebServer._hap._tcp.local.
    A: ESP32-WebServer._hap._tcp.local. SRV IN FLUSH 120 [19] 0 0 80 esp32-mdns.local.
    A: ESP32-WebServer._hap._tcp.local. TXT IN FLUSH 120 [44] path=/foobar; p=password; u=admin; board=esp32
    X: esp32-mdns.local. A IN FLUSH 120 [4] 192.168.10.101
```

The above results should be also need if replaying the test script shared above.

albrownwood, 2021-10-07:
Hi @david-cermak,
thanks, here is a device log with debug enabled.

`
W (4969) wifi:<ba-add>idx:1 (ifx:0, 0c:7c:28:e3:3c:d4), tid:0, ssn:0, winSize:64
I (4989) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (6539) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (6539) esp_netif_handlers: example_connect: sta ip: 192.168.1.223, mask: 255.255.255.0, gw: 192.168.1.254
I (6549) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.1.223
I (6559) example_connect: Connected to example_connect: sta
I (6569) example_connect: - IPv4 address: 192.168.1.223
I (6569) example_connect: - IPv6 address: fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (6589) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0

TX[0][0]: To: 224.0.0.251:5353, Packet[6220]:
    Q: *U* ESP32-WebServer._hap._tcp.local. ANY IN
    Q: *U* esp32-mdns...local. ANY IN
    Q: *U* esp32-mdns...local. ANY IN
    S: ESP32-WebServer._hap._tcp.local. SRV IN 120 [24] 0 0 52345 esp32-mdns.local.
    S: esp32-mdns.local. A IN 120 [4] 192.168.1.223
    S: esp32-mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[6320]:
    Q: *U* ESP32-WebServer._hap._tcp.local. ANY IN
    Q: *U* esp32-mdns...local. ANY IN
    Q: *U* esp32-mdns...local. ANY IN
    S: ESP32-WebServer._hap._tcp.local. SRV IN 120 [19] 0 0 52345 esp32-mdns.local.
    S: esp32-mdns.local. A IN 120 [4] 192.168.1.223
    S: esp32-mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c

TX[0][0]: To: 224.0.0.251:5353, Packet[6520]:
    Q: ESP32-WebServer._hap._tcp.local. ANY IN
    Q: esp32-mdns...local. ANY IN
    Q: esp32-mdns...local. ANY IN
    S: ESP32-WebServer._hap._tcp.local. SRV IN 120 [19] 0 0 52345 esp32-mdns.local.
    S: esp32-mdns.local. A IN 120 [4] 192.168.1.223
    S: esp32-mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[6620]:
    Q: ESP32-WebServer._hap._tcp.local. ANY IN
    Q: esp32-mdns...local. ANY IN
    Q: esp32-mdns...local. ANY IN
    S: ESP32-WebServer._hap._tcp.local. SRV IN 120 [8] 0 0 52345 esp32-mdns.local.
    S: esp32-mdns.local. A IN 120 [4] 192.168.1.223
    S: esp32-mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c

TX[0][0]: To: 224.0.0.251:5353, Packet[6820]:
    Q: ESP32-WebServer._hap._tcp.local. ANY IN
    Q: esp32-mdns...local. ANY IN
    Q: esp32-mdns...local. ANY IN
    S: ESP32-WebServer._hap._tcp.local. SRV IN 120 [19] 0 0 52345 esp32-mdns.local.
    S: esp32-mdns.local. A IN 120 [4] 192.168.1.223
    S: esp32-mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[6920]:
    Q: ESP32-WebServer._hap._tcp.local. ANY IN
    Q: esp32-mdns...local. ANY IN
    Q: esp32-mdns...local. ANY IN
    S: ESP32-WebServer._hap._tcp.local. SRV IN 120 [19] 0 0 52345 esp32-mdns.local.
    S: esp32-mdns.local. A IN 120 [4] 192.168.1.223
    S: esp32-mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c

TX[0][0]: To: 224.0.0.251:5353, Packet[7120]: AUTHORITATIVE
    A: _services._dns-sd._udp.local. PTR IN 4500 [17] ._hap._tcp.local.
    A: _hap._tcp.local. PTR IN 4500 [23] ESP32-WebServer._hap._tcp.local.
    A: ESP32-WebServer._hap._tcp.local. SRV IN FLUSH 120 [24] 0 0 52345 esp32-mdns.local.
    A: ESP32-WebServer._hap._tcp.local. TXT IN FLUSH 4500 [44] path=/foobar; p=password; u=admin; board=esp32
    A: esp32-mdns.local. A IN FLUSH 120 [4] 192.168.1.223
    A: esp32-mdns.local. AAAA IN FLUSH 120 [16] fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c

[... 28 lines cut ...]
    A: ESP32-WebServer._hap._tcp.local. TXT IN FLUSH 4500 [44] path=/foobar; p=password; u=admin; board=esp32
    A: esp32-mdns.local. A IN FLUSH 120 [4] 192.168.1.223
    A: esp32-mdns.local. AAAA IN FLUSH 120 [16] fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[9420]: AUTHORITATIVE
    A: _services._dns-sd._udp.local. PTR IN 4500 [17] ._hap._tcp.local.
    A: _hap._tcp.local. PTR IN 4500 [23] ESP32-WebServer._hap._tcp.local.
    A: ESP32-WebServer._hap._tcp.local. SRV IN FLUSH 120 [24] 0 0 52345 esp32-mdns.local.
    A: ESP32-WebServer._hap._tcp.local. TXT IN FLUSH 4500 [44] path=/foobar; p=password; u=admin; board=esp32
    A: esp32-mdns.local. A IN FLUSH 120 [4] 192.168.1.223
    A: esp32-mdns.local. AAAA IN FLUSH 120 [16] fe80:0000:0000:0000:9e9c:1fff:fec9:4e2c
`

`
➜  ~ dns-sd -B _hap._tcp
Browsing for _hap._tcp
DATE: ---Wed 06 Oct 2021---
20:27:10.542  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
20:27:51.666  Add        2   5 local.               _hap._tcp.           ESP32-WebServer
20:32:39.521  Rmv        0   5 local.               _hap._tcp.           ESP32-WebServer
`

Also attaching the wireshark log.
[capt3.pcapng 2.zip](https://github.com/espressif/esp-idf/files/7298523/capt3.pcapng.2.zip)

richardbecht, 2021-10-11:
Hello,

I have a similar problem but on the device side there is RX package but it log an mdns error. ESP-IDF version 4.3.1.

```
RX[0][0]: From: 192.168.1.218:5353, To: 224.0.0.251, Packet[11480450]: 
    Q: lb._dns-sd._udp.local. PTR IN
    Q: ._companion-link._tcp.local. PTR IN
    Q: ._hap._tcp.local. PTR IN
    Q: ._homekit._tcp.local. PTR IN
    Q: DC7BDC64-B822-5F48-9423-82E72836DB02._homekit._tcp.local. TXT IN
    Q: ._hap._udp.local. PTR IN
    Q: 70-35-60-63.1 Living Room (2)._sleep-proxy._udp.local. TXT IN
    Q: ._sleep-proxy._udp.local. PTR IN
ERROR: /Users/richard/esp32/esp-idf-v4.3.1/components/mdns/mdns.c:4930

```

I logged the variables inside the _mdns_parse_fqdn function:

name->host = "openthread.thread.home.arpa"
name->parts = 1
name->domain = empty string

If the client sends an RX that contains openthread.thread.home.arpa NS type query then esp side mdns log error and does not send TX. 
In line 2437 of the mdns.c file, the condition is false (domain is emtpy string), so the _mdns_parse_fqdn function returns 0.
I made a quick and dirty fix in the _mdns_read_fqdn (mdns.c line 252.) function so there are no more errors in the log.

```
...
            if (name->parts == 1 && buf[0] != '_'
                    && (strcasecmp(buf, MDNS_DEFAULT_DOMAIN) != 0)
                    && (strcasecmp(buf, "ip6") != 0)
                    && (strcasecmp(buf, "in-addr") != 0)
                    && (strcasecmp(buf, "arpa") != 0)) { // fix
                strlcat(name->host, ".", sizeof(name->host));
                strlcat(name->host, buf, sizeof(name->host));
            } else if (strcasecmp(buf, MDNS_SUB_STR) == 0) {
...

```

The request containing openthread.thread.home.arpa comes from my iPad, but I don't know why.
The _hap._tcp is now stable, there is no remove on client side, tested with 'dns-sd -B _hap._tcp' on macos.

Sorry for my bad English.

albrownwood, 2021-10-12:
Hi @david-cermak,

> Could you please quickly test if you experience the same behavior on IDF v4.2 ?
I tested with latest commit on branch 4.2 but I see the same behavior as with 4.4
> 
> > My observation is that the Macbook on 2.4Ghz radio never receives the mDNS broadcast packets when the esp comes online whereas the Macbook on 5Ghz receives them.
> 
> This is weird. I'd expect the clients on 5G to receive ESP multicasts correctly (the same way as the ESP receives only the multicasts on 2.4G), assuming they are on the same subnet. My understanding was that the router didn't forward packets for _pending_ clients.
The 5Ghz clients seem to be receiving multicast packets correctly but the 2.4Ghz clients (whether it's an esp or a Macbook) do not receive them.
> 
> > What do you think should be the next step with AT&T?
> 
> We are in the process of purchasing the device to analyse the problem (already have a lot of different brands of routers/switches, but none of this `BGW320` product family)
> 
sounds good, please let me know when you are able to try on your side.
> Maybe, as a workaround, you can try to setup a static IP and reserve this address in the router for Espressif devices.
in the meantime we will continue development of our device using other router brands we have.

david-cermak, 2021-10-13:
@richardbecht Thanks for explaining, understand now. It looks like there's a problem with parsing the FQDN in the first place, as the correct domain should be `arpa`, might have something to do with service sub-types, which are not currently supported.

This is unrelated to #7638 issue, so could you please create a separate issue for this problem? Also, would be great if could you attach a wireshark log with the packet from your iPad, so we can easily test it.

richardbecht, 2021-10-14:
@david-cermak Thanks for the help. I created a separate issue #7694 .

david-cermak, 2021-11-22:
Hi @albrownwood 

We have finally gotten the exact device, BGW320-505. I've quickly tested the basic scenario of connecting the Espressif board to 2.4GHz and my MBP to 5GHz and it seems that all multicasts were correctly propagated and the service appeared to be up and running.
Also, the DHCP seems to work correctly, the ESP device was connected, haven't seen any "Pending allocation" note:
[image]

Did you use any kind of non-standard settings? Anything that differs from the defaults?

albrownwood, 2021-11-22:
Hi @david-cermak,
no non-standard settings or configuration, we use the default as sent by ATT.
Also did you try after letting the router run for 1/2 days, some people on ATT forums seem to see this appear after some runtime.
What firmware version is installed on that router?

albrownwood, 2021-11-23:
Hi @david-cermak,
the DHCP issue was fairly repeatable. How can I capture the DHCP packets between the router and the ESP, any tool suggestion? Also we currently configured this router in passthrough mode so we only use it as a modem with another wifi router behind it due to the mdns issues, so I will have to reconfigure it as router to capture packets.

The firmware issue where these issues were seen is 2.14.4, are you able to upgrade to this version?

david-cermak, 2021-11-25:
Hi @albrownwood 

> How can I capture the DHCP packets between the router and the ESP

You can capture the air packet in wireshark, see https://wiki.wireshark.org/CaptureSetup/WLAN (and then share the connection token, or you can simply leave the WiFi open for the capturing only)

> The firmware issue where these issues were seen is 2.14.4, are you able to upgrade to this version?

This is appears to be quite difficult, as I'm not a AT&T customer, so not *eligible* for the firmware updates. But trying...

david-cermak, 2022-04-01:
@albrownwood  Any update with capturing the packets?

I'm sorry, I'm not able to update the firmware of the device that I purchased and wasn't able to reproduce any kind of mDNS issue, neither the problems with the DHCP packets you observed. Everything seems to work as expected with my device (firmware version: 2.6.5).

The only way to address this problem is to rely on your debug data and logs, and the packet capture.

closed 2022-05-10 by Alvin1Zhang