espressif/esp-idf #7639: Can't change value of NIMBLE parameter BLE_GATT_MAX_PROCS

JonasAxelsson, 2021-10-04:
## Environment

Development Kit: none
Module or chip used: ESP32-WROVER
IDF version (run git describe --tags to find it):
C:\repo\ble_test\dicm\esp32>git describe --tags
HMI_Rubicon_v1.4.0-12-g0d53d8d4
Build System: CMake
Compiler version (run xtensa-esp32-elf-gcc --version to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: Windows
(Windows only) environment type: Plain Command Prompt
Using an IDE?: No
Power Supply: USB

## Problem Description

It is not possible to set parameter MYNEWT_VAL_BLE_GATT_MAX_PROCS in the config tool.
I use ESP-IDF version 4.2.2

### Expected Behavior
Should be possible to alter the NIMBLE config parameter MYNEWT_VAL_BLE_GATT_MAX_PROCS 

### Actual Behavior
It is not possible to change parameter MYNEWT_VAL_BLE_GATT_MAX_PROCS  without altering an IDF file which means that we need to fork the IDF if we want to change the parameter.

The parameter is set in this file:
\components\bt\host\nimble\port\include\esp_nimble_cfg.h

### Steps to reproduce

### Code to reproduce this issue

## Debug Logs

## Other items if possible

rahult-github, 2021-10-04:
Hi @JonasAxelsson ,

Currently, the options exposed via menuconfig that user can select to configure were primarily decided based on common use cases and also exposing too many options may lead  to confusion. However, will revisit the list and expose such options for user to configure. 

Thanks,
 Rahul

rahult-github, 2022-09-20:
As part of new release versions, BLE_GATT_MAX_PROCS is included to be configured via menuconfig. Closing this issue.

closed 2022-09-20 by rahult-github

JonasAxelsson, 2022-11-16:
In which version of the IDF is the fix included?

JonasAxelsson, 2022-11-16:
> As part of new release versions, BLE_GATT_MAX_PROCS is included to be configured via menuconfig. Closing this issue.

In which version of the IDF is the fix included?