espressif/esp-idf #7645: esp蓝牙wifi同时使用+smartconfig 华为手机

fghjhuang, 2021-10-05:
1. 我的应用如下，
1）没有记录ssid和password情况下，初始化ble，通过ble配网，作为ble从机，手机做主机，给特定uuid发送数据进行配网
2）记录了ssid情况下，直接联网，不初始化ble
实际情况：很容易功耗过大，导致复位重启（是因为打开蓝牙的原因吗？），想知道使能蓝牙功能是否还需要设置什么东西？记录ssid的时候虽然没有初始化ble，但是menuconfig还是有配置蓝牙协议栈的，是否因此导致功耗过大？

2. 直接不使用ble配网，使用esp touch配网，但是使用华为手机安装esp touch app进行配网的时候发现配网失败，其他手机使用esp touch app是成功的，原因不明。

jsw-davidhuang, 2021-10-08:
关于第2点： 原来是我用smartconfig的时候忘记进行esp wifi start了，第二点close

xiewenxiang, 2021-10-11:
@fghjhuang  如何判断复位的原因是功耗过高引起的？ 有复位的日志吗？ 麻烦提供一下完整的日志.

jsw-davidhuang, 2021-10-11:
> @fghjhuang 如何判断复位的原因是功耗过高引起的？ 有复位的日志吗？ 麻烦提供一下完整的日志.

log里面最后一行是：Brownout detector was triggered
所以我查了百度发现是提示低电触发了重启

xiewenxiang, 2021-10-11:
模组是使用的espressif的官方模组还是自己设计的？电源供电能力多大呢？

jsw-davidhuang, 2021-10-11:
> 模组是使用的espressif的官方模组还是自己设计的？电源供电能力多大呢？

模组使用的是nodemcu，模块用的是乐鑫的esp-wroom-32，供电是用电脑的usb直接供电

jack0c, 2022-05-23:
@jsw-davidhuang 问题1应该是电源问题，电压不稳会导致browout reset

jack0c, 2022-05-23:
@jsw-davidhuang 我们会重新一下问题2

jsw-davidhuang, 2022-05-23:
> @jsw-davidhuang 问题1应该是电源问题，电压不稳会导致browout reset

嗯，是的，我用适配器供电就没有问题了

zhangyanjiaoesp, 2022-08-01:
@jsw-davidhuang please provide your IDF commit, we will reproduce it

closed 2023-05-26 by Alvin1Zhang