espressif/esp-idf #7646: dpp-enrollee hangs before getting IP address

mccleanp, 2021-10-05:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    v4.3.1-29-ge493a4c30e
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:    Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

After successful DPP authentication, the wifi connects but then appears to hang and the IP_EVENT_STA_GOT_IP event is never triggered.

### Expected Behavior

wifi should connect and trigger the IP_EVENT_STA_GOT_IP event

### Actual Behavior

code appears to hang and the IP_EVENT_STA_GOT_IP event is never 

### Steps to reproduce

1. Build, flash and monitor dpp-enrollee example
2. dpp-enrollee example code prints the QR code to console.
3. Use Android phone to add device to network using QR code.
4. dpp-enrollee example code indicates DPP authentication was successful and prints correct SSID
5. wifi connects but then it appears to hang
6. The IP_EVENT_STA_GOT_IP event is not triggered.

## Debug Logs

```
I (27) boot: ESP-IDF v4.3.1-29-ge493a4c30e 2nd stage bootloader
I (27) boot: compile time 20:38:24
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 3, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1654ch ( 91468) map
I (144) esp_image: segment 1: paddr=00026574 vaddr=3ffb0000 size=04150h ( 16720) load
I (151) esp_image: segment 2: paddr=0002a6cc vaddr=40080000 size=0594ch ( 22860) load
I (160) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=7faach (522924) map
I (350) esp_image: segment 4: paddr=000afad4 vaddr=4008594c size=0f1cch ( 61900) load
I (375) esp_image: segment 5: paddr=000beca8 vaddr=50000000 size=00010h (    16) load
I (386) boot: Loaded app from partition at offset 0x10000
I (386) boot: Disabling RNG early entropy source...
I (398) cpu_start: Pro cpu up.
I (398) cpu_start: Starting app cpu, entry point is 0x40081220
0x40081220: call_start_cpu1 at /home/mccleanp/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (412) cpu_start: Pro cpu start user code
I (412) cpu_start: cpu freq: 160000000
I (412) cpu_start: Application information:
I (417) cpu_start: Project name:     dpp-enrollee
I (422) cpu_start: App version:      v4.3.1-29-ge493a4c30e
I (428) cpu_start: Compile time:     Oct  5 2021 20:39:13
I (434) cpu_start: ELF file SHA256:  0d4d0d8964075842...
I (440) cpu_start: ESP-IDF:          v4.3.1-29-ge493a4c30e
I (447) heap_init: Initializing. RAM available for dynamic allocation:
I (454) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (460) heap_init: At 3FFB8008 len 00027FF8 (159 KiB): DRAM
I (466) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (472) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (479) heap_init: At 40094B18 len 0000B4E8 (45 KiB): IRAM
I (486) spi_flash: detected chip: generic
I (490) spi_flash: flash io: dio
W (493) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (508) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (608) wifi:wifi driver task: 3ffc1cec, prio:23, stack:6656, core=0
I (608) system_api: Base MAC address is not set
I (608) system_api: read default base MAC address from EFUSE
I (628) wifi:wifi firmware version: 88c8747
I (628) wifi:wifi certification version: v7.0
I (628) wifi:config NVS flash: enabled
I (628) wifi:config nano formating: disabled
I (638) wifi:Init data frame dynamic rx buffer num: 32
I (638) wifi:Init management frame dynamic rx buffer num: 32
I (648) wifi:Init management short buffer num: 32
I (648) wifi:Init dynamic tx buffer num: 32
I (658) wifi:Init static rx buffer size: 1600
I (658) wifi:Init static rx buffer num: 10
I (668) wifi:Init dynamic rx buffer num: 32
I (668) wifi_init: rx ba win: 6
I (668) wifi_init: tcpip mbox: 32
I (678) wifi_init: udp mbox: 6
I (678) wifi_init: tcp mbox: 6
I (678) wifi_init: tcp tx win: 5744
I (688) wifi_init: tcp rx win: 5744
I (688) wifi_init: tcp mss: 1440
I (698) wifi_init: WiFi IRAM OP enabled
I (698) wifi_init: WiFi RX IRAM OP enabled
I (1068) wifi dpp-enrollee: Scan below QR Code to configure the enrollee:

I (1068) QRCODE: Encoding below text with ECC LVL 0 & QR Code Version 10
I (1068) QRCODE: DPP:C:81/6;M:94:b9:7e:fa:57:f4;K:MDkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDIgADDGOOgEz1kB2vB8iwOvQd3uGrPjOU7z+WbV0u2tV+l10=;;
                                              
  █▀▀▀▀▀█ █▄█▄▄ ▀▀▀█▀█▄▄▀▀  ▀▀▀█ ▄▀ █▀▀▀▀▀█   
  █ ███ █ █  ▀▄▀▄ ▀▄▄  ▀▀██ █ █▀█▄  █ ███ █   
  █ ▀▀▀ █ ▄▄▀██ █ ▄▄██▀▄█▀▀▀  ▀█▄█  █ ▀▀▀ █   
  ▀▀▀▀▀▀▀ ▀ ▀ █ ▀▄█▄█▄▀▄▀▄█ ▀ █▄▀▄█ ▀▀▀▀▀▀▀   
  ▀▀ ▄█▀▀ ▄█▀ █▄▀▄█ ▄ ▄ █ ▀▄▀█▀▀▀▀▄▄▄▀▄██▀▀   
   ▀ ▀ ▀▀ ▄ █ ▀ ▄█▄█▄▄ █▄▀█ █▀ ▄▀▄ ▀ ▀▀  ▄█   
  █▀▀█▄▄▀█▀█▀ ▀█▀▀▄ ▀ ▄▄█▄ ▄▀▄▄  ▀ ▀ ▄▄ ▄█    
  ▀▀▄▀ ▀▀   ▀ ▄   ▀█▄▄▀█▀█  ▀██▄█ ▀▀ ▄ █▀▄█   
   ▄█▄▄ ▀▀█▀▀█▀▀▄▀▄▄  ▀ ▀▄ ▄▀▄▀▀▄ ▀▀ ▀▄█▀     
  █▄█▀ ▀▀▄▄▄██▀█▀▀█ ▀▄ ▄▄▀▀▀██ █▄▄ █▀ █▄▀▄█   
  █ ▀██ ▀▀▄█ ▀▄█▄▀▀ ▄ ▄█ █▄   █▀▄ ▀▄▄▀▄▀▀█    
  █▀▄  █▀▀ ▄▀█▀ ▀█▄█▄ ███▀▄▀▄▀███   ▄███▀▄█   
  ▄▄▄▀▀█▀  ▀█ ▀█▄ █    ▄▀▄▀▀▄▄▀▀▀ ▄▄▄ ▄██▀    
  █▄████▀█▀▄██▄ ▄▀ █▄ ▀▄▀█▄▀ ▀██▀ ▄█▀▄ ▀▀ █   
  ▄ ██▀▀▀  ▀█ ▀█▀▀▄▄▀ ▄█▀   ████▀▀▄█▄█▄       
     █ █▀███▄ ▄ ▄ ▀█▄  ▄▀▀ ▀▄▀ ▄ ▄ █   █▀ ▀   
  ▀▀  ▀▀▀ ▄██▀▀▀ █▄▄▄█▀█▀ ▀ ▀▄▀▀▄▀█▀▀▀██      
  █▀▀▀▀▀█ ▄▀ ▀▀█▀▀▄ ▀▄ █▄██▀▄█▄▄▄▀█ ▀ █ ▀ ▀   
  █ ███ █ ▀ ▀ ▄█▄▀▀▄▄ ▄█▄█▀▀  ▄▀ ▀█▀▀█▀ ▄█▄   
  █ ▀▀▀ █ ▄ ▄█▀ ▄█▄█▄ ▀▄ ▀█▀ █ ▄▀ ▀ ▄█ █▀▄█   
  ▀▀▀▀▀▀▀ ▀▀  ▀      ▀  ▀ ▀▀   ▀  ▀ ▀  ▀▀▀    
                                              

I (1398) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (1398) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
I (1608) wifi:mode : sta (94:b9:7e:fa:57:f4)
I (1608) wifi:enable tsf
I (1618) wifi dpp-enrollee: Started listening for DPP Authentication
I (39768) wifi dpp-enrollee: DPP Authentication successful, connecting to AP : redacted
I (39778) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (39778) wifi:state: init -> auth (b0)
I (39788) wifi:state: auth -> assoc (0)
I (39798) wifi:state: assoc -> run (10)
I (39828) wifi:connected with 40lisginny, aid = 2, channel 6, BW20, bssid = e0:63:da:07:5a:8a
I (39828) wifi:security: WPA2-PSK, phy: bgn, rssi: -49
I (39838) wifi:pm start, type: 1

I (39868) wifi:AP's beacon interval = 102400 us, DTIM period = 3

```

jkent, 2021-11-09:
I reproduce this problem as well.

Xtensa2C, 2021-11-16:
I can also confirm this issue.

closed 2022-01-07 by espressif-bot

Alvin1Zhang, 2022-01-07:
Thanks for reporting, we are also back porting the fix to release/4.3 and release/4.4.

jobo89, 2022-02-01:
I had the same issue and needed to go to:
`v5.0-dev-1350-g41761fc762`
for the example to get an IP address (I don't remember if v4.4-rc1 worked or not). However, when that happened, I now get a 4 MHz wave on the GPIO 0 pin. Is this from the fix or always there but now just reaching that part of the code?
[image]

nachiketkukade, 2022-02-07:
@jobo89 , the fix was purely software based and DPP specific. So I'm sure this has existed before and the execution is reaching that part of code now.