espressif/esp-idf #7647: WiFi Provisioning Bluedroid iOS Bug

divrajBevie, 2021-10-05:
## Environment

- Module or chip used:  ESP32-WROOM-32E with 16MB Flash
- IDF version v4.4
- Build System:         [CMAKE]
- Operating System:     [macOS]
- Using an IDE?: Yes VsCode extension
- Power Supply:         USB 5V

## Problem Description

When using the bluedroid stack with the WiFi Provisioning example, the local/advertised name does not get set. However NimBLE is working fine.

There are two scenarios where this can be observed. If your phone has never connected to the device, the name is advertised as NULL. However, if you previously connected to the ESP (paired) the phone caches the GAP name as the advertised. Changing the name in future firmware builds doesnt update the Advertised name until you pair/connect to it again.

With NimBLE stack the expected result occurs, that is , when a new iPhone tries to search for the device the advertised name is actually already set as the GAP name and any changes to the service name automatically sets the the new advertised name.

NOTE: with Android this is working fine and for more information regarding advertisedName and GAP name for IOS it can be seen on this article [https://developer.apple.com/forums/thread/72343](link).

NOTE2: I believe the Service Changed Characteristic is not being employed on the Bluedroid stack and I cannot find a way to trigger this.

### Steps to reproduce

1. Download nrf Connect app ON an **iPHONE** (Used to scan for devices)
2. Build the WiFi manager provisioning app using the nimble stack
3. Scan using nrf, Note that the name is equal to the one set via the firmware (i.e. PROV_)
4. Change the PROV_ value to something else and repeat step 3 with new build. 
5. Note the expected behavior.
6. Now change the nimble stack to Bluedroid and repeat steps 3-5
7. At steps 5 you will notice that the name does not change from the previous build.

sch0bert, 2022-03-02:
Hello,

I have the same problem in my application. I migrated my IDF from v4.0.3 to v4.4 and this problem began. I tested it with release version idf 4.4 by configuring my device using the protocomm example in the wiki page:

https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/provisioning/protocomm.html

Is not possible to see my esp32 with any iphone from 8 to above (haven't tested with previews versions). I believe Protocomm is using the bluedroid stack also. I already test it using the ios ESP Provisioning and Rainmaker app with the same result.

I see that this problem was originally reportes in the [ESP Provisioning ios project](https://github.com/espressif/esp-idf-provisioning-ios/issues/35) since Jul 2021 and haven't been fixed. **This is a very important matter and can not wait longer**, I will appreciated if you guys can take a look on this @igrr @projectgus @jack0c

jack0c, 2022-03-09:
@sch0bert Sorry for the delay. Looks like assigned to the wrong engineer internally. Will fix this soon.

jack0c, 2022-03-09:
It looks like a bug on Bluedroid. @xiewenxiang please help to fix this bug.

xiewenxiang, 2022-03-09:
Hi @divrajBevie @sch0bert ,
Could you please test with this patch?
[0001-test-phone-can-t-discovery-the-device-name-character.zip](https://github.com/espressif/esp-idf/files/8214122/0001-test-phone-can-t-discovery-the-device-name-character.zip)

sch0bert, 2022-03-09:
@xiewenxiang could you provide the instructions to add the .patch file to the idf?

xiewenxiang, 2022-03-10:
https://git-scm.com/docs/git-apply

```c
git apply --stat patchfile

git apply --check patchfile

git apply patchfile
```

sch0bert, 2022-03-15:
Hello @xiewenxiang, I applied the patch you provided and tested again but seems is not working for me. I still have the same problem. Just to double check, is this the only change (the right one is the modification) from the patch in file components/bt/host/bluedroid/stack/gatt/gatt_attr.c?

[image]

xiewenxiang, 2022-03-16:
@sch0bert  Yes, the modification is right. I can find the device name and Service Changed Characteristic based on this patch.
[image]

sch0bert, 2022-03-16:
@divrajBevie could you also check if the patch works on your side?

@xiewenxiang I just tried with different iphones (12, 11 & 9). In the cases where I paired before the device, I was able to find it again but in the ones that I didn't, is not able to see the device. Did you tried with a different cellphone?

sch0bert, 2022-04-04:
@KonssnoK try to make that little change that @xiewenxiang made a few of weeks ago inside your IDF components/bt/host/bluedroid/stack/gatt/gatt_attr.c

and tell us if that fixed your issue, it didn't in my case :(

KonssnoK, 2022-04-04:
> @KonssnoK try to make that little change that @xiewenxiang made a few of weeks ago inside your IDF components/bt/host/bluedroid/stack/gatt/gatt_attr.c
> 
> and tell us if that fixed your issue, it didn't in my case :(

Actually that change is already part of the 4.4 branch i used, so no, it didn't work.
[image]

Another clue could be that it's not working only on the S3 (i don't know which chipset are you using), but it works on the C3.
Since the two chipsets use the same library it should pin down the issue to either the proprietary lib (which we cannot check) or the differences (which i looked up and didn't seem so many).

KonssnoK, 2022-04-04:
Then it's even stranger, because ESP32 uses BL4.2, not BLE 5.0 :D
It means the issue is related to something shared by the xtensa architecture 🤷🏼‍♂️

sch0bert, 2022-04-04:
@xiewenxiang have you had a chance to look at this issue?

KonssnoK, 2022-04-04:
I think I'll open a commercial issue and then try to understand if i can get a commercial support for my company :) 
I'll update here when i have news.

KonssnoK, 2022-04-05:
I did some more testing on the issue i opened. It's exactly the same problem, but the provisioning fails anyway after trying to pair with the "wrong" name.

KonssnoK, 2022-04-06:
@sch0bert i was able to find the underlying issue, you can read the other issue and probably make a patch (or wait for one)

sch0bert, 2022-04-06:
This is great @KonssnoK! Thanks for sharing, I will try your fix later and give my feedback.

Alvin1Zhang, 2022-05-16:
Thanks for reporting, and sorry for slow turnaround, fix on master branch is available at https://github.com/espressif/esp-idf/commit/f7f6a929f2608cbcb67eafa240369ec4987971d1, we are back porting the fix to release/4.4 and release/4.3. Thanks.

sch0bert, 2022-05-18:
Please let us know when can we pull the fix in 4.4 @Alvin1Zhang

Alvin1Zhang, 2022-07-26:
Thanks for reporting, sorry for slow turnaround, fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/74dd39d8b07fc64abc69a4094af5745d4d846096, feel free to reopen.

closed 2022-07-26 by Alvin1Zhang