espressif/esp-idf #7648: Device Boot Error: Getting Flash Read Error : 1000

Vbansal1996, 2021-10-06:
Hello Community Members:

I am using the particular commit id of esp-idf : c40f2590bf759ff60ef122afa79b4ec04e7633d2
for my application.

the device is placed at the client side and we are getting the issue that the device is not able to load the bootloader after 15 days, When we checked the logs we are getting this:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
 
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
 
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

I had checked the efuse summary but found nothing suspicious:

Detecting chip type... ESP32
espefuse.py v3.0

> EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
> ----------------------------------------------------------------------------------------
> Calibration fuses:
> BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
> ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1121 R/W (0b00011)
>  
> Config fuses:
> XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
> XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
> XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
> CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 50 R/W (0x32)
> SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
> SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
> SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
> SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
> SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
> DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)
>  
> Efuse fuses:
> WR_DIS (BLOCK0):                         Efuse write disable mask                           = 0 R/W (0x0000)
> RD_DIS (BLOCK0):                         Efuse read disable mask                            = 0 R/W (0x0)
> CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme                
>    = NONE (BLK1-3 len=256 bits) R/W (0b00)
> KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)
>  
> Identity fuses:
> MAC (BLOCK0):                            Factory MAC Address                               
>    = a8:03:2a:d4:21:90 (CRC 0x4b OK) R/W 
> MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 75 R/W (0x4b)
> CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
> CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = False R/W (0b0)
> CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
> CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 1 R/W (0b001)
> MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)
>  
> Security fuses:
> FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 0 R/W (0b0000000)
> UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
> FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 0 R/W (0x0)
> CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
> ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
> ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
> JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = False R/W (0b0)
> DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
> DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = False R/W (0b0)
> DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = False R/W (0b0)

BLOCK1 (BLOCK1):                         Flash encryption key                              
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLOCK2 (BLOCK2):                         Secure boot key                                   
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLOCK3 (BLOCK3):                         Variable Block 3                                  
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
 
Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

Can anyone suggest what is the exact issue, why device boot fails after some power cycles.

negativekelvin, 2021-10-06:
You should dump the flash and compare it to a good image.

Vbansal1996, 2021-10-07:
I tried the same but it seems like device is not able to read any data from the flash. command used:
esptool.py read_flash 0x1000 0x8000 ./dump.bin

Here I am trying to read the application bootloader. But there is no response in return.

Does it means that flash got damaged?

closed 2023-03-27 by Alvin1Zhang