espressif/esp-idf #7652: Issue on WiFi handling on wakeup from sleep

DimitriLubenski, 2021-10-06:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (v4.3.1):
- Build System:         [idf.py]
- Operating System:     [Windows]
- Using an IDE?: [Yes (VSCODE)]
- Power Supply:         [USB]

## Problem Description

I am using the ESP32-WROOM-32 and Waveshare E-Ink Screen to implement the following scenario:

- The ESP wakes up 
- The WiFi connection is established to reach a local home WiFi access point
- The data is fetched via HTTP from a local server
- The controller goes to sleep for the next 10 minutes

I would like to be able to recognize the first boot for initialization of E-Ink.

To figure out if the controller performs the initial boot or wakes up from the sleep I am using the following code, which works well. On the first boot I get the value “0”, on subsequent “4” as a result of esp_sleep_get_wakeup_cause().

### Code to reproduce this issue

```cpp
#include "sdkconfig.h"
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "wifi.h"
#include "esp_wifi.h"
#include "soc/rtc.h"
#include <rom/rtc.h>
#include "esp_sleep.h"
#include <string> 

extern "C" {
	void app_main(void);
}

void delay(uint32_t millis) { vTaskDelay(millis / portTICK_PERIOD_MS); }

void app_main(void)
{
      printf("Cause ===========> %d \n", esp_sleep_get_wakeup_cause()) ;

      Wifi wifi;      
      //wifi.connect(); //ISSUE if uncommented!

      
      esp_sleep_enable_timer_wakeup(20ULL * 1000 * 1000);
      esp_deep_sleep_start();      
}
```

If I add to this flow the WiFi connectivity code, I am getting "two starts" (see thetrace below) on each wake up. 

Here is the class used to perfrom the WiFi connectivity - I am relying on standard examples. 

```cpp
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_http_client.h"
#include "nvs_flash.h"
#include <string>
[... 325 lines cut ...]
I (724) wifi_init: tcp mbox: 6
I (724) wifi_init: tcp tx win: 5744
I (734) wifi_init: tcp rx win: 5744
I (734) wifi_init: tcp mss: 1440
I (744) wifi_init: WiFi IRAM OP enabled
I (744) wifi_init: WiFi RX IRAM OP enabled
I (754) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (854) wifi:mode : sta (a8:03:2a:18:c7:f0)
I (854) wifi:enable tsf
I (854) wifi station: wifi_init_sta finished.
I (864) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (864) wifi:state: init -> auth (b0)
I (894) wifi:state: auth -> assoc (0)
I (904) wifi:state: assoc -> run (10)
I (934) wifi:connected with xx, aid = 4, channel 1, BW20, bssid = dc:15:c8:5e:96:9b
I (934) wifi:security: WPA2-PSK, phy: bgn, rssi: -83
I (944) wifi:pm start, type: 1

I (954) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (964) wifi:<ba-add>idx:0 (ifx:0, dc:15:c8:5e:96:9b), tid:6, ssn:2, winSize:64
I (2144) esp_netif_handlers: sta ip: 192.168.178.36, mask: 255.255.255.0, gw: 192.168.178.1
I (2144) wifi station: got ip:192.168.178.36
I (2144) wifi station: connected to ap SSID:XX password:XX

`

chegewara, 2021-10-06:
Fix this issue first:
`Brownout detector was triggered`
Its problem with power supply or USB cable.

Also remember to deinit wifi before go to deep sleep.

DimitriLubenski, 2021-10-06:
Thank you! 

I "solved" that for now like this and it works. Was experementing with differnt USB outputs and cabales without any success. Hope that is only my personal issue.
 
`
      uint32_t brown_reg_temp = READ_PERI_REG(RTC_CNTL_BROWN_OUT_REG); //save WatchDog register
      WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
      WifiConnector wifi;      
      wifi.connect();
      WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, brown_reg_temp); //enable brownout detector
`