espressif/esp-idf #7654: W6100

MadsHHLund, 2021-10-06:
Driver for w6100

I have had big problems with the stability of w5500, getting hot and unstable, and have been focusing og then w6100 chip.

I have modified then esp_eth_mac_w5500, to esp_eth_mac_w6100, and esp_eth_phy_w5500 to esp_eth_phy_w6100.

I have testet it, and in my application it's working, but I would be happy, if someone would test on, and perhaps make it an integrated part of EPS-IDF version x.x

[w6100.zip](https://github.com/espressif/esp-idf/files/7294445/w6100.zip)

david-cermak, 2021-11-10:
@MadsHHLund Any comment from you about the suggested approach? This driver won't be an integral part of IDF, but could be made an external component.

MadsHHLund, 2021-11-10:
I think its at good solution.

david-cermak, 2021-11-10:
@MadsHHLund Thanks for the feedback! Could you close this issue then?

MadsHHLund, 2025-01-13:
I have recently updated, the driver to wiz610, to support ESP-IDF v. 5.3.2
esp_eth_mac_w6100.c
esp_eth_phy_w6100.c
w6100.h
[w6100.zip](https://github.com/user-attachments/files/18397349/w6100.zip)

david-cermak, 2025-01-13:
@MadsHHLund Thanks for the update! Please note that it's already possible for everyone to publish their own components in IDF component registry. You can check this documentation on creating and publishing a new component: https://docs.espressif.com/projects/idf-component-manager/en/latest/guides/packaging_components.html

PS: As mentioned earlier, ESP-IDF would not accept this driver into the internal layers.