espressif/esp-idf #7656: Endless restart after several days of running

achao1104, 2021-10-07:
hello,

we used esp32s2 and sdk v4.3  release. It runs smoothly for a couple of days, and then enter "endless restart" (the log repeated quickly), if we pull down en pin for a while, it turns to normal again.

[0;32mI (562) spiram: SPI SRAM memory test OK[0m
Guru Meditation Error: Core  0 panic'ed (Cache error). 

Core  0 register dump:
PC      : 0x4003d48d  PS      : 0x00060d34  A0      : 0x800a47ad  A1      : 0x3fffe280  
A2      : 0x02000000  A3      : 0x00000000  A4      : 0x02000000  A5      : 0x0000000f  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fffe240  
A10     : 0x00fdffff  A11     : 0x6000e000  A12     : 0x00000000  A13     : 0x50000000  
A14     : 0x082c5e2d  A15     : 0x9b485e2d  SAR     : 0x00000007  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x50000000  LCOUNT  : 0x400259bc  

Backtrace:0x4003d48a:0x3fffe280 0x400a47aa:0x3fffe290 0x4008a015:0x3fffe2b0 0x4008a5d2:0x3fffe2d0 0x40025985:0x3fffe310 0x40050fb1:0x3fffe330 0x40051419:0x3fffe370 0x4004c265:0x3fffe4a0 0x40011371:0x3fffe560 0x4000f788:0x3fffe6f0 0x400072a5:0x3fffe710

ELF file SHA256: dc492b645f2a042d

[0;31mE (638) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0[0m
Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026ee6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x19cc
load:0x4004c000,len:0xc80
load:0x40050000,len:0x2d3c
entry 0x4004c214
[0;32mI (24) boot: ESP-IDF v4.3-dirty 2nd stage bootloader[0m
[0;32mI (24) boot: compile time 11:06:41[0m
[0;32mI (24) boot: chip revision: 0[0m
[0;32mI (26) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (32) boot.esp32s2: SPI Speed      : 80MHz[0m
[0;32mI (37) boot.esp32s2: SPI Mode       : QIO[0m
[0;32mI (41) boot.esp32s2: SPI Flash Size : 4MB[0m
[0;32mI (46) boot: Enabling RNG early entropy source...[0m
[0;32mI (51) boot: Partition Table:[0m
[0;32mI (55) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (62) boot:  0 nvs              WiFi data        01 02 00009000 00005000[0m
[0;32mI (70) boot:  1 otadata          OTA data         01 00 0000e000 00002000[0m
[0;32mI (77) boot:  2 ota_0            OTA app          00 10 00010000 00100000[0m
[0;32mI (85) boot:  3 ota_1            OTA app          00 11 00110000 00100000[0m
[0;32mI (92) boot:  4 storage          Unknown data     01 82 00210000 001e0000[0m
[0;32mI (100) boot:  5 coredump         Unknown data     01 03 003f0000 00010000[0m
[0;32mI (107) boot: End of partition table[0m
[0;32mI (111) esp_image: segment 0: paddr=00110020 vaddr=3f000020 size=25230h (152112) map[0m
[0;32mI (148) esp_image: segment 1: paddr=00135258 vaddr=3ffcd4c0 size=05758h ( 22360) load[0m
[0;32mI (153) esp_image: segment 2: paddr=0013a9b8 vaddr=40024000 size=05660h ( 22112) load[0m
[0;32mI (159) esp_image: segment 3: paddr=00140020 vaddr=40080020 size=8f694h (587412) map[0m
[0;32mI (273) esp_image: segment 4: paddr=001cf6bc vaddr=40029660 size=13e60h ( 81504) load[0m
[0;32mI (292) esp_image: segment 5: paddr=001e3524 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (306) boot: Loaded app from partition at offset 0x110000[0m
[0;32mI (306) boot: Disabling RNG early entropy source...[0m
[0;32mI (318) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte[0m
[0;32mI (318) cache: Data cache 		: size 8KB, 4Ways, cache line size 32Byte[0m
[0;32mI (324) spiram: Found 16MBit SPI RAM device[0m
[0;32mI (329) spiram: SPI RAM mode: sram 80m[0m
[0;32mI (333) spiram: PSRAM initialized, cache is in normal (1-core) mode.[0m
[0;32mI (340) cpu_start: Pro cpu up.[0m
[0;32mI (562) spiram: SPI SRAM memory test OK[0m
Guru Meditation Error: Core  0 panic'ed (Cache error). 

Core  0 register dump:
PC      : 0x4003d48d  PS      : 0x00060d34  A0      : 0x800a47ad  A1      : 0x3fffe280  
A2      : 0x02000000  A3      : 0x00000000  A4      : 0x02000000  A5      : 0x0000000f  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fffe240  
A10     : 0x00fdffff  A11     : 0x6000e000  A12     : 0x00000000  A13     : 0x50000000  
[... 385 lines cut ...]
[0;32mI (148) esp_image: segment 1: paddr=00135258 vaddr=3ffcd4c0 size=05758h ( 22360) load[0m
[0;32mI (153) esp_image: segment 2: paddr=0013a9b8 vaddr=40024000 size=05660h ( 22112) load[0m
[0;32mI (159) esp_image: segment 3: paddr=00140020 vaddr=40080020 size=8f694h (587412) map[0m
[0;32mI (273) esp_image: segment 4: paddr=001cf6bc vaddr=40029660 size=13e60h ( 81504) load[0m
[0;32mI (292) esp_image: segment 5: paddr=001e3524 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (306) boot: Loaded app from partition at offset 0x110000[0m
[0;32mI (306) boot: Disabling RNG early entropy source...[0m
[0;32mI (318) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte[0m
[0;32mI (318) cache: Data cache 		: size 8KB, 4Ways, cache line size 32Byte[0m
[0;32mI (324) spiram: Found 16MBit SPI RAM device[0m
[0;32mI (329) spiram: SPI RAM mode: sram 80m[0m
[0;32mI (333) spiram: PSRAM initialized, cache is in normal (1-core) mode.[0m
[0;32mI (340) cpu_start: Pro cpu up.[0m
[0;32mI (562) spiram: SPI SRAM memory test OK[0m
Guru Meditation Error: Core  0 panic'ed (Cache error). 

Core  0 register dump:
PC      : 0x4003d48d  PS      : 0x00060d34  A0      : 0x800a47ad  A1      : 0x3fffe280  
A2      : 0x02000000  A3      : 0x00000000  A4      : 0x02000000  A5      : 0x0000000f  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fffe240  
A10     : 0x00fdffff  A11     : 0x6000e000  A12     : 0x00000000  A13     : 0x50000000  
A14     : 0x12e2e6b8  A15     : 0x9f54e6b8  SAR     : 0x00000007  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x50000000  LCOUNT  : 0x400259bc  

Backtrace:0x4003d48a:0x3fffe280 0x400a47aa:0x3fffe290 0x4008a015:0x3fffe2b0 0x4008a5d2:0x3fffe2d0 0x40025985:0x3fffe310 0x40050fb1:0x3fffe330 0x40051419:0x3fffe370 0x4004c265:0x3fffe4a0 0x40011371:0x3fffe560 0x4000f788:0x3fffe6f0 0x400072a5:0x3fffe710

gadget-man, 2021-10-07:
I've noticed a similar issue. It's only happening on IDF 4.x, but devices seem to get stuck in a never-ending boot loop until you do a power down and restart. This is particularly tricky in my case as they are battery powered devices with no manual power-off.  I'm using Wrover with PSRAM, and in my case the issue always seems to happen just as the device attempts to go into deep sleep. Note I use mongoose-os so the log is slightly different:

[Jul 11 20:58:10.362] pn_sleep.c:40           idle
[Jul 11 20:58:10.366] pn_sleep.c:56           Entering Deep Sleep
[Jul 11 20:58:10.371] Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[Jul 11 20:58:10.378] 
[Jul 11 20:58:10.378] Core  0 register dump:
[Jul 11 20:58:10.380] PC      : 0x40126456  PS      : 0x00060133  A0      : 0x8011bb0d  A1      : 0x3ffda870  
[Jul 11 20:58:10.388] A2      : 0x3ffdacc4  A3      : 0x00000008  A4      : 0x3f402aea  A5      : 0x3ffdabb0  
[Jul 11 20:58:10.396] A6      : 0x3ffdab90  A7      : 0x00000004  A8      : 0x80126456  A9      : 0x3ffda850  
[Jul 11 20:58:10.404] A10     : 0x0000fff0  A11     : 0x00000004  A12     : 0x3ffb0e90  A13     : 0x00000001  
[Jul 11 20:58:10.411] A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
[Jul 11 20:58:10.419] EXCVADDR: 0x0000fff0  LBEG    : 0x4008a135  LEND    : 0x4008a145  LCOUNT  : 0xffffffff  
[Jul 11 20:58:10.427] 
[Jul 11 20:58:10.427] Backtrace:0x40126453 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 |<-CONTINUES
[Jul 11 20:58:10.524] 
[Jul 11 20:58:10.524] 
[Jul 11 20:58:10.524] ELF file SHA256: c47d9bc8a9532185
[Jul 11 20:58:10.527] 
[Jul 11 20:58:10.528] 
[Jul 11 20:58:10.528] --- BEGIN CORE DUMP ---
[Jul 11 20:58:10.530] mos: catching core dump
[Jul 11 20:58:10.541] mos: core dump aborted
[Jul 11 20:58:10.541] 
[Jul 11 20:58:10.541] Core  0 register dump:
[Jul 11 20:58:10.544] PC      : 0x4011da89  PS      : 0x00060433  A0      : 0x8012995c  A1      : 0x3ffda1f0  
[Jul 11 20:58:10.551] A2      : 0x3ffdacc4  A3      : 0x3ffda200  A4      : 0x3f402f41  A5      : 0x3ffda660  
[Jul 11 20:58:10.559] A6      : 0x3ffda640  A7      : 0x00000004  A8      : 0x8011da89  A9      : 0x3ffda1d0  
[Jul 11 20:58:10.567] A10     : 0x0000fff0  A11     : 0x00060e23  A12     : 0x00060e23  A13     : 0x3ffda810  
[Jul 11 20:58:10.575] A14     : 0x3ffdac5c  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[Jul 11 20:58:10.582] EXCVADDR: 0x0000fff0  LBEG    : 0x400d3d48  LEND    : 0x400d3d5c  LCOUNT  : 0x00000005  
[Jul 11 20:58:10.590] 
[Jul 11 20:58:10.590] Backtrace:0x4011da86 0x40129959 0x4012999f 0x400d3c29 0x400d3d07 0x40082d3b 0x4008b1e9 0x4008b2e5 0x4008b605 0x400d5fb1 0x40082df2 0x40126453 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 0x400845d5 0x401264dd 0x4011bb0a 0x400832e5 0x400833de 0x400921fd 0x4008e17f 0x4008e424 0x40084462 0x40084493 |<-CONTINUES
[Jul 11 20:58:10.687] 
[Jul 11 20:58:10.687] 
[Jul 11 20:58:10.688] ELF file SHA256: c47d9bc8a9532185
[Jul 11 20:58:10.691] 
[Jul 11 20:58:10.691] Re-entered core dump! Exception happened during core dump!
[Jul 11 20:58:10.696] Rebooting...
[Jul 11 20:58:10.698] ets Jul 29 2019 12:21:46
[Jul 11 20:58:10.700] 
[Jul 11 20:58:10.700] rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[Jul 11 20:58:10.705] configsip: 0, SPIWP:0xee
[Jul 11 20:58:10.707] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[Jul 11 20:58:10.714] mode:DIO, clock div:1
[Jul 11 20:58:10.716] load:0x3fff0018,len:4
[Jul 11 20:58:10.718] load:0x3fff001c,len:6776
[Jul 11 20:58:10.720] load:0x40078000,len:10708
[... cut ...]
[Jul 11 20:58:52.975] EXCVADDR: 0x00000000  LBEG    : 0x400899ee  LEND    : 0x400899f9  LCOUNT  : 0x00000000  
[Jul 11 20:58:52.983] 
[Jul 11 20:58:52.983] Backtrace:0x4008f1a9 0x400d29ae 0x400d6249 0x400d9c85 0x400840d0
[Jul 11 20:58:52.989] 
[Jul 11 20:58:52.989] 
[Jul 11 20:58:52.989] ELF file SHA256: c47d9bc8a9532185
[Jul 11 20:58:52.992] 
[Jul 11 20:58:52.993] 
[Jul 11 20:58:52.993] --- BEGIN CORE DUMP ---
[Jul 11 20:58:52.995] mos: catching core dump
[Jul 11 20:58:55.829] .............
[Jul 11 20:59:32.567] ---- END CORE DUMP ----

0xjakob, 2021-10-08:
@achao1104 @gadget-man Is it possible to decipher the backtrace, e.g. using `xtensa-esp32s2-elf-addr2line`?

Something you can try is using coredump or the built-in gdb, either one enabled via menuconfig, so that the device state can be examined after a crash.

achao1104, 2021-10-08:
it enters a dead loop as the log states.

0xjakob, 2021-10-08:
@achao1104 That's what I've seen. Now it would be nice to see where in the code it does that, i.e., what the `Backtrace:0x4003d48a:0x3fffe280 0x400a47aa:0x3fffe290 0x4008a015:0x3fffe2b0 0x4008a5d2:0x3fffe2d0 0x40025985:0x3fffe310 0x40050fb1:0x3fffe330 0x40051419:0x3fffe370 0x4004c265:0x3fffe4a0 0x40011371:0x3fffe560 0x4000f788:0x3fffe6f0 0x400072a5:0x3fffe710` translates to in terms of source code locations in IDF.

gadget-man, 2021-10-08:
I will try and capture a core dump and run analysis to find the backtrace

gadget-man, 2021-10-08:
Here's what the core dump shows:
Core dump by iParcelBox_firmware/esp32 2.0.247 20211007-144005
Using ELF file at: /Users/PaulNeedler/Documents/iParcelBox/3-Firmware/iParcelBox_firmware/build/objs/iParcelBox_firmware.elf
Using Docker image: docker.io/mgos/esp32-build:4.2-r1
Running docker run --rm -i --tty=true -v /Users/PaulNeedler/Documents/iParcelBox/3-Firmware/iParcelBox_firmware/build/objs/iParcelBox_firmware.elf:/fw.elf -v /Users/PaulNeedler/Documents/iParcelBox/3-Firmware/iParcelBox_firmware/core-iParcelBox_firmware-esp32-20211008-193057.420086182:/core -v /Users/PaulNeedler/Documents/iParcelBox/3-Firmware/iParcelBox_firmware:/Users/PaulNeedler/Documents/iParcelBox/3-Firmware/iParcelBox_firmware docker.io/mgos/esp32-build:4.2-r1 bash -c /usr/local/bin/serve_core.py --rom=/opt/Espressif/rom/rom.bin --rom_addr=0x40000000 --xtensa_addr_fixup=true /fw.elf /core & $MGOS_TARGET_GDB /fw.elf -ex 'target remote 127.0.0.1:1234' -ex 'set confirm off' -ex bt
Unable to find image 'mgos/esp32-build:4.2-r1' locally
4.2-r1: Pulling from mgos/esp32-build
35c102085707: Pull complete 
251f5509d51d: Pull complete 
8e829fe70a46: Pull complete 
6001e1789921: Pull complete 
6369aa549000: Pull complete 
c2364282aac9: Pull complete 
0c27615264fc: Pull complete 
ad3b4552b144: Pull complete 
ca278db297a8: Pull complete 
607e71a9215d: Pull complete 
f46dac7d5092: Pull complete 
f85856a6b5d4: Pull complete 
c8cd78ebe055: Pull complete 
5ec0ea2cd94f: Pull complete 
64e876348090: Pull complete 
5cd219b42dca: Pull complete 
c0668a2c39af: Pull complete 
b3b4a723e6ad: Pull complete 
Digest: sha256:66e3c273adb1f509cffc830b2ff68f06ec6137f8ce7b9a29c3081a1c47c8751c
Status: Downloaded newer image for mgos/esp32-build:4.2-r1
GNU gdb (crosstool-NG esp-2020r3) 8.1.0.20180627-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /fw.elf...Waiting for gdb on 1234
done.
Remote debugging using 127.0.0.1:1234
Found core at 23 - 458769
Mapping DRAM: 335872 @ 0x3ffae000
Mapping /opt/Espressif/rom/rom.bin at 0x40000000
Mapping /fw.elf .rtc.text: 58 @ 0x400c0000
Mapping /fw.elf .rtc.dummy: 58 @ 0x3ff80000
Mapping /fw.elf .rtc.force_fast: 2 @ 0x3ff8003a
Mapping /fw.elf .rtc.data: 16 @ 0x50000000
Mapping /fw.elf .iram0.vectors: 1024 @ 0x40080000
Mapping /fw.elf .iram0.text: 109774 @ 0x40080400
Mapping /fw.elf .dram0.data: 14144 @ 0x3ffb0000
Mapping /fw.elf .dram0.bss: 42600 @ 0x3ffb3740
Mapping /fw.elf .flash.rodata: 199400 @ 0x3f400020
Mapping /fw.elf .flash.text: 848951 @ 0x400d0018
Dump contains FreeRTOS task info
Loaded core dump from last snippet in  /core
_xt_coproc_restorecs () at /opt/Espressif/esp-idf/components/freertos/xtensa_context.S:577
577	/opt/Espressif/esp-idf/components/freertos/xtensa_context.S: No such file or directory.
#0  _xt_coproc_restorecs () at /opt/Espressif/esp-idf/components/freertos/xtensa_context.S:577
#1  0x400d29b1 in psram_get_size () at /opt/Espressif/esp-idf/components/esp32/spiram_psram.c:782
#2  0x400d6104 in _svfprintf_r (data=<optimized out>, fp=<optimized out>, fmt0=<optimized out>, ap=...) at ../../../.././newlib/libc/stdio/vfprintf.c:1536
#3  0x400d9a28 in _vfiprintf_r (data=<optimized out>, fp=<optimized out>, fmt0=<optimized out>, ap=...) at ../../../.././newlib/libc/stdio/vfprintf.c:1544
#4  0x400840cf in pthread_mutex_lock_internal (mux=<optimized out>, tmo=0) at /opt/Espressif/esp-idf/components/pthread/pthread.c:622

achao1104, 2021-10-13:
it is idf's issue, not related to business code. you might notice the device doesn't finish rebooting and then enter rebooint again.
you may notice " esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead"
also I tried to get a dump from flash, but it is unrelated to the endless rebooting issue.
[0;32mI (570) spiram: SPI SRAM memory test OK[0m
Guru Meditation Error: Core  0 panic'ed (Cache error). 

Core  0 register dump:
PC      : 0x4003e035  PS      : 0x00060d34  A0      : 0x800a4e55  A1      : 0x3fffe280  
A2      : 0x02000000  A3      : 0x00000000  A4      : 0x02000000  A5      : 0x0000000f  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fffe240  
A10     : 0x00fdffff  A11     : 0x6000e000  A12     : 0x00000000  A13     : 0x50000000  
A14     : 0x6226a905  A15     : 0xa46ae905  SAR     : 0x00000007  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x50000000  LCOUNT  : 0x40025a9c  

Backtrace:0x4003e032:0x3fffe280 0x400a4e52:0x3fffe290 0x40089fb1:0x3fffe2b0 0x4008a56e:0x3fffe2d0 0x40025a65:0x3fffe310 0x40051055:0x3fffe330 0x40051541:0x3fffe370 0x4004c269:0x3fffe4a0 0x40011371:0x3fffe560 0x4000f788:0x3fffe6f0 0x400072a5:0x3fffe710

ELF file SHA256: 4141155a9fe42a90

[0;31mE (646) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0[0m
Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026fc6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1a90
load:0x4004c000,len:0xc84

achao1104, 2021-10-13:
Hello,

part a blew is normal reboot log, part b is an endless loop reboot log. it looks a "Guru Meditation Error: Core  0 panic'ed (Cache error). " error occurs when calling "cpu_start: Pro cpu start user code", and then there is a flash crc error in "esp_core_dump_to_flash()", since it is in the early of rebooting, I didn't find a way to translate the backtrace to codes.
but from the flash crc error, I guess it might be due to flash is not loaded correctly.
idf version: v4.3 release, esp32s2 with core dump configed.

PART A
I (329) spiram: Found 16MBit SPI RAM device
I (334) spiram: SPI RAM mode: sram 80m
I (338) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (345) cpu_start: Pro cpu up.
I (567) s��ff���*��SRAM memory test OK
I (577) cpu_start: Pro cpu start user code
I (577) cpu_start: cpu freq: 240000000
PART B
[0;32mI (340) spiram: PSRAM initialized, cache is in normal (1-core) mode.[0m
[0;32mI (347) cpu_start: Pro cpu up.[0m
[0;32mI (570) spiram: SPI SRAM memory test OK[0m
Guru Meditation Error: Core  0 panic'ed (Cache error). 

Core  0 register dump:
PC      : 0x4003e035  PS      : 0x00060d34  A0      : 0x800a4e55  A1      : 0x3fffe280  
A2      : 0x02000000  A3      : 0x00000000  A4      : 0x02000000  A5      : 0x0000000f  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fffe240  
A10     : 0x00fdffff  A11     : 0x6000e000  A12     : 0x00000000  A13     : 0x50000000  
A14     : 0x5f24fb99  A15     : 0xa4363b99  SAR     : 0x00000007  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x50000000  LCOUNT  : 0x40025a9c  

Backtrace:0x4003e032:0x3fffe280 0x400a4e52:0x3fffe290 0x40089fb1:0x3fffe2b0 0x4008a56e:0x3fffe2d0 0x40025a65:0x3fffe310 0x40051055:0x3fffe330 0x40051541:0x3fffe370 0x4004c269:0x3fffe4a0 0x40011371:0x3fffe560 0x4000f788:0x3fffe6f0 0x400072a5:0x3fffe710

ELF file SHA256: 4141155a9fe42a90

[0;31mE (646) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0[0m
Rebooting...
ESP-ROM:esp32s2-rc4-20191025

achao1104, 2021-10-30:
Hello

Do you have any progress?

Thanks
Peter

gadget-man, 2021-11-05:
Is there any update on this? I'm seeing it crashing VERY regularly when trying to go into deep sleep (and then getting stuck in a reboot loop until power is removed):

My sleep code is as follows:
`rtc_gpio_isolate(GPIO_NUM_12);
    esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);
    esp_sleep_enable_timer_wakeup((uint64_t)seconds * 1000000); //365*24*60*60*1000000);
    ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(mgos_sys_config_get_board_lid_pin(),mgos_sys_config_get_board_lid_active_high()));
    ESP_ERROR_CHECK(esp_sleep_enable_ext1_wakeup(BUTTON_PIN_BITMASK, ESP_EXT1_WAKEUP_ALL_LOW)); // GPIO Pin 13 for button
    LOG(LL_INFO, ("Pre-Sleep settings updated"));
    esp_deep_sleep_start();`

And the log output:
`[Nov  5 12:20:29.922] pn_sleep.c:58           Entering Deep Sleep
[Nov  5 12:20:29.926] pn_sleep.c:66           Pre-Sleep settings updated
[Nov  5 12:20:29.931] /opt/Espressif/esp-idf/components/freertos/tasks.c:3095 (xTaskRemoveFromEventList)- assert failed!
[Nov  5 12:20:29.940] 
[Nov  5 12:20:29.940] abort() was called at PC 0x4008c430 on core 0
[Nov  5 12:20:29.944] 
[Nov  5 12:20:29.944] Backtrace:0x4008b005 |<-CORRUPTED
[Nov  5 12:20:29.947] 
[Nov  5 12:20:29.947] 
[Nov  5 12:20:29.948] ELF file SHA256: 4bcf1d6ed8153a9e
[Nov  5 12:20:29.951] 
[Nov  5 12:20:29.951] 
[Nov  5 12:20:29.951] --- BEGIN CORE DUMP ---
[Nov  5 12:20:29.953] mos: catching core dump
[Nov  5 12:20:29.973] mos: core dump aborted
[Nov  5 12:20:29.973] 
[Nov  5 12:20:29.973] Backtrace:0x4008deb7 |<-CORRUPTED
[Nov  5 12:20:29.976] 
[Nov  5 12:20:29.976] 
[Nov  5 12:20:29.976] ELF file SHA256: 4bcf1d6ed8153a9e
[Nov  5 12:20:29.979] 
[Nov  5 12:20:29.979] Re-entered core dump! Exception happened during core dump!
[Nov  5 12:20:29.984] Rebooting...
[Nov  5 12:20:29.986] ets Jul 29 2019 12:21:46
[Nov  5 12:20:29.988] 
[Nov  5 12:20:29.989] rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[Nov  5 12:20:29.993] configsip: 0, SPIWP:0xee
[Nov  5 12:20:29.996] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[Nov  5 12:20:30.002] mode:DIO, clock div:1
[Nov  5 12:20:30.004] load:0x3fff0018,len:4
[Nov  5 12:20:30.006] load:0x3fff001c,len:6776
[Nov  5 12:20:30.008] load:0x40078000,len:10708
[Nov  5 12:20:30.011] load:0x40080400,len:6512
[Nov  5 12:20:30.013] entry 0x40080714
[Nov  5 12:20:30.014] [0;32mI (63) boot: Chip Revision: 3[0m
[Nov  5 12:20:30.018] [0;32mI (65) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
[Nov  5 12:20:30.026] [0;32mI (39) boot: ESP-IDF v3.3-r5 2nd stage bootloader[0m
[Nov  5 12:20:30.032] [0;32mI (40) boot: compile time 13:05:40[0m
[Nov  5 12:20:30.036] [0;32mI (40) boot: Enabling RNG early entropy source...[0m
[Nov  5 12:20:30.042] [0;32mI (44) qio_mode: Enabling default flash chip QIO[0m
[Nov  5 12:20:30.047] [0;32mI (50) boot: SPI Speed      : 80MHz[0m
[Nov  5 12:20:30.051] [0;32mI (54) boot: SPI Mode       : QIO[0m
[Nov  5 12:20:30.055] [0;32mI (58) boot: SPI Flash Size : 16MB[0m
[Nov  5 12:20:30.059] [0;32mI (62) boot: Partition Table:[0m
[Nov  5 12:20:30.063] [0;32mI (66) boot: ## Label            Usage          Type ST Offset   Length   Flags[0m
[Nov  5 12:20:30.071] [0;32mI (74) boot:  0 nvs              WiFi data        01 02 00009000 00004000 00000000[0m
[Nov  5 12:20:30.079] [0;32mI (82) boot:  1 otadata          OTA data         01 00 0000d000 00002000 00000000[0m
[Nov  5 12:20:30.087] [0;32mI (90) boot:  2 app_0            OTA app          00 10 00010000 00180000 00000000[0m
[Nov  5 12:20:30.096] [0;32mI (98) boot:  3 fs_0             FS               01 82 00190000 00060000 00000000[0m
[Nov  5 12:20:30.104] [0;32mI (107) boot:  4 app_1            OTA app          00 11 001f0000 00180000 00000000[0m
[Nov  5 12:20:30.112] [0;32mI (115) boot:  5 fs_1             FS               01 82 00370000 00060000 00000000[0m
[Nov  5 12:20:30.121] [0;32mI (123) boot: End of partition table[0m
[Nov  5 12:20:30.125] [0;32mI (128) boot: OTA data 0: seq 0x000000e2, st 0x11, CRC 0x78cccf19, valid? 1[0m
[Nov  5 12:20:30.133] [0;32mI (135) boot: OTA data 1: seq 0x000000e3, st 0x10, CRC 0x88a337ae, valid? 1[0m
[Nov  5 12:20:30.140] [0;32mI (143) boot_comm: chip revision: 3, min. application chip revision: 0[0m
[Nov  5 12:20:30.147] [0;32mI (150) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2e3b4 (189364) map[0m
[Nov  5 12:20:30.195] [0;32mI (209) esp_image: segment 1: paddr=0x0003e3dc vaddr=0x3ffb0000 size=0x01c3c (  7228) load[0m
[Nov  5 12:20:30.204] [0;32mI (211) esp_image: segment 2: paddr=0x00040020 vaddr=0x400d0020 size=0xce2e8 (844520) map[0m
[... 30 lines cut ...]
[Nov  5 12:20:31.130] [0;32mI (1138) spi_flash: flash io: qio[0m
[Nov  5 12:20:31.134] [0;32mI (1139) cpu_start: Starting scheduler on PRO CPU.[0m
[Nov  5 12:20:31.139] [0;32mI (1142) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[Nov  5 12:20:31.148] 
[Nov  5 12:20:31.148] 
[Nov  5 12:20:31.148] mgos_freertos.c:155     iParcelBox_firmware 2.0.247 (20211007-144005)
[Nov  5 12:20:31.154] mgos_freertos.c:157     Mongoose OS 2.19.1 (20211007-144005/2.19.1-gd52032f)
[Nov  5 12:20:31.161] Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[Nov  5 12:20:31.168] 
[Nov  5 12:20:31.169] Core  0 register dump:
[Nov  5 12:20:31.171] PC      : 0x4008f158  PS      : 0x00060730  A0      : 0x800d29b1  A1      : 0x3ffb8350  
[... cut ...]
[Nov  5 12:20:31.202] A14     : 0x3ffae9a4  A15     : 0x00000001  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
[Nov  5 12:20:31.209] EXCVADDR: 0x00000000  LBEG    : 0x400899ea  LEND    : 0x400899f5  LCOUNT  : 0x00000000  
[Nov  5 12:20:31.217] 
[Nov  5 12:20:31.217] Backtrace:0x4008f155 0x400d29ae 0x400d6101 0x400d9a25 0x400840cc
[Nov  5 12:20:31.223] 
[Nov  5 12:20:31.223] 
[Nov  5 12:20:31.223] ELF file SHA256: 4bcf1d6ed8153a9e
[Nov  5 12:20:31.226] 
[Nov  5 12:20:31.227] 
[Nov  5 12:20:31.227] --- BEGIN CORE DUMP ---
[Nov  5 12:20:31.229] `

pacucha42, 2021-11-05:
No results yet, sorry. @gadget-man @achao1104 - would you mind sharing your ELF files? Thank you

gadget-man, 2021-11-05:
> No results yet, sorry. @gadget-man @achao1104 - would you mind sharing your ELF files? Thank you

No problem - how can I do this securely with you?

pacucha42, 2021-11-05:
> > No results yet, sorry. @gadget-man @achao1104 - would you mind sharing your ELF files? Thank you
> 
> No problem - how can I do this securely with you?

Thanks! I'll contact you directly on sales@iparcelbox.com - the email you published in your profile. Does that work for you?

gadget-man, 2021-11-05:
> > > No results yet, sorry. @gadget-man @achao1104 - would you mind sharing your ELF files? Thank you
> > 
> > 
> > No problem - how can I do this securely with you?
> 
> Thanks! I'll contact you directly on [sales@iparcelbox.com](mailto:sales@iparcelbox.com) - the email you published in your profile. Does that work for you?

Yes, fine thanks.

achao1104, 2021-11-15:
Hello,

Do you have any updates?  As you suggested, we have disabled "the Memory protection feature in the IDF config", it won't run into a loop, but occasionally crashed, here is a log crashed recently.

==================== THREAD 1 (TCB: 0x3ffe3958, name: 'wifi') =====================
#0  0x4002eae1 in panic_abort (details=0x3ffcd920 \"abort() was called at PC 0x40027357 on core 0\") at D:/project/esp32/idf-v4.3.1/components/esp_system/panic.c:379
#1  0x4002f25c in esp_system_abort (details=0x3ffcd920 \"abort() was called at PC 0x40027357 on core 0\") at D:/project/esp32/idf-v4.3.1/components/esp_system/system_api.c:112
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

AxelLin, 2022-01-17:
@pacucha42 Any update?

AxelLin, 2022-02-13:
> Thanks! I'll contact you directly on [sales@iparcelbox.com](mailto:sales@iparcelbox.com) - the email you published in your profile. Does that work for you?

@pacucha42 
Do you have any finding? Or any additional information required?

AxelLin, 2022-03-18:
@pacucha42 @0xjakob 
There is no response for this issue for several months. How is the status?

HariPrasadNagarajan, 2022-03-31:
Hi all,

We are currently working in ESP32S2 with esp-idf v4.3. We are also facing this endless reboot issue after running for few hours(12 hours to 1 day). But we are observing this error randomly, so we are trying to reproduce the error but couldn't reproduce manually. We also observed another error **Memory protection fault** which triggered the panic reboot at first before going to the endless reboot. Then on the sequential run we observed that device goes to the endless reboot state . I have attached the error log. , 

**[0;32mI (2896947) MESSAGE: Successfully data send 

[0m

Guru Meditation Error: Core  0 panic'ed (Memory protection fault).

abort() was called at PC 0x40026fcb on core 0

Backtrace:0x4002e393:0x3fffa8b0 0x4002ec65:0x3fffa8d0 0x40036226:0x3fffa8f0 0x40026fcb:0x3fffa960 0x40027119:0x3fffa990 0x400271a6:0x3fffa9b0 0x40118cfe:0x3fffa9e0 0x4011bd31:0x3fffacf0 0x401277f9:0x3fffad20 0x40034e7d:0x3fffad50 0x400262dc:0x3fffada0 0x4002e7b9:0x3fffadc0 0x4002e405:0x3fffadf0 0x4002ebf5:0x3fffae70 0x4002ec41:0x3fffaec0 0x40025706:0x3fffaee0 0x00040022:0x3fffaf70 |<-CORRUPTED

ELF file SHA256: 263fdb992f3333ff

Rebooting...

ESP-ROM:esp32s2-rc4-20191025

Build:Oct 25 2019

rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)

Saved PC:0x40026948

SPIWP:0xee

mode:DIO, clock div:1

load:0x3ffe6100,len:0x1920

load:0x4004c000,len:0xa04

load:0x40050000,len:0x2bec

entry 0x4004c1f4

[0;32mI (24) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader [0m

[0;32mI (24) boot: compile time 10:31:05 [0m

[0;32mI (24) boot: chip revision: 0 [0m

[0;32mI (27) boot.esp32s2: SPI Speed      : 80MHz [0m

[0;32mI (31) boot.esp32s2: SPI Mode       : DIO [0m

[0;32mI (36) boot.esp32s2: SPI Flash Size : 4MB [0m

[0;32mI (41) boot: Enabling RNG early entropy source... [0m

[0;32mI (46) boot: Partition Table: [0m

[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length [0m

[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00040000 [0m

[0;32mI (65) boot:  1 otadata          OTA data         01 00 00049000 00002000 [0m

[0;32mI (72) boot:  2 phy_init         RF data          01 01 0004b000 00001000 [0m

[0;32mI (80) boot:  3 ota_0            OTA app          00 10 00050000 00180000 [0m

[0;32mI (87) boot:  4 ota_1            OTA app          00 11 001d0000 00180000 [0m

[0;32mI (94) boot: End of partition table [0m

[0;32mI (99) esp_image: segment 0: paddr=00050020 vaddr=3f000020 size=274ech (161004) map [0m

[... 136 lines cut ...]
A2      : 0x02000000  A3      : 0x00000000  A4      : 0x02000000  A5      : 0x00000004  

A6      : 0x000000c0  A7      : 0x00000000  A8      : 0x3ffffd70  A9      : 0x3fffe240  

A10     : 0x00fdffff  A11     : 0x00000030  A12     : 0x00000000  A13     : 0x50000008  

A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000007  EXCCAUSE: 0x00000007  

EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x50000008  LCOUNT  : 0x400256d4  

Backtrace:0x4003af7e:0x3fffe280 0x400a8ad2:0x3fffe290 0x400889e3:0x3fffe2b0 0x40089196:0x3fffe2d0 0x4002569d:0x3fffe310 0x40050e51:0x3fffe330 0x400512b9:0x3fffe370 0x4004c245:0x3fffe4a0 0x40011371:0x3fffe560 0x4000f788:0x3fffe6f0 0x400072a5:0x3fffe710

ELF file SHA256: 263fdb992f3333ff

Rebooting...**

Questions
1. We are trying to find the root cause of this error, any inputs on where to get started.
2. Is **Memory protection fault** and **cache error** panic are related? 
3. Should we need to disable Memory protection in menuconfig?
4. Is this error is from ESP-idf v4.3?
5. If we change the ESP-idf version, will it solve the issue?
6. Is there anyways to reproduce this error? What are the possible places should I need to visits to debug this error?

Thanks in advance.

gadget-man, 2022-04-05:
Could a workaround for this be to enable RTC Watchdog Timer before going into deep sleep - if it gets stuck in a boot loop, the RWDT would then fire and force a full reboot?

AxelLin, 2023-05-03:
@achao1104 @gadget-man 
Do you still hit this issue with recent esp-idf releases?

Alvin1Zhang, 2025-03-03:
Thanks for reporting and sorry for the very slow turnaround, fix on master branch is available at https://github.com/espressif/esp-idf/commit/e81a9285fe0697ef1b6c1dec173b7c0332203bac, we are back porting this fix until release/5.0, feel free to reopen if the issue still happens. Thanks.

closed 2025-03-03 by Alvin1Zhang