espressif/esp-idf #7660: FAILED: esp-idf/mbedtls/x509_crt_bundle

Mair, 2021-10-07:
Hi @Typoception Thank you for the issue. The issue is because one of the certificates in the [cacrt_all.pem](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/esp_crt_bundle/cacrt_all.pem) file has been expired recently ( 30 sep ). That is causing the failure in build.
Can you please disable the following option and try again if it works.
`(Top) > Component config > mbedTLS > Certificate Bundle->Enable trusted root certificate bundle`

_Originally posted by @AdityaHPatwardhan in https://github.com/espressif/esp-idf/issues/7621#issuecomment-931241064_

mahavirj, 2021-10-08:
@Mair 

Can you please try fix from https://github.com/espressif/esp-idf/pull/7632? We are integrating this fix (till v4.2 release) internally.

Please see https://github.com/espressif/esp-idf/issues/7631#issuecomment-934212224 for more details.

mahavirj, 2021-10-10:
@Mair 

> Do you have a TM when it will be available on V4.2 then V4.3?

Fix has been merged internally till v4.2 branch, it shall appear on github in next few days. Process to sync with github is automated based on successful CI pass event but in case it takes longer we will consider an option to manually push out the fix.

closed 2021-10-12 by mahavirj

UnexpectedMaker, 2021-11-02:
Issue still seems to exist in `release/v4.3.1`

Sn0wfreezeDev, 2021-11-04:
I tested it with the release tagged with v4.3.1 and the issue still occurs. 
Switching to the release/v4.3 resolves it. It would be great to have a release is also able to build

igrr, 2021-11-04:
@Sn0wfreezeDev Definitely, the fix will be included in the next bugfix release of each branch, in this case 4.3.2. Our QA is currently testing 4.3.2 and we plan to release it as soon as the testing is finished.

aguaviva, 2021-11-21:
Hi, any updates on the solution or reasonable workaround? (Thanks for the good work!)