espressif/esp-idf #7663: esp32 ansc bug

donny681, 2021-10-08:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version DevKitC
- Module or chip used: ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it):
   v4.3.1-29-ge493a4c30e
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   gcc version 8.4.0 (crosstool-NG esp-2021r1)
- Operating System:     Linux
- (Windows only) environment type: PowerShell.
- Using an IDE?: No
- Power Supply:       USB

## Problem Description
There is something wrong with ble_ancs demo.It is ok to connect to the moble.When ios send nofication to ESP32 by ancs,the ble disconnect with the ios.If closing the "Share system notifications" in ios .The esp32 wouldn't disconnect the ios. 

### Expected Behavior
The esp32 would receive the CategoryID 
### Actual Behavior
When the iPhone receiving the message and share system notifications to esp32 ,the esp32 disconnect.

### Steps to reproduce

1. iPhone connect to  the ESP32 (it is ok)
2. Make a call to 1008611 by iPhone.After a few seconds the iphone would receive the message and share system notifications to esp32,the esp32 receive the notifications and disconnect with the esp32

// If possible, attach a picture of your setup/wiring here.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

> entry 0x40080680
I (29) boot: ESP-IDF v4.3.1-29-ge493a4c30e 2nd stage bootloader
I (29) boot: compile time 14:00:27
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1ae88h (110216) map
I (153) esp_image: segment 1: paddr=0002aeb0 vaddr=3ffbdb60 size=03eb4h ( 16052) load
I (159) esp_image: segment 2: paddr=0002ed6c vaddr=40080000 size=012ach (  4780) load
I (161) esp_image: segment 3: paddr=00030020 vaddr=400d0020 sizI (333) esp_image: segment 4: paddr=000a024c vaddr=400812ac size=15938h ( 88376) load
I (370) esp_image: segment 5: paddr=000b5b8c vaddr=50000000 size=00010h (    16) load
I (381) boot: Loaded app from partition at offset 0x10000
I (381) boot: Disabling RNG early entropy source...
I (393) cpu_start: Pro cpu up.
I (393) cpu_start: Starting app cpu, entry point is 0x400811d8
0x400811d8: call_start_cpu1 at /home/yates/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (407) cpu_start: Pro cpu start user code
I (407) cpu_start: cpu freq: 160000000
I (407) cpu_start: Application information:
I (411) cpu_start: Project name:     ble_ancs
I (416) cpu_start: App version:      1
I (421) cpu_start: Compile time:     Oct  8 2021 14:00:19
I (427) cpu_start: ELF file SHA256:  79a89579ff614a2d...
I (433) cpu_start: ESP-IDF:          v4.3.1-29-ge493a4c30e
I (439) heap_init: Initializing. RAM available for dynamic allocation:
I (446) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (452) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (459) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (465) heap_init: At 3FFC6008 len 00019FF8 (103 KiB): DRAM
I (471) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (477) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (484) heap_init: At 40096BE4 len 0000941C (37 KiB): IRAM
I (491) spi_flash: detected chip: gd
I (494) spi_flash: flash io: dio
W (498) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (512) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (557) BTDM_INIT: BT controller compile version [bfbbe1e]
I (557) system_api: Base MAC address is not set
I (557) system_api: read default base MAC address from EFUSE
I (567) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (907) BLE_ANCS: app_main init bluetooth
I (997) BLE_ANCS: REG_EVT
E (997) BT_BTM: BTM_BleConfigLocalIcon

I (1037) BLE_ANCS: advertising start success
E (1777) BT_APPL: bta_gattc_process_indicate indication/notif for unknown device, ignore
I (1787) BLE_ANCS: ESP_GATTC_OPEN_EVT
I (1987) addr: 63 1f 21 fb 79 d1 
I (1987) BLE_ANCS: pair status = success
I (5107) BLE_ANCS: ESP_GATTC_DIS_SRVC_CMPL_EVT
I (5167) BLE_ANCS: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (5167) BLE_ANCS: ESP_GATTC_SEARCH_CMPL_EVT
I (5167) BLE_ANCS: Find Apple control point char
I (5177) BLE_ANCS: Find Apple noticification source char
I (5177) BLE_ANCS: Find Apple data source char
W (32257) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x13
I (32257) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13
I (32277) BLE_ANCS: advertising start success
I (32837) BLE_ANCS: ESP_GATTC_OPEN_EVT
I (33097) addr: 63 1f 21 fb 79 d1 
I (33097) BLE_ANCS: pair status = success
I (36217) BLE_ANCS: ESP_GATTC_DIS_SRVC_CMPL_EVT
I (36277) BLE_ANCS: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (36277) BLE_ANCS: ESP_GATTC_SEARCH_CMPL_EVT
I (36277) BLE_ANCS: Find Apple control point char
I (36287) BLE_ANCS: Find Apple noticification source char
I (36287) BLE_ANCS: Find Apple data source char
W (63247) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x13
I (63247) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13
I (63267) BLE_ANCS: advertising start success
I (70077) BLE_ANCS: ESP_GATTC_OPEN_EVT
I (70347) addr: 63 1f 21 fb 79 d1 
I (70347) BLE_ANCS: pair status = success
I (73377) BLE_ANCS: ESP_GATTC_DIS_SRVC_CMPL_EVT
I (73437) BLE_ANCS: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (73437) BLE_ANCS: ESP_GATTC_SEARCH_CMPL_EVT
I (73437) BLE_ANCS: Find Apple control point char
I (73437) BLE_ANCS: Find Apple noticification source char
I (73447) BLE_ANCS: Find Apple data source char
W (100487) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x13
I (100497) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13
I (100507) BLE_ANCS: advertising start success
I (100727) BLE_ANCS: ESP_GATTC_OPEN_EVT
I (100967) addr: 63 1f 21 fb 79 d1 
I (100967) BLE_ANCS: pair status = success
I (103927) BLE_ANCS: ESP_GATTC_DIS_SRVC_CMPL_EVT
I (103987) BLE_ANCS: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (103987) BLE_ANCS: ESP_GATTC_SEARCH_CMPL_EVT
I (103997) BLE_ANCS: Find Apple control point char
I (103997) BLE_ANCS: Find Apple noticification source char
I (104007) BLE_ANCS: Find Apple data source char
W (131197) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x13
I (131197) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13
I (131217) BLE_ANCS: advertising start success
I (131297) BLE_ANCS: ESP_GATTC_OPEN_EVT
I (131567) addr: 63 1f 21 fb 79 d1 
I (131567) BLE_ANCS: pair status = success
I (134557) BLE_ANCS: ESP_GATTC_DIS_SRVC_CMPL_EVT
I (134617) BLE_ANCS: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (134617) BLE_ANCS: ESP_GATTC_SEARCH_CMPL_EVT
I (134617) BLE_ANCS: Find Apple control point char
I (134627) BLE_ANCS: Find Apple noticification source char
I (134637) BLE_ANCS: Find Apple data source char
W (161707) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x13
I (161707) BLE_ANCS: ESP_GATTC_DISCONNECT_EVT, reason = 0x13
I (161727) BLE_ANCS: advertising start success
I (161947) BLE_ANCS: ESP_GATTC_OPEN_EVT
I (162217) addr: 63 1f 21 fb 79 d1 
I (162217) BLE_ANCS: pair status = success
I (165267) BLE_ANCS: ESP_GATTC_DIS_SRVC_CMPL_EVT
I (165327) BLE_ANCS: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (165327) BLE_ANCS: ESP_GATTC_SEARCH_CMPL_EVT
I (165337) BLE_ANCS: Find Apple control point char
I (165337) BLE_ANCS: Find Apple noticification source char
I (165347) BLE_ANCS: Find Apple data source char

songzhishuo, 2022-07-12:
我也遇到了同样的问题，请问这个问题什么时候可以解决

aleksykrolczyk, 2022-11-11:
Any progress on that or suggestion how to deal with that issue?

zhp0406, 2023-09-11:
@aleksykrolczyk @songzhishuo @donny681 
Have you resolved your issue?

zhp0406, 2024-03-10:
Version 4.3 has been discontinued. Testing on the master version is normal. Please use the latest master version.

```
I (29) boot: ESP-IDF v5.3-dev-2489-g60a2bf6a68 2nd stage bootloader
I (29) boot: compile time Mar 10 2024 15:05:47
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=22f6ch (143212) map
I (153) esp_image: segment 1: paddr=00032f94 vaddr=3ffbdb60 size=04a38h ( 19000) load
I (160) esp_image: segment 2: paddr=000379d4 vaddr=40080000 size=08644h ( 34372) load
I (174) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=7a584h (501124) map
I (346) esp_image: segment 4: paddr=000ba5ac vaddr=40088644 size=0e584h ( 58756) load
I (381) boot: Loaded app from partition at offset 0x10000
I (381) boot: Disabling RNG early entropy source...
I (393) cpu_start: Multicore app
I (401) cpu_start: Pro cpu start user code
I (401) cpu_start: cpu freq: 160000000 Hz
I (402) app_init: Application information:
I (404) app_init: Project name:     ble_ancs
I (409) app_init: App version:      qa-test-master-20240301-192-g60
I (416) app_init: Compile time:     Mar 10 2024 15:05:40
I (422) app_init: ELF file SHA256:  a58d6fb36...
I (428) app_init: ESP-IDF:          v5.3-dev-2489-g60a2bf6a68
I (434) efuse_init: Min chip rev:     v0.0
I (439) efuse_init: Max chip rev:     v3.99 
I (444) efuse_init: Chip rev:         v3.0
I (449) heap_init: Initializing. RAM available for dynamic allocation:
I (456) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (462) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (468) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (474) heap_init: At 3FFC7498 len 00018B68 (98 KiB): DRAM
I (480) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (487) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (493) heap_init: At 40096BC8 len 00009438 (37 KiB): IRAM
I (501) spi_flash: detected chip: generic
I (504) spi_flash: flash io: dio
W (508) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (522) coexist: coex firmware version: a709fcdeb
I (527) main_task: Started on CPU0
I (537) main_task: Calling app_main()
I (547) BTDM_INIT: BT controller compile version [5ea2fca]
I (547) BTDM_INIT: Bluetooth MAC: 34:94:54:b7:0d:9e
I (547) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
W (557) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (877) BLE_ANCS: app_main init bluetooth
I (977) BLE_ANCS: REG_EVT
I (987) main_task: Returned from app_main()
I (1027) BLE_ANCS: advertising start success
I (22287) BLE_ANCS: ESP_GATTC_OPEN_EVT
W (25457) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (25847) addr: 69 7d 15 11 6d 4c 
I (25847) BLE_ANCS: pair status = success
I (26367) BLE_ANCS: ESP_GATTC_DIS_SRVC_CMPL_EVT
I (26417) BLE_ANCS: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (26427) BLE_ANCS: ESP_GATTC_SEARCH_CMPL_EVT
I (26427) BLE_ANCS: Find Apple control point char
I (26427) BLE_ANCS: Find Apple noticification source char
I (26437) BLE_ANCS: Find Apple data source char
I (44267) BLE_ANCS: EventID:New message EventFlags:0x12 CategoryID:Unknown CategoryID CategoryCount:1 NotificationUID:97
I (44267) BLE_ANCS: Get detailed information
I (44387) BLE_ANCS: recevice Notification Attributes response Command_id 0 NotificationUID 97
I (44387) Identifier: com.apple.mobile
I (44397) Identifier: phone
I (44397) Title: 10010
I (44397) Message: 当前通话
I (44407) MessageSize: 12
I (44407) Date: 20240310T150845
I (44407) NActionLabel: e7 bb 93 e6 9d 9f e9 80 9a e8 af 9d 
I (49727) BLE_ANCS: EventID:Removed message EventFlags:0x12 CategoryID:Unknown CategoryID CategoryCount:0 NotificationUID:97
I (53387) BLE_ANCS: EventID:New message EventFlags:0x10 CategoryID:Social CategoryCount:23 NotificationUID:98
I (53387) BLE_ANCS: Get detailed information
I (53487) BLE_ANCS: recevice Notification Attributes response Command_id 0 NotificationUID 98
I (53487) Identifier: com.apple.Mobile
I (53487) Identifier: SMS
I (53497) Title: 10010
I (53497) Message: 您的账户当�
I (53497) Message: ��可用余额19
I (53507) Message: 2.84元，本月
I (53507) Message: 消费30.00元�
I (53517) Message: �话���详单及
Multiple decode errors occured: Try checking the baud rate and XTAL frequency setting in menuconfig
I (53517) Message: 套餐余量查�
I (53517) Message: ��请点击链�
I (53527) Message: � https://cc.100
I (53527) Message: 10.com/i/4Lpn09 
I (53537) Message: 或登录中国�
I (53537) Message: ��通APP，更�
Multiple decode errors occured: Try checking the baud rate and XTAL frequency setting in menuconfig
I (53547) Message: �便���捷。
I (53547) MessageSize: 189
I (53547) Date: 20240310T150853
I (53557) NActionLabel: e6 b8 85 e9 99 a4 
```

closed 2024-03-13 by Alvin1Zhang