espressif/esp-idf #7668: Issue with esp_ble_set_encryption() function:  (17933) BT_APPL: bta_dm_set_encryption, not find peer_bdaddr or peer_bdaddr connection state error

rhr407, 2021-10-09:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version : [v4.4-dev-3235-g3e370c4296]
- Build System:         [df.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0

- Operating System:     [Linux]
- Using an IDE?: [Yes (VSCode)]
- Power Supply:         [USB]

## Problem Description

When I call esp_ble_set_encryption() to initiate an encrypted communication with my iPhone inside once the ESP32 is disconnected from the iPhone, I get this error. 

I want to make a connection to iPhone which is running the LightBlue BLE application. I can connect with pairing using my code but my target is to make a connection request from the ESP32 side so that I can connect the phone again. However, I get an error that apparently pertains to encryption context that might have changed by the phone after disconnect.

So, I want to know if there is a way I can make a connection request to a smartphone to which I had previously connected  (securely paired) sometime earlier?

I will highly appreciate it if someone can provide me with some useful advice.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. Open Use gatt_security example inside "esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server" folder
2. Go to the file example_ble_sec_gatts_demo.c
3. Copy and paste my code inside example_ble_sec_gatts_demo.c and test the code.

[image]

### Code to reproduce this issue

```
https://gist.github.com/rhr407/f0f9bd42c3ed3dc1db76d58ce71fd966
```

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2021-11-22:
@rhr407 Thanks for reporting, would you please help share if any updates for the issue? Thanks.

rhr407, 2021-11-22:
Hi @Alvin1Zhang and @xiewenxiang : Thanks a lot for your help. I think the issues should be closed now as I got the answer of my question from @xiewenxiang 

Kind regards

Alvin1Zhang, 2021-11-23:
@rhr407 Thanks for sharing the updates, good to hear you got the answer of the question, feel free to reopen.

closed 2021-11-23 by Alvin1Zhang