espressif/esp-idf #7669: Retrieve data after http POST request is broken in v4.1.2?

fpgamaster, 2021-10-09:
- Development Kit: none
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v4.1.2
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:  Linux

## Problem Description

Can not retrieve data after http POST request, esp_http_client_read() returns 0 after esp_http_client_perform().

The following snipped of code used to work with v4.1.1:

int my_http_request(const char *url, const char *http_request, int http_request_length,  char *http_response, int *http_response_length, int client_id)
{
        esp_http_client_config_t config;
        int err = 0;
        int status_code = 0;

        ESP_LOGI(LOG_TAG, "Requesting URL: %s", url);

        memset(&config, 0, sizeof(config));
        config.url = url;

        http_client[client_id] = esp_http_client_init(&config);
        if (http_client[client_id] == NULL) {
                ESP_LOGE(LOG_TAG, "Failed to initialise HTTP connection");
                return (-1);
        }

        esp_http_client_set_url(http_client[client_id], url);
        //ESP_LOGI(LOG_TAG, "Method POST");
        esp_http_client_set_method(http_client[client_id], HTTP_METHOD_POST);
        esp_http_client_set_post_field(http_client[client_id], http_request, http_request_length);

        err = esp_http_client_perform(http_client[client_id]);
        if (err == ESP_OK) {
                status_code = esp_http_client_get_status_code(http_client[client_id]);
                ESP_LOGD(LOG_TAG, "HTTP Status = %d, content_length = %d",
                esp_http_client_get_status_code(http_client[client_id]),
                esp_http_client_get_content_length(http_client[client_id]));
        } else {
                ESP_LOGE(LOG_TAG, "HTTP request failed: %s", esp_err_to_name(err));
        }

        if (status_code != 200) {
                http_cleanup(http_client[client_id]);
                return (-1);
        }

        ESP_LOGI(LOG_TAG, "buffer=%p, length=%d", http_response, *http_response_length);
        int len_received = 0;
        while (1) {
                int data_read = esp_http_client_read(http_client[client_id], http_response, *http_response_length);
                if (data_read < 0) {
                        ESP_LOGE(LOG_TAG, "Error: HTTP data read error");
                        http_cleanup(http_client[client_id]);
                        return (-2);
                } else if (data_read > 0) {
                        len_received += data_read;
                        ESP_LOGD(LOG_TAG, "Received %d bytes", data_read);
                        break;
                } else if (data_read == 0) {
                        ESP_LOGD(LOG_TAG, "Connection closed, all data received");
                        break;
                }
        }

        http_cleanup(http_client[client_id]);
        http_response[len_received] = '\0';
        *http_response_length = len_received;

        return (err);
}

The server returns the required data, esp_http_client_get_content_length() is not zero but esp_http_client_read() returns 0.

fpgamaster, 2021-10-11:
@ESP-YJM Thank you!
Removing this commit resolves the issue.

Best regards

shubhamkulkarni97, 2021-11-08:
Hi @fpgamaster

> AFAIR I have already tried the second approach you have proposed but it had failed too.

Could you share code snippet which fails to perform HTTP request with second approach?
Ideally it should work fine and is also demonstrated in esp_http_client example

fpgamaster, 2021-11-21:
Hi @shubhamkulkarni97

Yeah, the second approach works as expected.
Sorry, I have messed up something in the previous test.

Best regards

closed 2021-11-22 by Alvin1Zhang