espressif/esp-idf #7670: support flash > 16MB

chegewara, 2021-10-09:
Can we expect in near future to have support for flash size > 16MB?
Withe very small effort i could build bootloader that support 32MB, but that is not the end. This should be extended to file sytem, because `f_mkfs` failed with partition >16MB:
`E (3980) vfs_fat_spiflash: f_mkfs failed (14)`

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x17f0
load:0x4004c000,len:0xa60
load:0x40050000,len:0x2e68
entry 0x4004c208
[0;32mI (21) boot: ESP-IDF v4.4-dev-3235-g3e370c4296-dirty 2nd stage bootloader[0m
[0;32mI (21) boot: compile time 23:26:08[0m
[0;32mI (22) boot: chip revision: 0[0m
[0;32mI (26) boot.esp32s2: SPI Speed      : 80MHz[0m
[0;32mI (31) boot.esp32s2: SPI Mode       : DIO[0m
[0;32mI (35) boot.esp32s2: SPI Flash Size : 32MB[0m
[0;32mI (40) boot: Enabling RNG early entropy source...[0m
[0;32mI (46) boot: Partition Table:[0m
[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00005000[0m
[0;32mI (64) boot:  1 otadata          OTA data         01 00 0000e000 00002000[0m
[0;32mI (72) boot:  2 app0             OTA app          00 10 00010000 00060000[0m
[0;32mI (79) boot:  3 ffat             Unknown data     01 81 00070000 01e00000[0m
[0;32mI (86) boot: End of partition table[0m
[0;32mI (91) boot: No factory image, trying OTA 0[0m
[0;32mI (95) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=069cch ( 27084) map[0m
[0;32mI (109) esp_image: segment 1: paddr=000169f4 vaddr=3ffbc8b0 size=01c38h (  7224) load[0m
[0;32mI (114) esp_image: segment 2: paddr=00018634 vaddr=40022000 size=079e4h ( 31204) load[0m
[0;32mI (129) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=11b94h ( 72596) map[0m
[0;32mI (144) esp_image: segment 4: paddr=00031bbc vaddr=400299e4 size=02ecch ( 11980) load[0m
[0;32mI (147) esp_image: segment 5: paddr=00034a90 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (155) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (191) boot: Set actual ota_seq=1 in otadata[0][0m
[0;32mI (191) boot: Disabling RNG early entropy source...[0m
[0;32mI (202) ca⸮⸮fÆ~⸮⸮3?⸮⸮⸮⸮⸮,W⸮⸮⸮V⸮,ZY⸮B⸮⸮size 8KB, 4Ways, cache line size 32Byte[0m
[0;32mI (202) cpu_start: Pro cpu up.[0m
[0;32mI (214) cpu_start: Pro cpu start user code[0m
[0;32mI (214) cpu_start: cpu freq: 160000000[0m
[0;32mI (214) cpu_start: Application information:[0m
[0;32mI (218) cpu_start: Project name:     hello-world[0m
[0;32mI (224) cpu_start: App version:      v4.4-dev-3235-g3e370c4296-dirty[0m
[0;32mI (231) cpu_start: Compile time:     Oct  9 2021 23:26:08[0m
[0;32mI (237) cpu_start: ELF file SHA256:  5e63f98550541ccc...[0m
[0;32mI (243) cpu_start: ESP-IDF:          v4.4-dev-3235-g3e370c4296-dirty[0m
[0;32mI (250) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (257) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM[0m
[0;32mI (263) heap_init: At 3FFBED38 len 0003D2C8 (244 KiB): DRAM[0m
[0;32mI (270) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM[0m
[0;32mI (276) spi_flash: detected chip: winbond[0m
[0;32mI (281) spi_flash: flash io: dio[0m
[0;32mI (285) cpu_start: Starting scheduler on PRO CPU.[0m
Hello world!
This is esp32s2 chip with 1 CPU core(s), WiFi, silicon revision 0, 32MB external flash
Minimum free heap size: 256140 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
```

chegewara, 2021-10-09:
Here is proof of concept:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x17f0
load:0x4004c000,len:0xa60
load:0x40050000,len:0x2e68
entry 0x4004c208
[0;32mI (21) boot: ESP-IDF v4.4-dev-3235-g3e370c4296-dirty 2nd stage bootloader[0m
[0;32mI (21) boot: compile time 23:26:08[0m
[0;32mI (22) boot: chip revision: 0[0m
[0;32mI (26) boot.esp32s2: SPI Speed      : 80MHz[0m
[0;32mI (31) boot.esp32s2: SPI Mode       : DIO[0m
[0;32mI (35) boot.esp32s2: SPI Flash Size : 32MB[0m
[0;32mI (40) boot: Enabling RNG early entropy source...[0m
[0;32mI (46) boot: Partition Table:[0m
[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00005000[0m
[0;32mI (64) boot:  1 otadata          OTA data         01 00 0000e000 00002000[0m
[0;32mI (72) boot:  2 app0             OTA app          00 10 00010000 00060000[0m
[0;32mI (79) boot:  3 ffat             Unknown data     01 81 00070000 01e00000[0m
[0;32mI (86) boot:  4 ffat1            Unknown data     01 81 01e70000 00010000[0m
[0;32mI (94) boot: End of partition table[0m
[0;32mI (98) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0e968h ( 59752) map[0m
[0;32mI (118) esp_image: segment 1: paddr=0001e990 vaddr=3ffbfaa0 size=01688h (  5768) load[0m
[0;32mI (120) esp_image: segment 2: paddr=00020020 vaddr=40080020 size=2df7ch (188284) map[0m
[0;32mI (163) esp_image: segment 3: paddr=0004dfa4 vaddr=3ffc1128 size=007a8h (  1960) load[0m
[0;32mI (164) esp_image: segment 4: paddr=0004e754 vaddr=40024000 size=0ba98h ( 47768) load[0m
[0;32mI (181) esp_image: segment 5: paddr=0005a1f4 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (187) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (187) boot: Disabling RNG early entropy source...[0m
MSC lun 1 begin
E (3994) vfs_fat_spiflash: f_mkfs failed (14)
```
[image]

chegewara, 2021-10-11:
Actually its not possible yet. There is something odd with file system which does not allow to read/write properly flash that with that bootloader. I have no explanation for it yet, since there are no logs at all about any problems with file system.
The problem is i am testing it with arduino, so i cant change logging level and i also dont have time to investigate it further at the moment.

I can eventually prepare patches, so espressif could investigate it further. I mean patches, because it is small update in `bootloader_support` component and `esptool`.

0xjakob, 2021-10-11:
> I can eventually prepare patches, so espressif could investigate it further. I mean patches, because it is small update in bootloader_support component and esptool.

That would be great.

chegewara, 2021-10-11:
That part is very simple:
[32mbflash.patch.txt](https://github.com/espressif/esp-idf/files/7319325/32mbflash.patch.txt)
[32mbflash-esptool.patch.txt](https://github.com/espressif/esp-idf/files/7319326/32mbflash-esptool.patch.txt)

ESP32DE, 2021-10-11:
> 
> 
> Actually its not possible yet. There is something odd with file system which does not allow to read/write properly flash that with that bootloader. I have no explanation for it yet, ...
> 
[did you format the filesystem?
how you format the file system?](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/fatfs/vfs/vfs_fat_spiflash.c#L89-L97)

([ further](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/fatfs/src/ff.h#L380-L391) ) 

`
fresult = f_mkfs(drv, FM_ANY | FM_SFD, alloc_unit_size, workbuf, workbuf_size);
`

also how looks your menuconfig setup for sectore size ..
FAT32..EXFAT

[a](https://github.com/espressif/esp-idf/search?q=FM_ANY)) [b](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/fatfs/src/ff.c#L5609-L5633)) [c](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/fatfs/src/ff.c#L5751-L5861)) 

i am sure now you can complet it :)

best wishes
rudi ;-)

chegewara, 2021-10-11:
Yes, i am using the same procedure to format 15MB partition and 31MB partition. With standard bootloader i can format patition using ubuntu OS and with this modified bootloader i cant format even 1MB partition.

igrr, 2021-10-11:
@chegewara Are you able to run the `storage/wear_levelling` example with your changes, for example with 30MB FAT partition size? I seem to be able to do this locally on an ESP32-S3 with a 32 MB MXIC flash chip. Haven't checked on an ESP32-S2 yet.

The patches I'm using are basically the same as yours:
[flash_size_gt_16mb_esptool.patch.txt](https://github.com/espressif/esp-idf/files/7325063/flash_size_gt_16mb_esptool.patch.txt)
[flash_size_gt_16mb.patch.txt](https://github.com/espressif/esp-idf/files/7325064/flash_size_gt_16mb.patch.txt)

By the way, which flash chip are you using? `esptool.py flash_id` should print this information.

chegewara, 2021-10-11:
hi @igrr 
seems i am unlucky, because i have 2 atmegazero boards and on both debug port connected to onboard CP210x dont work. I also dont have luck getting `flash_id` over OTG port.
https://atmegazero.com/#/atmegazero_esp32s2_specifications

From PCB i can tell it is `winbond 25Q256JVEQ`.
And no, i didnt try esp-idf wera_levelling example yet.

When i built hello_world example i can see whole 32MB used and i can assign all space to factory partition, but like i said, there is something odd with USB MSC even when i create only 1MB FFAT partition. With regular bootloader it is working, but flashing upgraded bootloader cause problems.
Ubuntu is not reporting anything wrong, or i just didnt find anything, and i dont see any logs at all.

btw, is there any chance to have S3 devkit with 1GB flash and/or PSRAM from espressif any time soon?

chegewara, 2021-10-11:
@igrr i think all is good with our patches
before ive been testing 16MB flash setup with arduino bootloaders and it worked, but now i tested with bootloader i generated in hello_world example and i have the same issue as with 32MB pimped bootloader.

I believe it is some problem on my side and compatibility with arduino rather FS or esp-idf.

From patch i see we will have 32 and 64MB support, now we are awaiting new devkits

Thanks

ESP32DE, 2021-10-11:
@chegewara 
if you can't read twitter
herer your are
[image]

chegewara, 2021-10-11:
1GB means to me exactly the same as to any programmer or person with IT knowledge (which im not)
Thats why in title is 16MB, request to support 32MB and from @igrr patch we can see there will be up to 64MB support.
Capital `B` always will be bytes and lower `b` a bit

Since in official espressif TRM it is `1GB`  i do not expect nothing more and nothing less.

igrr, 2022-02-21:
We've just merged @ESP32DE's PR https://github.com/espressif/esp-idf/pull/7688 (sorry that it took a while!). The merge commit should appear on Github in the next few hours.

Now the basic usage of 32MB and larger flash for data storage should work.

Note that there are still some things which aren't supported, like placing the app itself outside of the lower 16MB area. See https://github.com/espressif/esp-idf/issues/8365 for a few more things which aren't working yet. We'll update that issue when support for large flash sizes is improved.

closed 2022-02-21 by igrr

thyclouds, 2022-06-15:
Hi @igrr, @ESP32DE, all!

@igrr said > Now the basic usage of 32MB and larger flash for data storage should work
Could you please elaborate how to use SPI flash (external or internal) above 16MB? We are using the latest esp-idf and are able to detect 128MB flash, but when we are using esptool.py to map out partitions, we are still locked within the first 16MB? How can we create one large or multiple partitions of 16MB? 
Your help would be greatly appreciated!