espressif/esp-idf #7677: Pthread pthread_mutex_lock() can't work with pthread_cond_wait()

wenxingguo, 2021-10-12:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): ESP-IDF v4.2-dirty
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3-5-g39b36b0) 8.4.0
- Operating System:  Linux

- Power Supply: USB
## Problem Description
This code does not work, but works well in linux, macos 

********************************************* main.c************************************
#include <pthread.h>
#include <stdio.h>
#include <unistd.h>

static char* message= "hello world";

pthread_cond_t cond = PTHREAD_COND_INITIALIZER;
pthread_mutex_t mutex = PTHREAD_MUTEX_INITIALIZER;

void* helloworld(void *args){
	pthread_mutex_lock(&mutex);
	while (1)
	{
		pthread_cond_wait(&cond, &mutex);
		printf("%s\n", message);
		continue;
	}
}

void app_main(void){

	pthread_t thread_id;
	pthread_create(&thread_id, NULL, helloworld, (void*) message);
	while (1)
	{
		pthread_cond_signal(&cond);
		sleep(1);
	}
}
****************************************main.c*************************************

### Expected Behavior

### Actual Behavior

### Steps to reproduce

### Code to reproduce this issue

```c
#include <pthread.h>
#include <stdio.h>
#include <unistd.h>

static char* message= "hello world";

pthread_cond_t cond = PTHREAD_COND_INITIALIZER;
pthread_mutex_t mutex = PTHREAD_MUTEX_INITIALIZER;

void* helloworld(void *args){
	pthread_mutex_lock(&mutex);
	while (1)
	{
		pthread_cond_wait(&cond, &mutex);
		printf("%s\n", message);
		continue;
	}
}

void app_main(void){

	pthread_t thread_id;
	pthread_create(&thread_id, NULL, helloworld, (void*) message);
	while (1)
	{
		pthread_cond_signal(&cond);
		sleep(1);
	}
}

```
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7028
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (29) boot: compile time 00:45:30
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x058e0 ( 22752) map
I (120) esp_image: segment 1: paddr=0x00015908 vaddr=0x3ffb0000 size=0x02058 (  8280) load
I (124) esp_image: segment 2: paddr=0x00017968 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /opt/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (130) esp_image: segment 3: paddr=0x00017d74 vaddr=0x40080404 size=0x082a4 ( 33444) load
I (153) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x13078 ( 77944) map
0x400d0020: _stext at ??:?

I (183) esp_image: segment 5: paddr=0x000330a0 vaddr=0x400886a8 size=0x01cc4 (  7364) load
0x400886a8: rtc_clk_cpu_freq_to_xtal at /opt/esp/esp-idf/components/soc/src/esp32/rtc_clk.c:431

I (192) boot: Loaded app from partition at offset 0x10000
I (192) boot: Disabling RNG early entropy source...
I (195) cpu_start: Pro cpu up.
I (198) cpu_start: Application information:
I (203) cpu_start: Project name:     xtaskblink
I (208) cpu_start: App version:      1
I (213) cpu_start: Compile time:     Oct 12 2021 00:42:06
I (219) cpu_start: ELF file SHA256:  7933ff8cb83c6db9...
I (225) cpu_start: ESP-IDF:          v4.2-dirty
I (230) cpu_start: Starting app cpu, entry point is 0x400816b8
0x400816b8: call_start_cpu1 at /opt/esp/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (241) heap_init: Initializing. RAM available for dynamic allocation:
I (247) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (253) heap_init: At 3FFB2890 len 0002D770 (181 KiB): DRAM
I (260) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (266) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (272) heap_init: At 4008A36C len 00015C94 (87 KiB): IRAM
I (279) cpu_start: Pro cpu start user code
I (297) spi_flash: detected chip: gd
I (297) spi_flash: flash io: dio
W (298) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (308) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008271f  PS      : 0x00060230  A0      : 0x80082884  A1      : 0x3ffb4660  
0x4008271f: lock_acquire_generic at /opt/esp/esp-idf/components/newlib/locks.c:126

A2      : 0xffffffff  A3      : 0xffffffff  A4      : 0x00000004  A5      : 0x3ffafc40  
A6      : 0x00000000  A7      : 0x3ffb4680  A8      : 0x800848e8  A9      : 0x3ffb4620  
A10     : 0x00000003  A11     : 0x00060423  A12     : 0x00060420  A13     : 0x00060423  
A14     : 0x00000001  A15     : 0x3f400010  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4008271c:0x3ffb4660 0x40082881:0x3ffb4690 0x400d54aa:0x3ffb46b0 0x400d300d:0x3ffb46d0 0x400d1cb2:0x3ffb4700 0x4008468d:0x3ffb4730
0x4008271c: lock_acquire_generic at /opt/esp/esp-idf/components/newlib/locks.c:125

0x40082881: _lock_acquire_recursive at /opt/esp/esp-idf/components/newlib/locks.c:170

0x400d54aa: pthread_cond_signal at /opt/esp/esp-idf/components/pthread/pthread_cond_var.c:55

0x400d300d: app_main at /Users/wood/Documents/esp_project/xtaskblink/main/main.c:27 (discriminator 1)

0x400d1cb2: main_task at /opt/esp/esp-idf/components/esp32/cpu_start.c:600 (discriminator 2)

0x4008468d: vPortTaskWrapper at /opt/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: 7933ff8cb83c6db9

Rebooting...
ets Jun  8 2016 00:22:57

## Other items if possible

wenxingguo, 2021-10-12:
but when I change the code
```c
#include <pthread.h>
#include <stdio.h>
#include <unistd.h>

static char* message= "hello world";

pthread_cond_t cond = PTHREAD_COND_INITIALIZER;
pthread_mutex_t mutex = PTHREAD_MUTEX_INITIALIZER;

void* helloworld(void *args){
	pthread_mutex_lock(&mutex);
	while (1)
	{
		pthread_cond_wait(&cond, &mutex);
		printf("%s\n", message);
		continue;
	}
}

void app_main(void){

	pthread_t thread_id;
	pthread_create(&thread_id, NULL, helloworld, NULL);
	sleep(1);
	while (1)
	{
		pthread_cond_signal(&cond);
		sleep(1);
	}
}
```
Other error appears

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7028
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (29) boot: compile time 00:45:30
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x058e0 ( 22752) map
I (121) esp_image: segment 1: paddr=0x00015908 vaddr=0x3ffb0000 size=0x02058 (  8280) load
I (124) esp_image: segment 2: paddr=0x00017968 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /opt/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (130) esp_image: segment 3: paddr=0x00017d74 vaddr=0x40080404 size=0x082a4 ( 33444) load
I (153) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x13078 ( 77944) map
0x400d0020: _stext at ??:?

I (183) esp_image: segment 5: paddr=0x000330a0 vaddr=0x400886a8 size=0x01cc4 (  7364) load
0x400886a8: rtc_clk_cpu_freq_to_xtal at /opt/esp/esp-idf/components/soc/src/esp32/rtc_clk.c:431

I (193) boot: Loaded app from partition at offset 0x10000
I (193) boot: Disabling RNG early entropy source...
I (195) cpu_start: Pro cpu up.
I (199) cpu_start: Application information:
I (204) cpu_start: Project name:     xtaskblink
I (209) cpu_start: App version:      1
I (213) cpu_start: Compile time:     Oct 12 2021 00:42:06
I (219) cpu_start: ELF file SHA256:  a248ce8ab29fc5fe...
I (225) cpu_start: ESP-IDF:          v4.2-dirty
I (230) cpu_start: Starting app cpu, entry point is 0x400816b8
0x400816b8: call_start_cpu1 at /opt/esp/esp-idf/components/esp32/cpu_start.c:287

I (223) cpu_start: App cpu up.
I (241) heap_init: Initializing. RAM available for dynamic allocation:
I (248) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (254) heap_init: At 3FFB2890 len 0002D770 (181 KiB): DRAM
I (260) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (267) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (273) heap_init: At 4008A36C len 00015C94 (87 KiB): IRAM
I (279) cpu_start: Pro cpu start user code
I (297) spi_flash: detected chip: gd
I (298) spi_flash: flash io: dio
W (298) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (308) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008271f  PS      : 0x00060230  A0      : 0x80082884  A1      : 0x3ffb4660  
0x4008271f: lock_acquire_generic at /opt/esp/esp-idf/components/newlib/locks.c:126

A2      : 0xffffffff  A3      : 0xffffffff  A4      : 0x00000004  A5      : 0x3ffafc40  
A6      : 0x00000000  A7      : 0x3ffb4680  A8      : 0x800848e8  A9      : 0x3ffb4620  
A10     : 0x00000003  A11     : 0x00060423  A12     : 0x00060420  A13     : 0x00060423  
A14     : 0x00000001  A15     : 0x3f400010  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4008271c:0x3ffb4660 0x40082881:0x3ffb4690 0x400d54aa:0x3ffb46b0 0x400d300d:0x3ffb46d0 0x400d1cb2:0x3ffb4700 0x4008468d:0x3ffb4730
0x4008271c: lock_acquire_generic at /opt/esp/esp-idf/components/newlib/locks.c:125

0x40082881: _lock_acquire_recursive at /opt/esp/esp-idf/components/newlib/locks.c:170

0x400d54aa: pthread_cond_signal at /opt/esp/esp-idf/components/pthread/pthread_cond_var.c:55

0x400d300d: app_main at /Users/wood/Documents/esp_project/xtaskblink/main/main.c:27 (discriminator 1)

0x400d1cb2: main_task at /opt/esp/esp-idf/components/esp32/cpu_start.c:600 (discriminator 2)

0x4008468d: vPortTaskWrapper at /opt/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: a248ce8ab29fc5fe

o-marshmallow, 2021-10-21:
I am closing this issue then. Feel free to re-open if you still encounter such issue.

closed 2021-10-21 by o-marshmallow