espressif/esp-idf #7687: esp_wifi_internal_set_fix_rate not affecting RTT when used with ESP Now

neel, 2021-10-13:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4] 
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-3235-g3e370c4296
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: Not Windows
- Using an IDE?: [No]
- Power Supply:  [USB]

## Problem Description

ESP NOW p2p RTT unchanged with or without `esp_wifi_internal_set_fix_rate`. 

[espnow-speed-test.zip](https://github.com/espressif/esp-idf/files/7340906/espnow-speed-test.zip)

In the linked project, I am sending a `request` from the source node that contains a origin time. The target node receives the request and sends a reply that contains the same origin time, and the targets node's time. The source node gets the reply and stores that in a container. The source node sends 20 requests, then wait for 5s and then prints whatever results it has gathered.

I am observing **no significant change in RTT with or without `esp_wifi_internal_set_fix_rate()`**

### Expected Behavior

RTT should be less when esp_wifi_internal_set_fix_rate() is used 

### Actual Behavior

#### When esp_wifi_internal_set_fix_rate() is used

|   origin  |    hit    |   return  |    RTT   |
|-----------|-----------|-----------|--------------|
| 885626886 | 877812654 | 885636045 | 9.159000ms   |
| 885627389 | 877815568 | 885641730 | 14.341000ms  |
| 885627581 | 877821395 | 885647545 | 19.964000ms  |
| 885627774 | 877827194 | 885653334 | 25.560000ms  |
| 885627966 | 877833018 | 885659168 | 31.202000ms  |
| 885628158 | 877838817 | 885664940 | 36.782000ms  |
| 885628351 | 877844664 | 885673852 | 45.501000ms  |
| 885628543 | 877850514 | 885679669 | 51.126000ms  |
| 885628736 | 877853536 | 885685502 | 56.766000ms  |
| 885628928 | 877859334 | 885691310 | 62.382000ms  |
| 885629120 | 877865173 | 885697107 | 67.987000ms  |
| 885629313 | 877870967 | 885702906 | 73.593000ms  |
| 885629505 | 877876795 | 885708721 | 79.216000ms  |
| 885629698 | 877882563 | 885714529 | 84.831000ms  |
| 885629893 | 877888387 | 885720326 | 90.433000ms  |
| 885630086 | 877894186 | 885726145 | 96.059000ms  |
| 885630407 | 877900010 | 885732038 | 101.631000ms |
| 885630599 | 877905809 | 885737828 | 107.229000ms |
| 885630791 | 877911606 | 885740797 | 110.006000ms |
| 885630984 | 877917503 | 885743736 | 112.752000ms |

#### When esp_wifi_internal_set_fix_rate() is NOT used

|   origin |    hit   |   return |    RTT       |
|----------|----------|----------|--------------|
| 15498063 | 12340151 | 15507240 | 9.177000ms   |
| 15498575 | 12343095 | 15512918 | 14.343000ms  |
| 15498799 | 12348923 | 15518715 | 19.916000ms  |
| 15499003 | 12354742 | 15529053 | 30.050000ms  |
| 15499208 | 12360535 | 15534821 | 35.613000ms  |
| 15499412 | 12365005 | 15540645 | 41.233000ms  |
| 15499616 | 12370826 | 15546421 | 46.805000ms  |
| 15499826 | 12376615 | 15552240 | 52.414000ms  |
| 15500030 | 12382422 | 15558054 | 58.024000ms  |
| 15500234 | 12388229 | 15563882 | 63.648000ms  |
| 15500439 | 12394036 | 15569697 | 69.258000ms  |
| 15500643 | 12399870 | 15575516 | 74.873000ms  |
| 15500847 | 12405659 | 15581293 | 80.446000ms  |
| 15501051 | 12411486 | 15587121 | 86.070000ms  |
| 15501256 | 12417302 | 15592916 | 91.660000ms  |
| 15501611 | 12423109 | 15598735 | 97.124000ms  |
| 15501831 | 12428907 | 15604614 | 102.783000ms |
| 15502035 | 12434730 | 15610415 | 108.380000ms |
| 15502239 | 12440622 | 15614883 | 112.644000ms |
| 15502444 | 12446403 | 15617789 | 115.345000ms |

### Steps to reproduce
[espnow-speed-test.zip](https://github.com/espressif/esp-idf/files/7340906/espnow-speed-test.zip)

1. Download the attached Project in zip
2. Unzip it
3. cd into that directory
4. Set the MAC address of two nodes (AP interface) in main/main.c line 32 and 33 
5. idf.py build
6. idf.py flash into two nodes

KWolfe81, 2021-11-23:
I've duplicated the @neel's results on the ESP32-S2 and am interested in finding a fix.

MaxwellAlan, 2021-11-24:
hi @neel @KWolfe81 since release/v4.3 we have add an new API to config espnow related rate , can you try with below api
[esp_err_t esp_wifi_config_espnow_rate(wifi_interface_t ifx, wifi_phy_rate_t rate);](https://github.com/espressif/esp-idf/blob/a18cb7090511d235ebe3c8b833180630f4f8bb33/components/esp_wifi/include/esp_wifi.h#L1225)

neel, 2021-11-26:
@MaxwellAlan Yes worked using the new API but the performance is little slower that before.

I am trying the following

    #if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 3, 0)
        esp_wifi_config_espnow_rate(WIFI_IF_AP, WIFI_PHY_RATE_MCS7_SGI);
        // esp_wifi_config_80211_tx_rate(WIFI_IF_AP, WIFI_PHY_RATE_MCS7_SGI);
    #endif 

Is the patch number in the version guard correct ? What surprises me is if I uncomment the `esp_wifi_config_80211_tx_rate` line then performance is even slower which is opposite of what I expected. I am using `release/v4.4` now. Also what is the impact of `esp_wifi_config_11b_rate` on performance ?

I see that, none of these functions require `AMPDU` to be disabled. Should it still be disabled ? 

Also, is the old internal function obsolete now ? Then it should be removed from the headers ? or at least throw some `#error` compile time ?

MaxwellAlan, 2022-03-30:
Hi @neel 
> ```
> #if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 3, 0)
>     esp_wifi_config_espnow_rate(WIFI_IF_AP, WIFI_PHY_RATE_MCS7_SGI);
>     // esp_wifi_config_80211_tx_rate(WIFI_IF_AP, WIFI_PHY_RATE_MCS7_SGI); [...]

The patch number is right.
As for esp_wifi_config_80211_tx_rate, I believe it does not have any influence on espnow, this api is only for 80211_tx.
While esp_wifi_config_11b_rate this api is target for to turn off 11b rate, which has a little improvement for performace.

> I see that, none of these functions require `AMPDU` to be disabled. Should it still be disabled ?

Not needed now. 

> 
> Also, is the old internal function obsolete now ? Then it should be removed from the headers ? or at least throw some `#error` compile time ?

Old internal function still works now, but for espnow and raw 80211_tx , we add new api to config these pkts rates. Old internal api will affect all pkts rates.

What's more, we add HT40 support for espnow, you can call below two apis to support ht40 now.
esp_wifi_set_bandwidth();
esp_wifi_set_channel();
This will improve some performace, please have a try.

Thanks.

jack0c, 2022-05-20:
@neel any updates?

closed 2022-07-12 by Alvin1Zhang