espressif/esp-idf #7699: Connection to random addressed BLE clients always fail

danergo, 2021-10-15:
## Environment

- Development Kit: NodeMCU-ESP-32S
- Module or chip used:  ESP-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-2533-g83956ebbae
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description

I'm using ESP32 as a Bluetooth co-processor, with the ESP-IDF provided hciuart sample. (https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/controller_hci_uart_esp32).

Host is a RPi, with very up to date softwares:

```
BlueZ: 5.62 (compiled from source)
Kernel: 5.10.63+
Release: Raspbian GNU/Linux 11 (Bullseye)
```

However, I can't connect to devices which requires random addressing.

### Expected Behavior

ESP32 should be able to connect to random addressed clients.

### Actual Behavior

ESP32 is not able to connect to random addressed clients.

### Steps to reproduce

```
gatttool -t random -b MAC_TO_CONN -I
[MAC_TO_CONN][LE]> connect
Attempting to connect to MAC_TO_CONN
Error: connect to MAC_TO_CONN: Function not implemented (38)
```

During this operation here is my btmon output (recorded parallel):

```
Bluetooth monitor ver 5.62
= Note: Linux version 5.10.63+ (armv6l)                                0.788425
= Note: Bluetooth subsystem version 2.22                               0.788450
= New Index: HCI_MAC (Primary,UART,hci0)              [hci0] 0.788457
= Open Index: HCI_MAC                                 [hci0] 0.788463
= Index Info: HCI_MAC (RivieraWaves S.A.S)            [hci0] 0.788468
@ MGMT Open: bluetoothd (privileged) version 1.18             {0x0001} 0.788476
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7   #1 [hci0] 6.115369
        Type: Passive (0x00)
        Interval: 60.000 msec (0x0060)
        Window: 60.000 msec (0x0060)
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
> HCI Event: Command Complete (0x0e) plen 4                  #2 [hci0] 6.118578
      LE Set Scan Parameters (0x08|0x000b) ncmd 5
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2       #3 [hci0] 6.118737
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                  #4 [hci0] 6.121824
      LE Set Scan Enable (0x08|0x000c) ncmd 5
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 18                    #5 [hci0] 6.181503
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Random (0x01)
        Address: MAC_TO_CONN (Static)
        Data length: 6
        Flags: 0x04
          BR/EDR Not Supported
        TX power: -8 dBm
        RSSI: -95 dBm (0xa1)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2       #6 [hci0] 6.181690
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                  #7 [hci0] 6.183300
      LE Set Scan Enable (0x08|0x000c) ncmd 5
        Status: Success (0x00)
< HCI Command: LE Create Connection (0x08|0x000d) plen 25    #8 [hci0] 6.183450
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: Accept list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: MAC_TO_CONN (Static)
        Own address type: Public (0x00)
        Min connection interval: 30.00 msec (0x0018)
        Max connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                    #9 [hci0] 6.242149
      LE Create Connection (0x08|0x000d) ncmd 5
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31                   #10 [hci0] 6.345657
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 0
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: MAC_TO_CONN (Static)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Central clock accuracy: 0x00
@ MGMT Event: Device Connected (0x000b) plen 19        {0x0001} [hci0] 6.345823
        LE Address: MAC_TO_CONN (Static)
        Flags: 0x00000000
        Data length: 6
        Flags: 0x04
          BR/EDR Not Supported
        TX power: -8 dBm
< HCI Command: LE Read Remote Used... (0x08|0x0016) plen 2  #11 [hci0] 6.346246
        Handle: 0
> HCI Event: Command Status (0x0f) plen 4                   #12 [hci0] 6.355264
      LE Read Remote Used Features (0x08|0x0016) ncmd 5
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4              #13 [hci0] 6.636850
        Status: Success (0x00)
        Handle: 0
        Reason: Connection Failed to be Established (0x3e)
@ MGMT Event: Device Disconnected (0x000c) plen 8      {0x0001} [hci0] 6.636961
        LE Address: MAC_TO_CONN (Static)
        Reason: Unspecified (0x00)
```

Also, during this operation this gets printed to ESP32's debug port:

```
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
```

Please help me with this, or either confirm me if ESP32's Bluetooth is not working with random addressing.

danergo, 2021-10-16:
Nice idea, have just done that! Will update this thread tomorrow with the new results.

Thank you so much!

danergo, 2021-10-16:
Well, sorry, I missed something: it seems I don't have any usable logs:

[image]

Those are in Bluedroid, but this example is not using that, so it's marked as red, thus not usable.

Could you tell me where are the logging settings for normal bluetooth?

chegewara, 2021-10-16:
Ok, my bad. Maybe regular log level verbose can help.

xulongzhuang, 2021-10-19:
Does the peer device use a static random address or a private random address? Is it possible that the private random address has been updated and the host connected to the old address caused this situation?
I use the modified `gatt_server_service_table` as the random address device and `gatt_sectury_server` as the private address device. The linux PC is used as the host to connect to the above two devices respectively, and they can be connected normally.

danergo, 2021-10-19:
@xulongzhuang: honestly, I don't know if the peer device is using static or private random address. However I know, that with RPi4's Cypress BT module with same BlueZ package, same Bullseye Debian OS, this command can connect:

```
gatttool -t random -b MAC_TO_CONN -I
[MAC_TO_CONN][LE]> connect
```

From this I assume that the "MAC_TO_CONN" peer has a static random address, as it's not changing.

So to conclude: command which works with Cypress BT is not working with ESP32 and I'm searching for the reason.

Thank you so much for your reply!

xiewenxiang, 2021-10-21:
@danergo 

The log indicates that the connection failed to be established(reason:0x3e). 

the log `lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)` just indicate that the connection is abnormal and requirs further analysis.

I have a suspicion tjat the connection request paramters are not correct. such as:

```c
        Peer address type: Random (0x01)
        Peer address: MAC_TO_CONN (Static)
```

It's conflicting.

According to the type of crash, it maybe access a NULL pointer.

xiewenxiang, 2021-10-21:
```c
> HCI Event: Disconnect Complete (0x05) plen 4              #13 [hci0] 6.636850
        Status: Success (0x00)
        Handle: 0
        Reason: Connection Failed to be Established (0x3e)
```

The 0x3E error code indicates that the connection has failed to be established. what's the probability of that happening?

Could you please use bluedroid host to verify whether the connection can be established?